verilator/verilator #3852: Lint

KingkingLiu, 2023-01-10:
How can i add some new check rule on the basis of lint, because the check rule is too fewe help - what is your question?

wsnyder, 2023-01-10:
Sorry, I don't understand your question.

KingkingLiu, 2023-01-11:
> 

> 

We know that verilator lint can help check for errors in code, but its checking items are limited.I wanted to extend some new checks, but the complex code kept me from finding the interface.How can I find lint's checker code so I can make changes or additions

wsnyder, 2023-01-11:
There is no single place for lint, checks occur where the other algorithms have appropriate data structures.

Please file a separate bug for each check you want to add, or at least the first one or two, and we can discuss where to add it.

closed 2023-01-11 by wsnyder

KingkingLiu, 2023-01-12:
> You can add a new rule, but you need to write C++ code to do it.
> 
> There is some description of internals under docs/internals.rst. In general you will need to dig in to the code to understand things.
> 
> As mentioned above, if you say what rule you want added it will be easier to discuss.

I want to debug verilator locally with vscode, but I dont know how to start.

wsnyder, 2023-01-13:
I don't use vscode, I presume there are good youtube tutorials.

KingkingLiu, 2023-01-30:
> I want to debug verilator locally with vscode, but I dont know how to start.

Thank you for generous help! As far as I know, the quantity of lint rules is about 98. You tell me that I need to provide the specific  lint rule to you. Hence I take the rule from spyglass lint, named W88, which flags memories where all their elements are not set in the design. Below is the code of the example :
module test(in1, in2, out1);
input [3:0] in1, in2;
output [3:0] out1;
reg [3:0] out1;
reg [7:0] mem [0:3];
always@(in1) begin 
out1 = in1 & in2;
mem[0] = {in1, out1};
mem[2] ={in1, in2};
end 
endmodule
Looking forward to your reply.

wsnyder, 2023-01-30:
Makes sense, please file a new issue to add that warning.