verilator/verilator #3858: Localparams in interfaces derived from parameters on the interface can be incorrectly computed

justinthiel, 2023-01-11:
Posted a test for this in [#3857](https://github.com/verilator/verilator/pull/3857).

It appears that the value of localparams defined within an interface that reference parameters defined on the interface will not always be computed properly when different forms of said interface are instantiated.

More specifically we see the following:
``
    top.t.intf:              symbolsPerBeat  16, symbolsPerBeatDivBy2  8, mismatch 0
    top.t.theCore.core_intf: symbolsPerBeat  64, symbolsPerBeatDivBy2  8, mismatch 1
``    
In this case 'intf' is created and symbolsPerBeatDivBy2 is extracted from the interface in a submodule. Once this is done future interfaces of the same type (i.e. 'core_intf' above) seem to inherit the symbolsPerBeatDivBy2 computed for the first declared interface ('intf') even though the symbolsPerBeat (which determines the correct value for symbolsPerBeatDivBy2) value is different for the second instantiated interface.

We are attempting to debug this locally, but I wanted to post a test in the mean time in case anyone has an ideas on how to fix the problem.

wsnyder, 2023-01-11:
Thanks for the test and debugging. I suspect in V3Param the dotted reference isn't making sure that the interface referenced has been processed before committing the value.

AndrewNolte, 2023-01-26:
Any idea where to go on this @wsnyder? I am able to get the test to pass by setting any_overrides to true, or removing cellPinCleanup. I'm not quite sure how nodeDeparamCommon works. Would I put a check in here to see if the interface has been processed?

wsnyder, 2023-01-26:
nodeDeparam converts different parameterized modules/interfaces to different names.   If setting any_overrides works perhaps the code thinks there is a duplicate where there isn't - that is it doesn't recognize a parameter change?  Note #3866 was an issue like that.

closed 2024-01-11 by wsnyder