verilator/verilator #3868: Call abstract class' ctor from a non-abstract class' ctor

Zvord, 2023-01-13:
Hello!

The following code does not compile:

```systemverilog
virtual class foo;
    function new(); endfunction
endclass

class bar extends foo;
    function new();
        super.new();
    endfunction
endclass
```

The error message:
> Illegal to call 'new' using an abstract virtual class (IEEE 1800-2017 8.21)

The referenced chapter explicitly allows to call a constructor of the abstract class if the call is initiated in an non-abstract class constructor:
> An object of an abstract class shall not be constructed directly. Its constructor may only be called indirectly through the chaining of constructor calls originating in an extended non-abstract object.

Version: Verilator 5.004 2022-12-14 rev UNKNOWN.REV
OS: macOS Ventura 13.1

wsnyder, 2023-01-13:
Thanks for reporting this, makes sense. Perhaps you could submit a pull with the test case and a fix (probably just disable message for new)?

Zvord, 2023-01-13:
I'll see what I can do. I'm new to Verilator though, so it's not gonna be fast :)

closed 2023-01-17 by wsnyder