verilator/verilator #3872: Internal Error: V3DfgPeephole.cpp:1403: Mismatched width

YikeZhou, 2023-01-14:
Hi all, I came across an "internal error" related to the power (`**`) operator.
(Found a similar issue here #3714. But I guess they're not the same.)

### Example code

```verilog
module pow_of_2(x, o);
  input [1:0] x;
  output [3:0] o;

  assign o = 4'd2 ** x;
endmodule
```

Verilator cmdline:
```
./bin/verilator -cc pow_of_2.v
```

Verilator output:
```
%Error: Internal Error: pow_of_2.v:5:19: ../V3DfgPeephole.cpp:1403: Mismatched width
    5 |   assign o = 4'd2 ** x;
      |                   ^~
                        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```

### Verilator version

The latest git master (`e5eb7d893018fadaccaaa2be14068e2ddeb1c6bd`)

### OS version

Ubuntu 22.04.1 LTS

gezalore, 2023-01-14:
Thanks for the neat test case, I will take a look.

gezalore, 2023-01-22:
Now fixed in master.

commit 3069860f (2023-01-22): Allow mismatched widths in operands of shifts in DFG

Fixes #3872.

Testing this is a bit tricky, as the front-end fixes up the operand
widths in shifts to match, and we need V3Const to introduce a mismatched
one by reducing `4'd2 ** x` (with x being 2 2-bit wide signal) to `4'd1
<< x`, but t_dfg_peephole runs with V3Const disabled exactly because it
makes it hard to write tests. Rather than fixing this one case in
V3Const (which we should do systematically at some point), I fixed DFG
to accept these just in case V3Const generates more of them. The
assertions were there only because of paranoi [... cut]
closed 2023-01-22 by gezalore