verilator/verilator #3873: Cannot find file containing module on windows

waruqi, 2023-01-14:
Thanks for taking the time to report this.

I'm getting xmake to support verilator, https://github.com/xmake-io/xmake/issues/3257

I've got it working successfully on macos/linux, but I'm having some problems on windows, but I don't know what's causing the problem.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

main.v

```
module hello;
  initial begin
    $display("hello world!");
    $finish ;
  end
endmodule
```

```console
verilator --cc --make cmake --prefix test --Mdir C:\Users\ruki\AppData\Local\Temp\.xmake\230114\_A1905FCCCD594F708649E5D19828E0C0.dir C:\Users\ruki\AppData\Local\Temp\.xmake\230114\_A1905FCCCD594F708649E5D19828E0C0.dir\main.v

%Error: Cannot find file containing module: C:\Users\ruki\AppData\Local\Temp\.xmake\230114\_B16190110D014270819F09DD1F59C8C0.dir\main.v
%Error: This may be because there's no search path specified with -I<dir>.
        ... Looked in:
             C:\Users\ruki\AppData\Local\Temp\.xmake\230114\_B16190110D014270819F09DD1F59C8C0.dir\main.v
             C:\Users\ruki\AppData\Local\Temp\.xmake\230114\_B16190110D014270819F09DD1F59C8C0.dir\main.v.v
             C:\Users\ruki\AppData\Local\Temp\.xmake\230114\_B16190110D014270819F09DD1F59C8C0.dir\main.v.sv
             C:\Users\ruki\AppData\Local\Temp\.xmake\230114\_B16190110D014270819F09DD1F59C8C0.dir/C:\Users\ruki\AppData\Local\Temp\.xmake\230114\_B16190110D014270819F09DD1F59C8C0.dir\main.v
             C:\Users\ruki\AppData\Local\Temp\.xmake\230114\_B16190110D014270819F09DD1F59C8C0.dir/C:\Users\ruki\AppData\Local\Temp\.xmake\230114\_B16190110D014270819F09DD1F59C8C0.dir\main.v.v
             C:\Users\ruki\AppData\Local\Temp\.xmake\230114\_B16190110D014270819F09DD1F59C8C0.dir/C:\Users\ruki\AppData\Local\Temp\.xmake\230114\_B16190110D014270819F09DD1F59C8C0.dir\main.v.sv
%Error: Exiting due to 2 error(s)

```

What 'verilator --version' are you using?  Did you try it with the git master version?

 rev v5.004-58-g5fce23e90 (mod)

What OS and distribution are you using?

Windows 7

May we assist you in trying to fix this in Verilator yourself?

sure

wsnyder, 2023-01-14:
And does that file exists?  If not, I suspect a problem in whatever you used to call Verilator.

waruqi, 2023-01-14:
> And does that file exists? If not, I suspect a problem in whatever you used to call Verilator.

exists, and if use relative path `main.v`, it works.

```console
cd C:\Users\ruki\AppData\Local\Temp\.xmake\230114\_A1905FCCCD594F708649E5D19828E0C0.dir
verilator --cc --make cmake --prefix test --Mdir C:\Users\ruki\AppData\Local\Temp\.xmake\230114\_A1905FCCCD594F708649E5D19828E0C0.dir main.v
```

wsnyder, 2023-01-17:
I suspect the backslashes are confusing Verilator.  Verilator is still fundamentally a Linux program, it wants forward slashes to separate directory paths.

wsnyder, 2023-01-17:
And note that it also heavily relies on GNU Make which has similar assumptions, so fixing this is beyond teaching Verilator the Microsoft filename parsing.

waruqi, 2023-01-18:
so, verilator is not yet supported on windows ? Only linux/macOS is supported?

How should I use it on windows? gnu make only runs in environments like msys2, but I don't want it to depend on it.

And I passed `-make cmake` to tell it to generate the cmake file instead of gnu make.

wsnyder, 2023-01-18:
Verilator builds now under cmake, but the work hasn't been done to make it really support Windows natively yet, so I would suggest sticking with WSL ro msys2.  (Or, look into contributing fixes on the issues needed to natively support Windows.)

waruqi, 2023-01-18:
> I suspect the backslashes are confusing Verilator. Verilator is still fundamentally a Linux program, it wants forward slashes to separate directory paths.

you are right, and I have used `/` to fix it. It works now on windows without msys2/gnu make.

```console
$ xmake -v
[  0%]: compiling.verilog src\main.v
C:\Users\ruki\AppData\Local\.xmake\packages\v\verilator\2023.1.10\7bc908f2aaa54ce3afb95b6d8dc08944\bin\verilator --cc --make cmake --prefix hello --Mdir build\.gens\hello\windows\x64\release\rules\verilator src/main.v
[  8%]: cache compiling.release C:\\Users\\ruki\\AppData\\Local\\.xmake\\packages\\v\\verilator\\2023.1.10\\7bc908f2aaa54ce3afb95b6d8dc08944/include/verilated.cpp
"C:\\Program Files\\Microsoft Visual Studio\\2022\\Community\\VC\\Tools\\MSVC\\14.34.31933\\bin\\HostX64\\x64\\cl.exe" -c -nologo -std:c++20 -Ibuild\.gens\hello\windows\x64\release\rules\verilator -IC:\Users\ruki\AppData\Local\.xmake\packages\v\verilator\2023.1.10\7bc908f2aaa54ce3afb95b6d8dc08944\include -IC:\Users\ruki\AppData\Local\.xmake\packages\v\verilator\2023.1.10\7bc908f2aaa54ce3afb95b6d8dc08944\include\vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_THREADS=1 -DVM_TIMING=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 /EHsc -Fobuild\.objs\hello\windows\x64\release\7cbdc69e81584c6d99444cb52d4864dc\verilated.cpp.obj C:\\Users\\ruki\\AppData\\Local\\.xmake\\packages\\v\\verilator\\2023.1.10\\7bc908f2aaa54ce3afb95b6d8dc08944/include/verilated.cpp
[  8%]: cache compiling.release C:\\Users\\ruki\\AppData\\Local\\.xmake\\packages\\v\\verilator\\2023.1.10\\7bc908f2aaa54ce3afb95b6d8dc08944/include/verilated_threads.cpp
"C:\\Program Files\\Microsoft Visual Studio\\2022\\Community\\VC\\Tools\\MSVC\\14.34.31933\\bin\\HostX64\\x64\\cl.exe" -c -nologo -std:c++20 -Ibuild\.gens\hello\windows\x64\release\rules\verilator -IC:\Users\ruki\AppData\Local\.xmake\packages\v\verilator\2023.1.10\7bc908f2aaa54ce3afb95b6d8dc08944\include -IC:\Users\ruki\AppData\Local\.xmake\packages\v\verilator\2023.1.10\7bc908f2aaa54ce3afb95b6d8dc08944\include\vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_THREADS=1 -DVM_TIMING=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 /EHsc -Fobuild\.objs\hello\windows\x64\release\7cbdc69e81584c6d99444cb52d4864dc\verilated_threads.cpp.obj C:\\Users\\ruki\\AppData\\Local\\.xmake\\packages\\v\\verilator\\2023.1.10\\7bc908f2aaa54ce3afb95b6d8dc08944/include/verilated_threads.cpp
[  8%]: cache compiling.release build\.gens\hello\windows\x64\release\rules\verilator/hello___024root__Slow.cpp
"C:\\Program Files\\Microsoft Visual Studio\\2022\\Community\\VC\\Tools\\MSVC\\14.34.31933\\bin\\HostX64\\x64\\cl.exe" -c -nologo -std:c++20 -Ibuild\.gens\hello\windows\x64\release\rules\verilator -IC:\Users\ruki\AppData\Local\.xmake\packages\v\verilator\2023.1.10\7bc908f2aaa54ce3afb95b6d8dc08944\include -IC:\Users\ruki\AppData\Local\.xmake\packages\v\verilator\2023.1.10\7bc908f2aaa54ce3afb95b6d8dc08944\include\vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_THREADS=1 -DVM_TIMING=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 /EHsc -Fobuild\.objs\hello\windows\x64\release\gens\rules\verilator\hello___024root__Slow.cpp.obj build\.gens\hello\windows\x64\release\rules\verilator/hello___024root__Slow.cpp
[ 16%]: cache compiling.release build\.gens\hello\windows\x64\release\rules\verilator/hello___024root__DepSet_hddd5c129__0__Slow.cpp
"C:\\Program Files\\Microsoft Visual Studio\\2022\\Community\\VC\\Tools\\MSVC\\14.34.31933\\bin\\HostX64\\x64\\cl.exe" -c -nologo -std:c++20 -Ibuild\.gens\hello\windows\x64\release\rules\verilator -IC:\Users\ruki\AppData\Local\.xmake\packages\v\verilator\2023.1.10\7bc908f2aaa54ce3afb95b6d8dc08944\include -IC:\Users\ruki\AppData\Local\.xmake\packages\v\verilator\2023.1.10\7bc908f2aaa54ce3afb95b6d8dc08944\include\vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_THREADS=1 -DVM_TIMING=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 /EHsc -Fobuild\.objs\hello\windows\x64\release\gens\rules\verilator\hello___024root__DepSet_hddd5c129__0__Slow.cpp.obj build\.gens\hello\windows\x64\release\rules\verilator/hello___024root__DepSet_hddd5c129__0__Slow.cpp
[ 25%]: cache compiling.release build\.gens\hello\windows\x64\release\rules\verilator/hello.cpp
"C:\\Program Files\\Microsoft Visual Studio\\2022\\Community\\VC\\Tools\\MSVC\\14.34.31933\\bin\\HostX64\\x64\\cl.exe" -c -nologo -std:c++20 -Ibuild\.gens\hello\windows\x64\release\rules\verilator -IC:\Users\ruki\AppData\Local\.xmake\packages\v\verilator\2023.1.10\7bc908f2aaa54ce3afb95b6d8dc08944\include -IC:\Users\ruki\AppData\Local\.xmake\packages\v\verilator\2023.1.10\7bc908f2aaa54ce3afb95b6d8dc08944\include\vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_THREADS=1 -DVM_TIMING=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 /EHsc -Fobuild\.objs\hello\windows\x64\release\gens\rules\verilator\hello.cpp.obj build\.gens\hello\windows\x64\release\rules\verilator/hello.cpp
[ 33%]: cache compiling.release build\.gens\hello\windows\x64\release\rules\verilator/hello___024root__DepSet_h1aa89519__0.cpp
"C:\\Program Files\\Microsoft Visual Studio\\2022\\Community\\VC\\Tools\\MSVC\\14.34.31933\\bin\\HostX64\\x64\\cl.exe" -c -nologo -std:c++20 -Ibuild\.gens\hello\windows\x64\release\rules\verilator -IC:\Users\ruki\AppData\Local\.xmake\packages\v\verilator\2023.1.10\7bc908f2aaa54ce3afb95b6d8dc08944\include -IC:\Users\ruki\AppData\Local\.xmake\packages\v\verilator\2023.1.10\7bc908f2aaa54ce3afb95b6d8dc08944\include\vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_THREADS=1 -DVM_TIMING=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 /EHsc -Fobuild\.objs\hello\windows\x64\release\gens\rules\verilator\hello___024root_ [... cut ...] .cpp.obj build\.objs\hello\windows\x64\release\gens\rules\verilator\hello___024root__DepSet_hddd5c129__0__Slow.cpp.obj build\.objs\hello\windows\x64\release\gens\rules\verilator\hello.cpp.obj build\.objs\hello\windows\x64\release\gens\rules\verilator\hello___024root__DepSet_h1aa89519__0.cpp.obj build\.objs\hello\windows\x64\release\gens\rules\verilator\hello___024root__DepSet_hddd5c129__0.cpp.obj build\.objs\hello\windows\x64\release\gens\rules\verilator\hello__Syms.cpp.obj
[100%]: build ok!
```

waruqi, 2023-01-18:
Maybe this code should be fixed.

https://github.com/verilator/verilator/blob/3fe81a3832809bdec63605f0d3caf2e733a86cb4/src/V3Os.cpp#L147

donlon, 2023-08-15:
Hi @waruqi, would you like to try this new version? (download here https://github.com/verilator/verilator/actions/runs/5866555068) It seems to work fine for me.

waruqi, 2023-08-16:
It works fine, thanks!

```
D:\projects\personal\xmake\tests\projects\embed\verilator\hello>xmake -rv
[  0%]: compiling.verilog src\main.v
verilator --cc --make cmake --prefix hello --Mdir build\.gens\hello\windows\x64\release\rules\verilator src\main.v
checking for cl.exe ... C:\Program Files\Microsoft Visual Studio\2022\Preview\VC\Tools\MSVC\14.37.32705\bin\HostX64\x64\cl.exe
checking for the c++ compiler (cxx) ... cl.exe
checking for the c++ compiler (cxx) ... cl.exe
checking for the c++ compiler (cxx) ... cl.exe
checking for the c++ compiler (cxx) ... cl.exe
checking for the c++ compiler (cxx) ... cl.exe
checking for the c++ compiler (cxx) ... cl.exe
checking for C:\Program Files\Microsoft Visual Studio\2022\Preview\VC\Tools\MSVC\14.37.32705\bin\HostX64\x64\cl.exe ... ok
checking for flags (-std:c++20) ... ok
[ 15%]: cache compiling.release src\sim_main.cpp
"C:\\Program Files\\Microsoft Visual Studio\\2022\\Preview\\VC\\Tools\\MSVC\\14.37.32705\\bin\\HostX64\\x64\\cl.exe" -c -nologo -std:c++20 -Ibuild\.gens\hello\windows\x64\release\rules\verilator -IC:\Users\wangrunqing\Downloads\vh\install\include -IC:\Users\wangrunqing\Downloads\vh\install\include\vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_THREADS=1 -DVM_TIMING=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 /EHsc -Fobuild\.objs\hello\windows\x64\release\src\sim_main.cpp.obj src\sim_main.cpp
[ 15%]: cache compiling.release C:\\Users\\wangrunqing\\Downloads\\vh\\install/include/verilated_threads.cpp
"C:\\Program Files\\Microsoft Visual Studio\\2022\\Preview\\VC\\Tools\\MSVC\\14.37.32705\\bin\\HostX64\\x64\\cl.exe" -c -nologo -std:c++20 -Ibuild\.gens\hello\windows\x64\release\rules\verilator -IC:\Users\wangrunqing\Downloads\vh\install\include -IC:\Users\wangrunqing\Downloads\vh\install\include\vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_THREADS=1 -DVM_TIMING=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 /EHsc -Fobuild\.objs\hello\windows\x64\release\5bcf64cbdc7e4d11a05caea798424157\verilated_threads.cpp.obj C:\\Users\\wangrunqing\\Downloads\\vh\\install/include/verilated_threads.cpp
[ 15%]: cache compiling.release C:\\Users\\wangrunqing\\Downloads\\vh\\install/include/verilated.cpp
"C:\\Program Files\\Microsoft Visual Studio\\2022\\Preview\\VC\\Tools\\MSVC\\14.37.32705\\bin\\HostX64\\x64\\cl.exe" -c -nologo -std:c++20 -Ibuild\.gens\hello\windows\x64\release\rules\verilator -IC:\Users\wangrunqing\Downloads\vh\install\include -IC:\Users\wangrunqing\Downloads\vh\install\include\vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_THREADS=1 -DVM_TIMING=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 /EHsc -Fobuild\.objs\hello\windows\x64\release\5bcf64cbdc7e4d11a05caea798424157\verilated.cpp.obj C:\\Users\\wangrunqing\\Downloads\\vh\\install/include/verilated.cpp
[ 15%]: cache compiling.release build\.gens\hello\windows\x64\release\rules\verilator/hello___024root__DepSet_hddd5c129__0__Slow.cpp
"C:\\Program Files\\Microsoft Visual Studio\\2022\\Preview\\VC\\Tools\\MSVC\\14.37.32705\\bin\\HostX64\\x64\\cl.exe" -c -nologo -std:c++20 -Ibuild\.gens\hello\windows\x64\release\rules\verilator -IC:\Users\wangrunqing\Downloads\vh\install\include -IC:\Users\wangrunqing\Downloads\vh\install\include\vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_THREADS=1 -DVM_TIMING=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 /EHsc -Fobuild\.objs\hello\windows\x64\release\gens\rules\verilator\hello___024root__DepSet_hddd5c129__0__Slow.cpp.obj build\.gens\hello\windows\x64\release\rules\verilator/hello___024root__DepSet_hddd5c129__0__Slow.cpp
[ 15%]: cache compiling.release build\.gens\hello\windows\x64\release\rules\verilator/hello___024root__Slow.cpp
"C:\\Program Files\\Microsoft Visual Studio\\2022\\Preview\\VC\\Tools\\MSVC\\14.37.32705\\bin\\HostX64\\x64\\cl.exe" -c -nologo -std:c++20 -Ibuild\.gens\hello\windows\x64\release\rules\verilator -IC:\Users\wangrunqing\Downloads\vh\install\include -IC:\Users\wangrunqing\Downloads\vh\install\include\vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_THREADS=1 -DVM_TIMING=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 /EHsc -Fobuild\.objs\hello\windows\x64\release\gens\rules\verilator\hello___024root__Slow.cpp.obj build\.gens\hello\windows\x64\release\rules\verilator/hello___024root__Slow.cpp
[ 15%]: cache compiling.release build\.gens\hello\windows\x64\release\rules\verilator/hello.cpp
"C:\\Program Files\\Microsoft Visual Studio\\2022\\Preview\\VC\\Tools\\MSVC\\14.37.32705\\bin\\HostX64\\x64\\cl.exe" -c -nologo -std:c++20 -Ibuild\.gens\hello\windows\x64\release\rules\verilator -IC:\Users\wangrunqing\Downloads\vh\install\include -IC:\Users\wangrunqing\Downloads\vh\install\include\vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_THREADS=1 -DVM_TIMING=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 /EHsc -Fobuild\.objs\hello\windows\x64\release\gens\rules\verilator\hello.cpp.obj build\.gens\hello\windows\x64\release\rules\verilator/hello.cpp
checking for flags (cl_sourceDependencies) ... ok
[ 23%]: cache compiling.release build\.gens\hello\windows\x64\release\rules\verilator/hello___024root__DepSet_h1aa89519__0.cpp
"C:\\Program Files\\Microsoft Visual Studio\\2022\\Preview\\VC\\Tools\\MSVC\\14.37.32705\\bin\\HostX64\\x64\\cl.exe" -c -nologo -std:c++20 -Ibuild\.gens\hello\windows\x64\release\rules\verilator -IC:\Users\wangrunqing\Downloads\vh\install\include -IC:\Users\wangrunqing\Downloads\vh\install\include\vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_THREADS=1 -DVM_TIMING=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 /EHsc -Fobuild\.objs\hello\windows\x64\release\gens\rules\verilator\hello___024root__DepSet_h1aa89519__0.cpp.obj build\.gens\hello\windows\x64\release\rul [... cut ...] d\.objs\hello\windows\x64\release\gens\rules\verilator\hello___024root__DepSet_hddd5c129__0__Slow.cpp.obj build\.objs\hello\windows\x64\release\gens\rules\verilator\hello.cpp.obj build\.objs\hello\windows\x64\release\gens\rules\verilator\hello___024root__DepSet_h1aa89519__0.cpp.obj build\.objs\hello\windows\x64\release\gens\rules\verilator\hello___024root__DepSet_hddd5c129__0.cpp.obj build\.objs\hello\windows\x64\release\gens\rules\verilator\hello__Syms.cpp.obj
[100%]: build ok, spent 5.407s
```

PR #4421 (2023-08-16): Fix Windows filename format, etc. (#3873)
Just combined some similar changes into a single PR to make things easier

Path processing functions now support back slash, and some compiler warnings on MSVC are fixed

It seems that this also fixes #3873.
closed 2023-08-16 by waruqi