verilator/verilator #3875: Internal error `Broken link in node`

goekce, 2023-01-14:
#3874 contains the test code which leads to the error. The modifications stated in the comments lead to a successful compilation:

```sv
interface intf #(A=0);  // A=1 ok
        typedef logic logic_t;

        function logic_t f();  // logic_t => logic ok
                return 0;
        endfunction

        modport if_m (
                import f
        );
endinterface

module system;
        intf #(1) intf1();  // 1 => 0 ok.
        m m0(.*);
endmodule

module m (intf.if_m intf1);
        if (intf1.f())  // if(0) ok
                $error();
```

goekce, 2023-01-15:
Note that `if` is a generate statement which expects a elaboration time constant expression. VCS and vsim do not seem to accept any external functions in the `if` statement, but Verilator does. Verilator compiles the following without errors.
```sv
module mm;
  function logic f();
    return 0;
  endfunction
endmodule

module m;
  mm mm1();
  if (mm1.f()) 
    $error("ok\n")
endmodule
```
But the bug is probably related to `typedef` and type propagation anyway.

wsnyder, 2023-01-17:
When the interface is elaborated it seems to get a confused data type.  Needs some more debug - perhaps you'd like to take a look at a pull to fix this?

goekce, 2023-01-24:
I am trying to find the cause for the segfault and have some questions.

The segfault happens during `V3Dead` pass. In debugging mode `V3Broken::brokenAll` is used to check the tree for broken links. It is recursively run on the children of any node. The segfault happens after deadifying the modules `m` and `intf` (probably because their parametered, constrained copies are used).

My hunch is `V3Dead` creates a dangling pointer during edit, but I have difficulties finding this pointer. Here is an excerpt from my debugging session:

```
Breakpoint 27, BrokenCheckVisitor::processAndIterate (this=0x7fffffffdd90, nodep=0x555556166ab0) at ../V3Broken.cpp:198                                                                                                                      
198             processEnter(nodep);                                                                                                                                                                                                         
(gdb)                                                                                                                                                                                                                                        
199             iterateChildrenConst(nodep);                                                                                                                                                                                                 
(gdb) pnt nodep                                                                                                                                                                                                                              
This=TYPETABLE 0x555556166ab0 back=0x5555561663a0 next=0x555556166c90 headtail=0x555556166c90 op1p=0x555556184f40                                                                                                                            
  Fileline = <built-in>:0:0:                                                                                                                                                                                                                 
     TYPETABLE 0x555556166ab0 <e2> {a0aa}                                                                                                                                                                                                    
                detailed  ->  BASICDTYPE 0x555556183f00 <e166> {c13ak} @dt=this@(G/w1)  logic [GENERIC] kwd=logic                                                                                                                            
                detailed  ->  BASICDTYPE 0x555556182540 <e222> {c9as} @dt=this@(G/nw1)  LOGIC_IMPLICIT [GENERIC] kwd=LOGIC_IMPLICIT                                                                                                          
                detailed  ->  BASICDTYPE 0x555556184f40 <e11> {c9au} @dt=this@(G/swu32/1)  logic [GENERIC] kwd=logic range=[31:0]                                                                                                            
                detailed  ->  BASICDTYPE 0x5555561f7460 <e77> {c28ad} @dt=this@(G/str)  string [GENERIC] kwd=string                                                                                                                          
         This=TYPETABLE 0x555556166ab0 back=0x5555561663a0 next=0x555556166c90 headtail=0x555556166c90 op1p=0x555556184f40                                                                                                                   
    1: BASICDTYPE 0x555556184f40 <e11> {c9au} @dt=this@(G/swu32/1)  logic [GENERIC] kwd=logic range=[31:0]                                                                                                                                   
    1:     This=BASICDTYPE 0x555556184f40 back=0x555556166ab0 next=0x5555561f7460 headtail=0x555556182540                                                                                                                                    
- V3FileLine.cpp:169: Get Stream[ct1+14]: interface intf #(A=0);                                                                                                                                                                             
    9 | interface intf #(A=0);                                                                                                                                                                                                               
      |                    ^                                                                                                                                                                                                                 
    1: BASICDTYPE 0x5555561f7460 <e77> {c28ad} @dt=this@(G/str)  string [GENERIC] kwd=string                                                                                                                                                 
    1:     This=BASICDTYPE 0x5555561f7460 back=0x555556184f40 next=0x5555561f4740 headtail=0                                                                                                                                                 
[... cut ...]
#30 0x00005555557902e2 in main (argc=<optimized out>, argv=<optimized out>) at ../Verilator.cpp:751
</code></pre>
</details>

After printing the details of `BASICDTYPE` at `0x*460`, the gdb proceed with the nextp `*740` and then goes to its child `*0x555556185310`. which has a child `AstRange` at address `0x1`. So this is causing the problem. `*310` is a `BASICDTYPE` of logic, so maybe `AstRange` pointer of `*310` is set to a value instead of an address.

- Does my story make sense?
- When is `headtail` set to 0?

wsnyder, 2023-01-24:
A crash in V3Dead or V3Broken often indicates an earlier transform messed something up.   A pointer of 0x1 is used in an object that was earlier deleted.  So the crash is really that earlier a node was deleted and a pointer still exists to it as you suspected.  If you print the broken node (pn node won't work) you should see an editCount you can use to find where it was changed (hopefully), see the internals.txt for more details.  Alternatively, use a gdb watchpoint to see where the 0x1 was written and backtrace.

goekce, 2023-01-24:
> When is `headtail` set to 0?

`headtail` is only set for the head and tail node on the same hierarchy. `headtail` of a head node is set to the address of the tail node and vice-versa. Middle nodes have the address 0. So nothing fishy here.

I got the last edit number of the the `BASICDTYPE` by using its address after the segfault.
```
p (struct AstNode*) 0xADDR
```
As you mentioned all the tree pointers are set to 0x1.

Breaking at the edit id leaves you in the middle of a tree delete operation. Probably the edit happens during `V3Dead` during `deadCheckMod()`. The `BASICDTYPE` `logic` is deleted during elimination of the `IFACE` `intf` which is the second dead module after `m`. After `deadCheckMod()` the type table is regenerated:

https://github.com/verilator/verilator/blob/4a8cfe367d32e7d775f3c63bf88e7f5bc226b932/src/V3Dead.cpp#L526-L530

(It is somehow also cleared in the beginning of V3Dead, which I do not understand: )
https://github.com/verilator/verilator/blob/4a8cfe367d32e7d775f3c63bf88e7f5bc226b932/src/V3Dead.cpp#L507-L510

My hunch is that during regeneration the type table picks up the wrong `BASICDTYPE` `logic`. AFAI understand the type table is regenerated using scopes.

wsnyder, 2023-01-25:
Yes, it probably got mistracked in scope creation.  Interfaces are complicated to handle correctly as they are both data types and modules.

goekce, 2023-01-25:
I did not look into scope creation yet.

While I was peeking into `nodep->typeTablep()->repairCache()` I noticed the difference between the *cache* and the actual children of the type table. `repairCache()` only creates a cache of the children which are `BASICDTYPE`s. So `repairCache()` works fine.

I checked the children of the type table again, there are four `BASICDTYPE`s and one `REFDTYPE` (`logic_t`) which was pointing to the `logic` in the interface `intf`. `V3Dead` eliminated `intf`, so `logic_t`'s type pointer is dangling. For reference I attached the node graph before `V3Dead` pass–you can use CTRL+Mouse-wheel to zoom after opening in the browser:

[image]

I believe `intf` should *not* have been removed because `REFDTYPE` `logic_t` needs the `intf`->`logic_t`->`logic` .

Does this make sense?

goekce, 2023-01-25:
> I believe `intf` should _not_ have been removed because `REFDTYPE` `logic_t` needs the `intf`->`logic_t`->`logic` .

I noticed `intf__A1` (parameterized copy of `intf`) already contains a copy of `TYPEDEF` `logic_t` and its child `BASICDTYPE` `logic`. So removing the `intf` is not the problem. `logic_t` in the type table should point to `logic` of `intf__A1` instead.

goekce, 2023-01-25:
> Still I do not understand why `V3Dead` would remove a data type which is used by the type table (by removing the whole `intf` tree).

I found the reason:
https://github.com/verilator/verilator/blob/4a8cfe367d32e7d775f3c63bf88e7f5bc226b932/src/V3LinkDot.cpp#L1562-L1566

`V3LinkDot` anticipates that existing references like `VARXREF` could point to a dead module so it postpones the removal to `V3Dead` step. So *V3Dead should cleanup* means that it should not only repair the cache but also the children.

wsnyder, 2023-01-26:
V3Dead should get a correct tree as an input, it shouldn't have to do other resolution work, so it's presumably a problem in V3LinkDot(?)

goekce, 2023-01-27:
> V3Dead should get a correct tree as an input

Thanks, this helps me a lot.

I added a parametered module to the design to compare the differences between a parameterized interface and a module. There is indeed a difference:  the type table picks up the right datatypes in case of the parameterized module.

The datatypes are added to the data type table via `WidthVisitor::iterateEditMoveDTypep(AstNode*, AstNodeDType*)` during V3Width. `iterateEditMoveDTypep` is called during `W3Width::visit()` with parameters `(nodep, nodep->subtypep() )`, e.g., (`VAR`, `REFDTYPE`), in other words, WidthVisitor iterates on the node, then *width*s the children (e.g., `logic`) and modifies the sub datatype of the node to the widthed datatype (`logic`) like this:

```
REFDTYPE 0x5555561f53f0 <e161> {c12al} @dt=0@  logic_t -> TYPEDEF 0x5555561f5090 <e37> {c10aq} @dt=0@  logic_t -> BASICDTYPE 0x5555561f4f30 <e21> {c10ak} @dt=this@(w1)  logic kwd=logic
```
Is transformed to
```
REFDTYPE 0x5555561f53f0 <e237#> {c12al} @dt=0x5555561f4f30@(w1)  logic_t refdt=0x5555561f4f30(w1) -> BASICDTYPE 0x5555561f4f30 <e21> {c10ak} @dt=this@(w1)  logic kwd=logic
```

Finally the child `REFDTYPE` is moved to the data type table, so the `VAR` does not have any children anymore. For example `VAR` `f` on the above graph is without any children due to `iterateEditMoveDTypep`.

Somehow this operation is only applied to `intf` but not to `intf__A1`. But in my second design with the additional parameterized module `V3Width` applies `iterateEditMoveDTypep` only on the parameterized module instead of the generic module.  The latter operation is correct, because the generic module is removed by `V3Deadify` later. 

I am searching for the reason why `V3Width` applies `iterateEditMoveDTypep` only on the generic interface `intf` instead of `intf__A1`.

For completeness I attach the tree image of the second design here. The parameterized module is `x__K1`.

<details>

[image]
</details>

goekce, 2023-01-27:
I noticed that `intf` is on level 4 compared to other modules like `m` and `x` which are on level 3 even `intf` is instantiated directly under `t`. Probably because `intf` is actually used by `m0`:

[image]

I don't think this difference is related to the problem though.

wsnyder, 2023-01-27:
What matters is the interface level > level of any modules using it.  It is a rank of the graph you posted.

goekce, 2023-01-28:
`V3Width` applies `iterateEditMoveDTypep`  on the generic interface `intf` instead of `intf__A1`, because the function inside the generate statement `if (intf1.f())` in module `m__Iz1` points to the function in `intf1`:

```
FUNCREF 0x5555561fef00 <e177> {c28am} @dt=0@  f -> .=intf1 FUNC 0x5555561f5520 <e33> {c12at} @dt=0@  f
```

I believe `intf1.f()` in `GENIF` should be linked to to `intf1__A1` instead of `intf` . This happens probably during `V3Param`.

For example the `IFACEREFDTYPE` in `m__Iz1` is indeed linked to `intf__A1` which is correct.

```
IFACEREFDTYPE 0x5555561fed00 <e74> {c27al} @dt=0@ if=intf mp=if_m -> IFACE 0x555556183220 <e218#> {c9al}  intf__A1 ...
```

goekce, 2023-01-28:
> This happens probably during `V3Param`.

Actually first linking happens during `V3LinkDot` before the modules are deparameterized in `V3Param`, so all variables, functions etc get linked to `intf`:

```
    1:2:1: IFACEREFDTYPE 0x5555561f7e50 <e74> {c27al} @dt=0@ if=intf mp=if_m -> IFACE 0x5555561f4980 <e158> {c9al} u1=0x5555561fca80 u4=0x1  intf  L4 [LIB] [1ps]
```

During `V3Param` `intf` is cloned, deparameterized and the result is `intf__A1`. The `intf1` cells point now to `intf__A1`.
```
- V3Param.cpp:1311:   param:                                                                                          
- V3Param.cpp:865:    De-parameterize: CELL 0x5555561f67c0 <e49> {c22am}  intf1 -> IFACE 0x5555561f4980 <e158> {c9al}  intf  L4 [LIB] [1ps]
...         
- V3Width.cpp:7194:   widthParamsEdit: CONST 0x5555561f69e0 <e47> {c22aj} @dt=0x5555561f4b90@(G/swu32/1)  ?32?sh1                                                                                                                            
- V3Param.cpp:373:    Name: intf->intf__A1->intf__A1                                                                  
- V3Param.cpp:593:         De-parameterize to new: IFACE 0x555556183220 <e218#> {c9al}  intf__A1  L4 [LIB] [1ps]                                                                                                                             
- V3Param.cpp:626:           set param VAR 0x5555561fdb30 <e15> {c9as} u4=0x41 @dt=0@  A [VSTATIC]  GPARAM = CONST 0x5555561f69e0 <e47> {c22aj} @dt=0x5555561f4b90@(G/swu32/1)  ?32?sh1
- V3Param.cpp:817:         Done with IFACE 0x555556183220 <e218#> {c9al}  intf__A1  L4 [LIB] [1ps]
- V3Param.cpp:882:         Done with CELL 0x5555561f67c0 <e49> {c22am}  intf1 -> IFACE 0x555556183220 <e218#> {c9al}  intf__A1  L4 [LIB] [1ps]
```

After that `IFACEREFDTYPE`s of `m0` must be reconnected, which I could not understand. I thought the step above would make all the `intf1` cells point to `intf__A1` :confused: :

```
- V3Width.cpp:7194:   widthParamsEdit: CELL 0x5555561f6d60 <e55> {c23ae}  m0 -> MODULE 0x5555561f77f0 <e156> {c27ai}  m  L3 [1ps]
- V3Param.cpp:750:         portIfaceRef IFACEREFDTYPE 0x5555561f7e50 <e74> {c27al} @dt=0@ if=intf mp=if_m -> IFACE 0x5555561f4980 <e158> {c9al}  intf  L4 [LIB] [1ps]
- V3Param.cpp:760:         pinIfaceRef IFACEREFDTYPE 0x5555561fb0b0 <e142> {c22am} @dt=0@ cell=intf1 if=intf -> CELL 0x5555561f67c0 <e49> {c22am}  intf1 -> IFACE 0x555556183220 <e218#> {c9al}  intf__A1  L4 [LIB] [1ps]
- V3Param.cpp:762:         IfaceRefDType needs reconnect  IFACEREFDTYPE 0x5555561fb0b0 <e142> {c22am} @dt=0@ cell=intf1 if=intf -> CELL 0x5555561f67c0 <e49> {c22am}  intf1 -> IFACE 0x555556183220 <e218#> {c9al}  intf__A1  L4 [LIB] [1ps]
- V3Param.cpp:373:    Name: m->m__Iz1->m__Iz1
- V3Param.cpp:593:         De-parameterize to new: MODULE 0x5555561fea50 <e227#> {c27ai}  m__Iz1  L3 [1ps]
- V3Param.cpp:606:         IfaceOld IFACEREFDTYPE 0x5555561f7e50 <e74> {c27al} @dt=0@ if=intf mp=if_m -> IFACE 0x5555561f4980 <e158> {c9al}  intf  L4 [LIB] [1ps]
- V3Param.cpp:607:         IfaceTo  IFACEREFDTYPE 0x5555561fb0b0 <e142> {c22am} @dt=0@ cell=intf1 if=intf -> CELL 0x5555561f67c0 <e49> {c22am}  intf1 -> IFACE 0x555556183220 <e218#> {c9al}  intf__A1  L4 [LIB] [1ps]
- V3Param.cpp:609:         IfaceClo IFACEREFDTYPE 0x5555561fed00 <e74> {c27al} @dt=0@ if=intf mp=if_m -> IFACE 0x5555561f4980 <e158> {c9al}  intf  L4 [LIB] [1ps]
- V3Param.cpp:611:         IfaceNew IFACEREFDTYPE 0x5555561fed00 <e74> {c27al} @dt=0@ if=intf mp=if_m -> IFACE 0x555556183220 <e218#> {c9al}  intf__A1  L4 [LIB] [1ps]
```

I see that `IFACEREFDTYPE`s and `CELL`s are updated. Maybe the reconnect  operation forgets to relink `intf1.f()` in `GENIF`, because it is neither a cell or somehow related to a `IFACEREFDTYPE`.

wsnyder, 2026-03-29:
Example passes on master; many interface improvements over the last several months.