verilator/verilator #3876: Verilator_coverage toggle coverage score differs from VCS

Bounti, 2023-01-16:
# Issue

Verilator_coverage reports a very different toggle coverage score compared to VCS. I made the comparison on an open source SystemVerilog based design (i.e., OpenTitan AES block), and found a toggle score of 27% for Verilator while VCS reports 55.89%. I put all the details below on how I computed this score with Verilator. After investigating, I could identify one major difference on how Verilator_coverage computes the score. Verilator_coverage groups hit count based on the line number. Packed structure are then considered as a single signal in some cases, and the computed toggle coverage score is then very different. 

# Details 

For instance, in aes.sv the tl_i is an input of type tlul_pkg::tl_h2d_t containing multiple signals:
```
  input  tlul_pkg::tl_h2d_t                         tl_i,
```

The coverage.dat properly logs all the hit count for each field of the tl_i structure. Please, note that each fields of tl_i has its own logged hit count in coverage.dat and all share the same line number.
```
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[0]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[10]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[11]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[12]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[13]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[14]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[15]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[16]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[17]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[18]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[19]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[1]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[20]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[21]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[22]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[23]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[24]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[25]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[26]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[27]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[28]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[29]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[2]hTOP.aes_tb.dut' 32
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[30]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[31]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[3]hTOP.aes_tb.dut' 28
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[4]hTOP.aes_tb.dut' 29
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[5]hTOP.aes_tb.dut' 29
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[6]hTOP.aes_tb.dut' 29
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[7]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[8]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_address[9]hTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[0]hTOP.aes_tb.dut' 24
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[10]hTOP.aes_tb.dut' 8
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[11]hTOP.aes_tb.dut' 12
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[12]hTOP.aes_tb.dut' 20
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[13]hTOP.aes_tb.dut' 12
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[14]hTOP.aes_tb.dut' 2
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[15]hTOP.aes_tb.dut' 10
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[16]hTOP.aes_tb.dut' 16
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[17]hTOP.aes_tb.dut' 2
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[18]hTOP.aes_tb.dut' 14
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[19]hTOP.aes_tb.dut' 14
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[1]hTOP.aes_tb.dut' 14
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[20]hTOP.aes_tb.dut' 6
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[21]hTOP.aes_tb.dut' 8
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[22]hTOP.aes_tb.dut' 2
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[23]hTOP.aes_tb.dut' 14
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[24]hTOP.aes_tb.dut' 12
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[25]hTOP.aes_tb.dut' 12
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[26]hTOP.aes_tb.dut' 12
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[27]hTOP.aes_tb.dut' 12
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[28]hTOP.aes_tb.dut' 12
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[29]hTOP.aes_tb.dut' 8
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[2]hTOP.aes_tb.dut' 18
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl49n53pagev_toggle/aesotl_i.a_data[30]hTOP.aes_tb.dut' 12
[... cut ...]
git clone https://github.com/googleinterns/hw-fuzzing.git
git -C ./hw-fuzzing apply ../diff.patch
cd hw-fuzzing
export HW_FUZZING=$(pwd) && export PYTHONPATH="./:../:$(pwd)/infra/hwfp:$(pwd)/infra/base-sim/hwfutils"
python3 ./infra/hwfp/hwfp/fuzz.py hw/opentitan/aes/cpp_afl.hjson

compile
./bin/aes < ./seeds/auto_cbc_128bit_decrypt_1block.hwf
cat logs/coverate.dat
verilator_coverage --annotate . ./logs/coverage.dat 
```

Reported toggle score is 27%, while VCS reports 55.89% score.
```
Total coverage (191/702) 27.00%
See lines with '%00' in .
```

Bounti, 2023-01-17:
Hi @wsnyder,
Thank you for your quick reply, and for confirming the issue.
I would be happy to contribute by submitting a PR.
I will look at the changes you suggested.

wsnyder, 2023-02-07:
@Bounti have you looked at this?  If not I'll look at fixing it.

wsnyder, 2023-02-07:
If you'd like to debug it, I certainly would take the patch and hold off, but otherwise I don't spend any effort on xemacs, sorry.

May I ask why you are still using xemacs?  It hasn't been updated in 14 years, I was planning on removing support for it.

Bounti, 2023-02-08:
Hi @wsnyder,

Sorry, I haven't had time to work on a fix yet. However, I made a small python script to compute toggle coverage score properly. I found a relatively closed results but still a gap of 2%: 53.88% for Verilator and 55.89% for VCS. So I extended my script to diff VCS and Verilator logs, and it turns out they have a different set of signals even if it is exactly the same code.

One example of it:
In aes.sv, the input signals of the packed structure alert_rx_i are not traced by vcs but vcs logs the usage later on. While Verilator logs both.

```
Verilator trace:
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl53n53pagev_toggle/aesoalert_rx_i[0].ack_nhTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl53n53pagev_toggle/aesoalert_rx_i[0].ack_phTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl53n53pagev_toggle/aesoalert_rx_i[0].ping_nhTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl53n53pagev_toggle/aesoalert_rx_i[0].ping_phTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl53n53pagev_toggle/aesoalert_rx_i[1].ack_nhTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl53n53pagev_toggle/aesoalert_rx_i[1].ack_phTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl53n53pagev_toggle/aesoalert_rx_i[1].ping_nhTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_ip_aes_0.6/rtl/aes.svl53n53pagev_toggle/aesoalert_rx_i[1].ping_phTOP.aes_tb.dut' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_prim_all_0.1/rtl/prim_alert_sender.svl45n21pagev_toggle/prim_alert_senderoalert_rx_i.ack_nhTOP.aes_tb.dut.gen_alert_tx[*].u_alert_sender_i' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_prim_all_0.1/rtl/prim_alert_sender.svl45n21pagev_toggle/prim_alert_senderoalert_rx_i.ack_phTOP.aes_tb.dut.gen_alert_tx[*].u_alert_sender_i' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_prim_all_0.1/rtl/prim_alert_sender.svl45n21pagev_toggle/prim_alert_senderoalert_rx_i.ping_nhTOP.aes_tb.dut.gen_alert_tx[*].u_alert_sender_i' 0
C 'fbuild/lowrisc_ip_aes_0.6/src/lowrisc_prim_all_0.1/rtl/prim_alert_sender.svl45n21pagev_toggle/prim_alert_senderoalert_rx_i.ping_phTOP.aes_tb.dut.gen_alert_tx[*].u_alert_sender_i' 0
Verilator reports 12 signals
VCS trace:
/home/dev/src/fuzzers/opentitan-fuzzer-vcs/build/lowrisc_ip_aes_0.6/syn-vcs/../src/lowrisc_prim_all_0.1/rtl/prim_alert_sender.sv
tb_aes.dut.gen_alert_tx[1].u_alert_sender_i:
alert_rx_i.ack_n: covered 0/2
alert_rx_i.ack_p: covered 0/2
alert_rx_i.ping_n: covered 0/2
alert_rx_i.ping_p: covered 0/2
tb_aes.dut.gen_alert_tx[0].u_alert_sender_i:
alert_rx_i.ack_n: covered 0/2
alert_rx_i.ack_p: covered 0/2
alert_rx_i.ping_n: covered 0/2
alert_rx_i.ping_p: covered 0/2
VCS reports only 8 signals, the inputs in aes.sv are ignored.
```

I am not sure if any standard way exists to compute the toggle coverage, but yet VCS and Verilator score for toggle are not comparable directly.

@wsnyder I did not get your question on xemacs? I am not using xemacs at all.
I will work on a fix this Friday, and share the PR. Sorry for the delay.

wsnyder, 2023-02-08:
I'm not aware of any standard in what signals should be covered/combined for coverage.
Ignore the xemacs comment, I posted to wrong thread.

closed 2023-02-09 by wsnyder