verilator/verilator #3879: SVA: named property not working in latest/devel version?

svenka3, 2023-01-17:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

> Trying sv-tests:  tests/chapter-16/16.15--property-disable-iff.sv

```log
%Error: /home/srini/proj/git_sv-tests/sv-tests/tests/chapter-16/16.15--property-disable-iff.sv:34:27: Unsupported in C: Variable has the same name as PROPERTY 'prop': 'prop'
```

I did a bit more digging as to how the runners.py seems to work fine on this example, and below is what I found. I could be wrong, am new to this repo.

I see that the top module is passed as "clk_gen" (by its default module search algorithm).
> --top-module clk_gen

So my wild guess is that it hides that named property issue in module top. I hacked it a bit to have module top as first in the file, then I see this issue.

What 'verilator --version' are you using?  Did you try it with the git master version?

> Verilator 5.005 devel rev v5.004-61-ga1160a85a

What OS and distribution are you using?
> Windows 10, WSL, Ubuntu

May we assist you in trying to fix this in Verilator yourself?

wsnyder, 2023-01-17:
Can you please submit the fixed test (to fail) to sv-tests?

Anyhow named properties are relatively new. Perhaps you could look at fixing this, or @RRozak perhaps you could?

PeterMonsson, 2023-01-19:
@svenka3 For step 1 you can update test_regress/t/t_assert_named_property.v with the following code snippet (based on sv-tests):

```
    logic out = 1;
   
    property prop_a;
        @(posedge clk) disable iff (cyc <= 10) out;
    endproperty

    assert property (prop_a) else $error($sformatf("property check failed :assert: (False)"));
```

You can run test_regress/t/t_assert_named_property.v with the following command:

```
cd test_regress
perl driver.pl t/t_assert_named_property.pl
```

For a workaround in your own code, you can add a dummy argument to all your named properties like this:

````
    property prop_a(dummy);
        @(posedge clk) disable iff (cyc <= 10) out;
    endproperty

    assert property (prop_a(10)) else $error($sformatf("property check failed :assert: (False)"));

```

For step 2 (trying to fix it) I would look at src/verilog.y and see if there is a reason why assert property takes a  method call but not a property ID.

RRozak, 2023-01-20:
You can even just add `()` to a property call, like this:
`assert property (prop_a()) else ...`
Assert property takes a function call instead of a property ID, because it is hard to distinguish them on a parser stage.
Here is an example:
```systemverilog
   function logic func;      
     return 1'b1;
   endfunction

   property prop;
      1'b1;
   endproperty
   assert property(@(posedge clk) prop());
   assert property(@(posedge clk) func());
```
Another reason is that `AstProperty` inherits from `AstNodeFTask`, because the linking process is almost the same, except the case from this issue.

RRozak, 2023-01-20:
I prepared a fix for it.

PeterMonsson, 2023-01-20:
Wow! Thank you!

svenka3, 2023-01-20:
Thanks a lot for the quick turnaround, appreciate it.