verilator/verilator #3880: SVA: Assertion failures not reported

svenka3, 2023-01-17:
Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

I don't see any of the SVA-s below firing in Verilator. What am I missing please?

```sh
verilator --timing --binary sva_imm.sv
```

```verilog
module top;

  bit pclk;
  bit pselx;

  always #10 pclk = ~pclk;

  initial begin : test
    #100;
    pselx = 1;
    #100;
    pselx = 0;
    #100;
    pselx = 1;
    #100;
    $finish (2);
  end : test
  always @(pselx) begin
    $display ("%t psel: %b", $time, pselx);
    a1 : assert (!pselx) else $error ("a1");
    a2 : assert (pselx)  else $error ("a2");
  end
  a3 : assert property (@(posedge pclk) pselx)  else $error ("a3");

endmodule : top

```

What 'verilator --version' are you using?  Did you try it with the git master version?

> Verilator 5.005 devel rev v5.004-61-ga1160a85a

What OS and distribution are you using?

> Windows 10, WSL

May we assist you in trying to fix this in Verilator yourself?

wsnyder, 2023-01-17:
Use --assert

closed 2023-01-17 by wsnyder

svenka3, 2023-01-17:
That worked, thanks!