verilator/verilator #3888: no warning/error about multi-driven signal

adambagley, 2023-01-19:
Hi, I noticed with Verilator 5.002 if I accidentally multi-drive a signal in my RTL with both an `assign` statement and `always_comb` block, there's no compiler warning/error. Example: 
`logic foo, bar;`
`assign foo = 1'b0;`
`always_comb foo = bar;`
Thank you

wsnyder, 2023-01-19:
Specifically IEEE 9.2.2.2 says anything driven by always_comb should have no other drivers.

Perhaps you'd be willing to make a pull request to test and add this check?

adambagley, 2023-01-19:
Thanks for finding the part of the spec that mentions this. I'm going to look into adding this check today and see how involved it is, I'm guessing it's pretty straightforward

adambagley, 2023-01-21:
Good news, I was able to do some hacking of the source code to get the linting check to catch this. I'd like to do a cleaner fix and then add a test before opening a pull request. I'm hoping to have something ready sometime next week. Cheers

adambagley, 2023-01-29:
Hi, I'm a little stuck trying to figure out where/how to add  this check, and would appreciate if you could help point me in the right direction.

This example is not caught by the current linting:
`input clk;`
   `input d;`
   `output reg out;`
   
  ` assign out = 1'b0;  // <--- Warning`
  ` always_comb begin`
     ` out = d;  // <--- Warning`
   `end`
   
   But this one is: 
`input clk;`
   `input d;`
   `output reg out;`
   
  ` assign out = d;  // <--- Warning`
  ` always_comb begin`
     ` out = d;  // <--- Warning`
   `end`
   
   The reason being, in the first example the first assignment to `out` is constant so is 'synthesized away' due to this line in Verilator.cpp:
   `if (v3Global.opt.fConstBeforeDfg()) V3Const::constifyAllLint(v3Global.rootp());`
   
   When I comment out this line in Verilator.cpp, the linting throws warnings on both examples. However, I'm aware this isn't an actual solution. As this is my first time looking at Verilator source code, I'm not quite sure where's the best spot to add this specific check: that a variable written to in an `always_comb` shouldn't be written to by another process. 
   
   The check would have to occur _before_ any constant assignments are 'synthesized away'.  
   
  I'm not sure if V3DfgAstToDfg.cpp is the best spot,  or V3Delayed.cpp, or other? I've been trying to use the commit history and existing tests as hints of where to add the check but haven't found a great spot yet.
   
   Thank you,
   Adam

wsnyder, 2023-01-30:
I believe some constification is done before V3Undriven so that constant array indices can be properly analyzed.  V3const was not supposed to impact V3Undriven (see comments in Verilator.cpp).  But I can see how it is in this case.

I suspect this could be fixed by V3Undriven looking for the init()'s that V3Const has converted the otherwise missing expression into.

adambagley, 2023-01-30:
So I don't believe V3Undriven is being impacted by V3Const. Perhaps we're confusing `V3Const::constifyAllLive()` with `V3Const::constifyAllLint()` ?

The function calls happen in this order:
`V3Const::constifyAllLive();`
`V3Undriven::undrivenAll();`
`V3Const::constifyAllLint();`

When I comment out the _last_ of these, the existing linting check catches the multidriven signals.
 I.e. the signals aren't being constified until _after_ V3Undriven, and this works as a hack:
`V3Const::constifyAllLive();`
`V3Undriven::undrivenAll();`
`//V3Const::constifyAllLint();`

So I believe I can add the check to V3Undriven without having to worry about constifyAll**Live** having impacted/converted anything relevant for this check?

wsnyder, 2023-02-01:
Yes, the intent is constifyAllLive should not impact V3Undriven.

closed 2023-02-23 by wsnyder