verilator/verilator #3891: Treatment of VL_CPU_RELAX on unknown architectures

ldoolitt, 2023-01-19:
A portion of `include/verilatedos.h` attempts to define useful (and architecture-dependent) values of the VL_CPU_RELAX macro.
But the default case, for unknown architectures, is to error.

The documented reason for this macro is efficiency on SMT CPUs.  Is that reason to block building on an unknown architecture?
This condition was hit for Verilator 5.004 on Debian auto-builders for mipsel, mips64el, and s390x.

https://buildd.debian.org/status/package.php?p=verilator

ldoolitt, 2023-01-19:
The VL_CPU_RELAX macro seems to be the cause of failure on armel, too.
The build phase succeeds, but the test phase hits

```
/tmp/cchHO0hs.s: Assembler messages:
/tmp/cchHO0hs.s:1269: Error: selected processor does not support `yield' in ARM mode
make[2]: *** [/<<PKGBUILDDIR>>/test_regress/../include/verilated.mk:233: verilated.o] Error 1
```

wsnyder, 2023-01-19:
It's power hungry to not have a VL_CPU_RELAX.  I would prefer we have an error message for people compiling Verilator using github sources on a new architectures so they know they need to define it. 

I see two paths:
1. If we can figure out the architectures of interest and the appropriate defines I can ifdef check, I'll add appropriate ifdefs.
2. For a distro that is highly cross-platform, I'm fine if we have a new define that just makes the macro empty for unknown architectures.  Especially as some architectures don't even implement the appropriate instruction.

Your preference?

Note in either option armel will need a separate ifdef to work around what seems a bug in their yield.

ldoolitt, 2023-01-19:
I don't mind an out-of-the-box default build-from-git experience that mandates a working VL_CPU_RELAX.
I like the idea of a config-level option turning unknown-architecture VL_CPU_RELAX into a no-op, so Debian can deliver working (as in, passes "make test") binaries for mipsel, mips64el, and s390x.  And probably even hppa, powerpc, and sparc64.
I'm not good at naming things, so I won't propose a name for this option.  :-(

wsnyder, 2023-01-19:
I found source code for s930, mips, armel, sparc, and implemented them.  I don't have system to test on but I think this should clear up the rows in white in the report you pointed to.  I added VL_IGNORE_UNKNOWN_ARCH for anything else.

Let me know once you think all the distro issues are good I'll push a new stable release (5.006).

closed 2023-01-19 by wsnyder

tijuca, 2023-01-19:
Thanks for looking into that quickly, I'll try some testing builds tomorrow on porter boxes.

tijuca, 2023-01-20:
So, a first rebuild on s390x was successful.

Building on armel works also mostly, running `make test` on armel is still failing. I can workaround this by excluding calling that target for armel. OTOH there is a reason why this is failing. :)

```
...
g++ -g -O2 -ffile-prefix-map=/home/tijuca/verilator=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -I.  -MMD -I/home/tijuca/verilator/test_regress/../include -I
/home/tijuca/verilator/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-
uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a1_first_cc -DVM_PREFIX=Vt_a1_first_cc -DVM_PREFI
X_INCLUDE="<Vt_a1_first_cc.h>" -DVM_PREFIX_ROOT_INCLUDE="<Vt_a1_first_cc___024root.h>" -DT_A1_FIRST_CC   -O0 -c -o verilated_threads.o /home/tijuca/verilator/test_regress/../include/verilated_threads.cpp
/usr/bin/perl /home/tijuca/verilator/test_regress/../bin/verilator_includer -DVL_INCLUDE_OPT=include Vt_a1_first_cc.cpp Vt_a1_first_cc___024root__DepSet_hc6dd0e50__0.cpp Vt_a1_first_cc___024root__DepSet_h4
a66bce2__0.cpp Vt_a1_first_cc__Trace__0.cpp Vt_a1_first_cc___024root__Slow.cpp Vt_a1_first_cc___024root__DepSet_h4a66bce2__0__Slow.cpp Vt_a1_first_cc__Syms.cpp Vt_a1_first_cc__Trace__0__Slow.cpp > Vt_a1_fi
rst_cc__ALL.cpp
g++ -g -O2 -ffile-prefix-map=/home/tijuca/verilator=. -fstack-protector-strong -Wformat -Werror=format-security -Wdate-time -D_FORTIFY_SOURCE=2 -I.  -MMD -I/home/tijuca/verilator/test_regress/../include -I
/home/tijuca/verilator/test_regress/../include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=1 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-
uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -DVERILATOR=1 -DVL_DEBUG=1 -DTEST_OBJ_DIR=obj_vlt/t_a1_first_cc -DVM_PREFIX=Vt_a1_first_cc -DVM_PREFI
X_INCLUDE="<Vt_a1_first_cc.h>" -DVM_PREFIX_ROOT_INCLUDE="<Vt_a1_first_cc___024root.h>" -DT_A1_FIRST_CC   -O0 -c -o Vt_a1_first_cc__ALL.o Vt_a1_first_cc__ALL.cpp
echo "" > Vt_a1_first_cc__ALL.verilator_deplist.tmp
Archive ar -rcs Vt_a1_first_cc__ALL.a Vt_a1_first_cc__ALL.o
g++ -Wl,-z,relro -Wl,-z,now    Vt_a1_first_cc__main.o verilated.o verilated_vcd_c.o verilated_threads.o Vt_a1_first_cc__ALL.a    -pthread -lpthread -latomic   -o Vt_a1_first_cc
rm Vt_a1_first_cc__ALL.verilator_deplist.tmp
make[2]: Leaving directory '/home/tijuca/verilator/test_regress/obj_vlt/t_a1_first_cc'
driver: Leaving directory '/home/tijuca/verilator/test_regress/obj_vlt/t_a1_first_cc'
        obj_vlt/t_a1_first_cc/Vt_a1_first_cc    > obj_vlt/t_a1_first_cc/vlt_sim.log
Vt_a1_first_cc: ./test_regress/../include/verilated_imp.h:97: VerilatedEvalMsgQueue::VerilatedEvalMsgQueue(): Assertion `atomic_is_lock_free(&m_depth)' failed.
%Warning: vlt/t_a1_first_cc: Exec of obj_vlt/t_a1_first_cc/Vt_a1_first_cc failed: Vt_a1_first_cc: ./test_regress/../include/verilated_imp.h:97: VerilatedEvalMsgQueue::VerilatedEvalMsgQueue(): Assertion `at
omic_is_lock_free(&m_depth)' failed.

vlt/t_a1_first_cc: %Error: Exec of obj_vlt/t_a1_first_cc/Vt_a1_first_cc failed: Vt_a1_first_cc: ./test_regress/../include/verilated_imp.h:97: VerilatedEvalMsgQueue::VerilatedEvalMsgQueue(): Assertion `atom
ic_is_lock_free(&m_depth)' failed.
vlt/t_a1_first_cc: FAILED: Exec of obj_vlt/t_a1_first_cc/Vt_a1_first_cc failed: Vt_a1_first_cc: ./test_regress/../include/verilated_imp.h:97: VerilatedEvalMsgQueue::VerilatedEvalMsgQueue(): Assertion `atom
ic_is_lock_free(&m_depth)' failed.
==SUMMARY: Passed 0  Failed 1  Time 1:15
==SUMMARY: Passed 0  Failed 1  Time 1:15

======================================================================
        #vlt/t_a1_first_cc: %Error: Exec of obj_vlt/t_a1_first_cc/Vt_a1_first_cc failed: Vt_a1_first_cc: ./test_regress/../include/verilated_imp.h:97: VerilatedEvalMsgQueue::VerilatedEvalMsgQueue(): Assert
ion `atomic_is_lock_free(&m_depth)' failed.
                make && test_regress/t/t_a1_first_cc.pl  --vlt
TESTS DONE, FAILED: Passed 0  Failed 1  Time 1:15
make[1]: *** [Makefile:171: smoke-test] Error 10
```

Builds for mips{,64}el are still running, will add some potential fallout here if somethings still goes wrong there.

wsnyder, 2023-01-20:
I'd suggest blacklist armel, it looks like something is broken in its atomic library.

tijuca, 2023-01-22:
The build on mipsel and mips64el were also failing with the same error message. I turned off the call for the target `make test` on these platforms and the build process on these machines is now also successful.

I know that the architectures armel, mips{,64}el are not that important for serious daily work but the failing targets showing there is something not working correctly and ignoring this is not solving the real issues. I know of some upstream projects that have written their own platform specific wrapper or sub component to shipping around such problems.
Just want to mention this, I have no strong feelings on this, the final decision is always up to the upstream maintainers.