verilator/verilator #3896: Visual Studio support

sanggyu78, 2023-01-21:
Hi, 

I am developing Visual Studio Extensions for SystemVerilog. 
Although there are many IDEs available now, including Eclipse, Visual Studio Code, etc, 
I think Visual Studio is the most convenient and advanced IDE, which lacks SystemVerilog support. 

It was natural for me to try to support Verilator as a primary simulation tool. 
Following Video demonstrates Verilator works very well in my Visual Studio Extension. 
https://youtu.be/Jqf2EyqFHtI

To use Verilator in Visual Studio, the error messages should be formatted to be compatible with Visual Studio.
The other IDEs, such as Visual Studio Code, support modifying regular expressions to parse error messages.
Unfortunately, however, Visual Studio does not support such. 

I modified Verilator to generate Visual Studio-compatible error messages just by adding one function.

In V3Error.cpp, I added the function report_visual_studio_error(). 
This function translates the original message into Visual-Studio-compatible format. 

And then, I modified V3Error::v3errorEnd() as follows. 

```
    if (
#ifndef V3ERROR_NO_GLOBAL_
        !(v3Global.opt.quietExit() && s_errorCode == V3ErrorCode::EC_FATALEXIT)
#else
        true
#endif
    ) {
		if(__adhoc_extension_vs_support_) 
			report_visual_studio_error(msg);
		else
                        std::cerr << msg;
    }
```

I think it will be the best if some command line option, for example  -message_format_vs, to turn-on message format translation.

```
void report_visual_studio_error(string msg)
{
	// Convert Verilator's regular error message into Visual studio's standard message
	// Verilator:     %Warning-UNOPTFLAT: MCU/RV32_Xhpla_ctrl.sv:107:15: Signal unoptimizable: Circular combinational logic: 'RV32E_Xhpla_core.u_ctrl.exe_ready_w
	// Visual studio: C:\\Workspace\\HPA\\HPLA\\MCU\\RV32_Xhpla_ctrl.sv(107,15): error UNOPTFLAT: Signal unoptimizable: Circular combinational logic: 'RV32E_Xhpla_core.u_ctrl.exe_ready_w
	
	const char* c_msg = msg.c_str();
	char* p_msg = new char[strlen(c_msg)+1];
	strcpy(p_msg, c_msg);
	char* ptr_head = p_msg;

	const char* category = NULL;
	const char* code = NULL;
	const char* filename = NULL;
	const char* message = NULL;

	if(!strncmp(ptr_head, "%Warning-", 9))
	{
		category = ptr_head;
		ptr_head += 9;
	}
	else if(!strncmp(ptr_head, "%Error-", 7))
	{
		category = ptr_head;
		ptr_head += 7;
	}
	else if(!strncmp(ptr_head, "%Warning:", 9))
	{
		category = ptr_head;
		ptr_head += 9;
		code = "";
	}
	else if(!strncmp(ptr_head, "%Error:", 7))
	{
		category = ptr_head;
		ptr_head += 7;
		code = "";
	}

	if(code == NULL)
	{
		char* p = strchr(ptr_head, ':');
		if(p != NULL)
		{
			*p = 0;
			code = ptr_head;
			ptr_head = p + 1;
		}
		else
		{
			code = "";
		}
	}
	
	char* p = strchr(ptr_head, ':');
	if(p != NULL)
	{
		*p = 0;
		if(*ptr_head == ' ') ptr_head++;
		filename = ptr_head;

		char* q = ptr_head;
		while(*q)
		{
			if(*q == '/')
				*q = '\\';
			q++;
		}

		ptr_head = p + 1;
	}

	message = ptr_head;
	if(*message == ' ')
		message++;

	if(filename != NULL && code != NULL && message != NULL)
		std::cerr << filename << ": error " << code << ": " << message;
	else
		std::cerr << msg;

	delete p_msg;
}

```

wsnyder, 2023-01-21:
That's a good video!

I believe Visual Studio does allow custom regexps, please see https://learn.microsoft.com/en-us/visualstudio/msbuild/msbuild-diagnostic-format-for-tasks?view=vs-2022 and let us know if that does what you need without a change, thanks.

sanggyu78, 2023-01-28:
Hi Wilson, 

Really thanks for giving good advice on my issue. 

I have tried to add CustomErrorRegularExpression for MSBuild as guided. 

I added the following custom regex: 
CustomErrorRegularExpression="^\s*%(?&lt;CATEGORY&gt;Error):\s*(?&lt;FILENAME&gt;[^:]+):(?&lt;LINE&gt;\d*):(?&lt;COLUMN&gt;\d*):\s*(?&lt;CODE&gt;[^:]+),\s*(?&lt;TEXT&gt;.+)$"

It works to detect errors from the original Verilator messages and lists them up in the Error Window Pane. 
Unfortunately, however, it seems that jumping to the location of the error is not supported well. 

As a temporary solution, 
I will develop a utility program that translates Verilator's messages to the Visual Studio's standard message format. 

[image]

Best Regards
Sanggyu Park

closed 2023-01-28 by wsnyder