verilator/verilator #3897: Null pointer check in a wrong place in array locator method

Zvord, 2023-01-22:
Consider the following code (written in the fashion of `t_with.v`)
```systemverilog
`define stop $stop
`define checkh(gotv,expv) do if ((gotv) !== (expv)) begin $write("%%Error: %s:%0d:  got='h%x exp='h%x\n", `__FILE__,`__LINE__, (gotv), (expv)); `stop; end while(0);
class Cls;
   int i;
   function int get_i();
      return i;
   endfunction
endclass

module t (/*AUTOARG*/);
   initial begin
      Cls aliases[$];
      Cls found[$];
      int tofind;
      for (int j = 1; j <= 8; j++) begin
         Cls obj = new;
         obj.i = j;
         aliases.push_back(obj);
      end
      tofind = 4;
      found = aliases.find() with (item.i == tofind);
      `checkh(found.size, 1);
      found = aliases.find() with (item.get_i() == tofind);
      `checkh(found.size, 1);
      $write("*-* All Finished *-*\n");
      $finish;
   end
endmodule
```

Compilation fails with
```
./Vt_with___024root__DepSet_ha94cac4d__0__Slow.cpp:67:19: error: use of undeclared identifier 'item'
    VL_NULL_CHECK(item, "t/t_with.v", 34)->__VnoInFunc_get_i(vlSymsp, __Vtask_get_i__1__Vfuncout);
```

This is because the null check is hanging in the air:
```c++
    VL_NULL_CHECK(item, "t/t_with.v", 34)->__VnoInFunc_get_i(vlSymsp, __Vtask_get_i__1__Vfuncout);
    t__DOT__unnamedblk1__DOT__found = vlSelf->t__DOT__unnamedblk1__DOT__aliases.find(
                                                                                [=](
                                                                                IData/*31:0*/ item__DOT__index,
                                                                                VlClassRef<Vt_with___024unit__03a__03aCls> item) { return 
                                                                                (4U 
                                                                                == __Vtask_get_i__1__Vfuncout); }
    );
```

Note, that only `find() with (item.get_i()...` fails. The null check is placed inside the lambda when we access the object's property.
```c++
    t__DOT__unnamedblk1__DOT__found = vlSelf->t__DOT__unnamedblk1__DOT__aliases.find(
                                                                                [=](
                                                                                IData/*31:0*/ item__DOT__index,
                                                                                VlClassRef<Vt_with___024unit__03a__03aCls> item) { return 
                                                                                (4U 
                                                                                == VL_NULL_CHECK(
                                                                                item, "t/t_with.v", 32)
                                                                                ->__PVT__i); }
    );
```

wsnyder, 2023-01-23:
Presumably the insertion into the Ast tree needs to be relocated.  Might you be able to look at the sources to see where it should go and attempt a pull request to fix it (and add your test)?

Zvord, 2023-01-23:
Sure, I will take a look.

wsnyder, 2023-01-25:
Yes, the call should land inside the WITH, with the ASSIGN that is formed also inside the WITH.  Likely V3Task isn't tracking the insertion point for With blocks.

wsnyder, 2025-03-09:
The test here now passes on master, presuming fixed earlier.

closed 2025-03-09 by wsnyder