verilator/verilator #3901: Constrain Randomization using Verilator 5.005

raulbehl, 2023-01-24:
Does verilator support constraint randomization? From the recent blog post at https://antmicro.com/blog/2023/01/open-source-systemverilog-uvm-support-in-verilator/ it seems that the crave library is integrated and should work. But I still see that verilator doesn't consider constraints during randomization. I tried testing it using the following example:

```systemverilog
class foo;

  rand bit [1:0]  num;

  constraint gen_num {
    num == 2'b00;
  }

endclass
    
// Construct the class in the TB
module foo_tb ();

  foo foo_h;

  initial begin
    foo_h = new();
    repeat (8) begin
      void'(foo_h.randomize());
      $display("%2b\n", foo_h.num);
    end

    $finish();
  end

endmodule
```

Used the following command to compile: `verilator --binary --exe --timing foo.sv`

wsnyder, 2023-01-24:
I haven't seen a pull request yet so there is no constraint solver. @kbieganski I presume an "experimental" is needed in that sentence in the blog?  I think the other features mentioned in that blog are all in 5.006.

tgorochowik, 2023-01-24:
Hi @raulbehl 

I am sorry for making it confusing in the blog note, we will change it to more explicitly what is already mainlined and what is not - for now the features that are merged have a PR linked in the text, but that's definitely not clear enough.

The constraints/crave support can be found in a fork, it's not yet in a state where we would be happy enough with it to submit it here, if you want to play with it, please take a look at this [test suite](https://antmicro.github.io/verilator-verification-features-tests/log.html) under "randomize-constraints" to see what is supported and to what extent.

The sources for the tests used there can be found [here](https://github.com/antmicro/verilator-verification-features-tests/tree/8de426ded230383e527e0f8c298cccc754f1b919/tests/randomize-constraints) (note that not all of these pass yet in the test suite above)

Verilator with crave support can be found in [our fork](https://github.com/antmicro/verilator-1/tree/c975a61f9300a89c0589b84b63200a0d1eb84bde) for now.

We plan to submit this work once it's a little bit more mature, at this moment there is still quite a few things to figure out.

wsnyder, 2023-01-24:
Thanks @tgorochowik, as always, I appreciate the massive work behind this and look forward to the eventual pull.

closed 2023-01-24 by wsnyder

raulbehl, 2023-01-28:
Thank you for the quick feedback on this. 

@wsnyder @tgorochowik I greatly appreciate the incredible efforts put into incorporating numerous features in Verilator. Thank you for your hard work.