verilator/verilator #3902: Function call without () not working

svenka3, 2023-01-24:
While porting SVUnit to Verilator I stumbled upon a subtle issue. Function calls in SV are allowed without braces - (). Looks like Verilator does not like that style. Sample error:

```log
%Error: vw_vlt_issue_1.sv:11:42: Found definition of 'get_start_tag_contents' as a FUNC but expected a variable
   11 |     result = $sformatf("%s<%s>", indent, get_start_tag_contents);
      
```

Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

Commands:
```sh
verilator --binary vw_vlt_issue_1.sv -DBUG
verilator --binary vw_vlt_issue_1.sv (Works)
```

```verilog
class c;
  function string get_start_tag_contents;
  endfunction : get_start_tag_contents

  local function string as_string_with_indent(string indent);
    string result;
    string tmp_s;
   `ifndef BUG
    result = $sformatf("%s<%s>", indent, get_start_tag_contents());
  `else
    result = $sformatf("%s<%s>", indent, get_start_tag_contents);
  `endif // BUG
  endfunction : as_string_with_indent
endclass : c
module top;
  c c_0;

  initial begin
    c_0 = new();

    #100 $finish (2);
  end
endmodule : top

```

What 'verilator --version' are you using?  Did you try it with the git master version?
> Verilator 5.005 devel rev v5.004-75-gbe53eec5c

What OS and distribution are you using?

> WSL, Ubuntu

May we assist you in trying to fix this in Verilator yourself?

> Yes please, though my Lex/yacc expertise is read-only so far.

wsnyder, 2023-01-24:
Running --debug --dumpi-tree 9 you should be able to see what the parser creates.  Then this is fed to V3LinkDot that analyzes the symbols and determines what's a function/method call; I suspect that needs improvement.  Run with the parens to see the debug output also so you know what the LinkDot output should be. Thanks for looking into it.

RRozak, 2023-01-24:
I fixed it. I just changed the fix for properties to be more general. There is also the same problem with method calls and it would require a separate fix.

svenka3, 2023-03-31:
I was trying to remove the previous workaround on SVUnit - from where this issue was reported. I see the issue is still open. I see t_func.v is updated and it does test some part. However the below code still fails with same error:

```verilog

   module t;

  initial begin
    int a;
    a = fn_call_wo_paren;
  end

   function int fn_call_wo_paren;
     fn_call_wo_paren = 3;
   endfunction

endmodule

```
Log:
```log
%Error: /home/srini/f.v:5:9: Found definition of 'fn_call_wo_paren' as a FUNC but expected a variable
    5 |     a = fn_call_wo_paren;
      |         ^~~~~~~~~~~~~~~~
```
Thanks

wsnyder, 2023-03-31:
The original issue was resolved, so it would be better to file a new issue, but I suppose we can reopen this one instead since it does seem similar.

@svenka3 perhaps you could make a pull request improving V3LinkDot.cpp to resolve AstVarRef's as function calls?

svenka3, 2023-03-31:
I would love to attempt a PR. I did spend sometime looking at that CPP file. I also dumped the tree files, got about 10 of them, which one to start looking for as first suspect please

svenka3, 2023-04-01:
I looked at the tree files and I see that Vf_001_cells.tree itself has a difference - for a function call with/wo parenthesis.

```log
Verilator Tree Dump (format 0x3900) from <e0> to <e737>
     NETLIST 0x5555572e6000 <e1#> {a0aa}  $root [1ps/1ps]
    1: MODULE 0x5555572e6360 <e737#> {d1ai}  t  L0 [1ps]
    1:2: INITIAL 0x555557352210 <e540#> {d3ad}
    1:2:2: BEGIN 0x55555730e7e0 <e529#> {d3al} [UNNAMED]
    1:2:2:1: VAR 0x5555572f1b00 <e532#> {d4aj} @dt=0@  a VAR
    1:2:2:1:1: BASICDTYPE 0x555557350dd0 <e533#> {d4af} @dt=this@(sw32)  int kwd=int range=[31:0]
    1:2:2:1: ASSIGN 0x555557352160 <e539#> {d5ah} @dt=0@
    1:2:2:1:1: PARSEREF 0x555557350f70 <e537#> {d5aj} @dt=0@  fn_call_wo_paren [TEXT]
```
Good one (with parenthesis during call)
```log
Verilator Tree Dump (format 0x3900) from <e0> to <e740>
     NETLIST 0x5555572e6000 <e1#> {a0aa}  $root [1ps/1ps]
    1: MODULE 0x5555572e6360 <e740#> {d1ai}  t  L0 [1ps]
    1:2: INITIAL 0x555557352210 <e543#> {d3ad}
    1:2:2: BEGIN 0x55555730e7e0 <e529#> {d3al} [UNNAMED]
    1:2:2:1: VAR 0x5555572f1b00 <e532#> {d4aj} @dt=0@  a VAR
    1:2:2:1:1: BASICDTYPE 0x555557350ea0 <e533#> {d4af} @dt=this@(sw32)  int kwd=int range=[31:0]
    1:2:2:1: ASSIGN 0x555557352160 <e542#> {d6ah} @dt=0@
    1:2:2:1:1: FUNCREF 0x55555734adc0 <e540#> {d6aj} @dt=0@  fn_call_wo_paren -> UNLINKED
```
I guess that FUNCREF is the key. So is it still V3LinkDot.cpp the suspect or another file?

svenka3, 2023-04-01:
OK, with some prints etc. I realized the issue could well be within V3LinkDot.cpp as pointed out earlier.

Looking at the fix https://github.com/verilator/verilator/pull/3903/files

```cpp
src/V3LinkDot.cpp
if (!ftaskp->isFunction()) {
                    // The condition is true for tasks, properties and void functions.
                    // In these cases, the parentheses may be skipped.
```
Even for non-void functions, the parens are optional. I just now tried making that if-cond true-always and see it worked fine for few function scenarios. Should I run a regression? Curious as to why that comment and the check for !isFunction() ?

wsnyder, 2023-04-01:
@RRozak can you comment?

@svenka3 can you prepare a pull along with tests?

svenka3, 2023-04-01:
I ran a regression and results are as below:

```log

======================================================================
        #dist/t_a5_attributes_include: %Error: Missing ok
                make -j && test_regress/t/t_a5_attributes_include.pl  --dist
        #dist/t_a5_attributes_src: %Error: Missing ok
                make -j && test_regress/t/t_a5_attributes_src.pl  --dist
        #vlt/t_func_no_parentheses_bad: %Error: Exec of perl ok, but expected to fail
                make -j && test_regress/t/t_func_no_parentheses_bad.pl  --vlt
        #vltmt/t_func_no_parentheses_bad: %Error: Exec of perl ok, but expected to fail
                make -j && test_regress/t/t_func_no_parentheses_bad.pl  --vltmt
TESTS DONE, FAILED: Passed 3040  Failed 4  Failed-First 4  Skipped 293  Time 66:39
make: *** [Makefile:51: test] Error 10
```

Is there a hitchhiker's guide to regression triaging here? Should I chase those failed-first alone?

Thanks

svenka3, 2023-04-01:
I debugged t/t_func_no_parentheses_bad.pl - looks like the gold/out file needs update. I am also improving that test. Will send a PR.

svenka3, 2023-04-01:
This test was originally intended to fail (hence that _bad), it has "fails" attribute in PL file. Should we rename/delete this as the "bad " test is indeed a "good" test in this case?

wsnyder, 2023-04-01:
Yes, remove the test (assuming nothing is still "bad"), and add whatever checks it has to the non-bad .v file.

RRozak, 2023-04-03:
Hi, unfortunately, the code you provided is incorrect. Parens can't be skipped in nonvoid functions. The section 13.5 of IEEE 1800-2017 says:
> It shall be illegal to omit the parentheses in a tf_call unless the subroutine is a task, void function, or class method. If the subroutine is a nonvoid class function method, it shall be illegal to omit the parentheses if the call is directly recursive.

t_func_no_parentheses_bad.v tests if appropriate error is thrown if a call of a function that returns `int` doesn't have parentheses.

Thanks for taking the time, but I think that the behavior of Verilator is correct.

svenka3, 2023-04-03:
Thanks for the pointer to LRM. Indeed you are dot-on with functions in static (module/interface/package). However, class methods are allowed as the referred quote also mentions. I reduced the SVUnit code that showed the original issue to below and it fails on Verilator:

```verilog
class XmlElement;
  function string get_start_tag_contents();
    return "Local";
  endfunction : get_start_tag_contents
  function string as_string_with_indent(string indent = "");
    string result;
    result = get_start_tag_contents;
  endfunction 
endclass

```
So perhaps we still need some clause to differentiate class method vs module non-void function call.

Thanks

RRozak, 2023-04-03:
In order to check if a function is a method, You can call `classMethod()` on a function object.

svenka3, 2023-04-03:
For a simple class method (as above) - I get the following:
```log
ftaskp->isFunction() : 1 ftaskp->classMethod : 0
```
Code:
```verilog
              cout  << " ftaskp->isFunction() : "
                   << ftaskp->isFunction() 
                   << " ftaskp->classMethod : "
                   << ftaskp->classMethod();

                // AF if (!ftaskp->isFunction()) {
                if (!ftaskp->isFunction() || 
                    ftaskp->classMethod() ) {
```

Any clues? 
Thanks

RRozak, 2023-04-03:
`isFunction()`checks if an AstNodeFTask object returns a value: https://github.com/verilator/verilator/blob/7e1980af7a67f90967556dda5be07fef72452876/src/V3AstNodeOther.h#L128
So if it is called on a method that returns a value, it should return 1.
Maybe it is not the best name, but this method is used only to distinguish between tasks and functions.

`classMethod()` starts working properly after V3LinkResolve stage, because it is set there: https://github.com/verilator/verilator/blob/7e1980af7a67f90967556dda5be07fef72452876/src/V3LinkResolve.cpp#L123. I didn't know about it when I proposed using it. Sorry for that.

I think that the best approach is to make `classMethod()` working at the moment You want to use it.

svenka3, 2023-04-03:
@RRozak Thanks for the hints.
> I think that the best approach is to make classMethod() working at the moment You want to use it.

I now borrowed some code from V3LinkResolve.cpp to V3LinkDot.cpp and am able to detect classMethod inside LinkDot. @wsnyder - please let me know if you are OK with this approach, will update my PR if so (I may delete and send a new PR as I had wrongly "git rm" that _bad test).

Will run regression prior to the new PR.

Thanks

wsnyder, 2023-04-04:
Please push the new code to the pull request and I'll review thanks.

wsnyder, 2023-04-10:
Thanks for the fix.

closed 2023-04-10 by wsnyder