verilator/verilator #3905: t_dist_cinclude does not do anything in some distros

Zvord, 2023-01-25:
Some tests from `make test` fail on my macOS system for some reason. One of them is `t_dist_cinclude`:
```
zvord@Mac-mini-Ilia verilator % ./test_regress/t/t_dist_cinclude.pl
==SUMMARY: Left 2  Passed 0  Failed 0  Time 0:00
======================================================================
dist/t_dist_cinclude: ==================================================
src/V3FunctionTraits.h:22:#include <stddef.h>
%Warning: dist/t_dist_cinclude: Files like stdint.h instead of cstdint: src/V3FunctionTraits.h
dist/t_dist_cinclude: %Error: Files like stdint.h instead of cstdint: src/V3FunctionTraits.h
dist/t_dist_cinclude: FAILED: Files like stdint.h instead of cstdint: src/V3FunctionTraits.h
==SUMMARY: Left 1  Passed 0  Failed 1  Time 0:01
======================================================================
vlt/t_dist_cinclude: ==================================================
-Skip: vlt/t_dist_cinclude: scenario 'vlt' not enabled for test
vlt/t_dist_cinclude: -Skip: Skip: scenario 'vlt' not enabled for test
==SUMMARY: Passed 0  Failed 1  Time 0:01

======================================================================
        #dist/t_dist_cinclude: %Error: Files like stdint.h instead of cstdint: src/V3FunctionTraits.h
                make && test_regress/t/t_dist_cinclude.pl  --dist
TESTS DONE, FAILED: Passed 0  Failed 1  Time 0:01
```

And it is right to fail because `V3FunctionTraits.n` includes `stddef.h`, while the test is expecting `cstddef.h`.

Then why does it pass in GitHub CI, as well as in a clean Ubuntu installation?
```perl
my $cmd = "cd $root && fgrep -n include $files | sort";
my $grep = `$cmd`;
```

I found that the second line is silently discarded. It probably has something to do with the length of `$files`. It's 170k symbols. When I trim `$files` to something around 130k symbols, the script starts to work. I do not understand this limit. `ARG_MAX` is much bigger and I don't know what else to check.

I suggest to avoid calling the shell and stick to perl. [Here's my take](https://github.com/Zvord/verilator/commit/f23a4b1c3859aa6fe99de5775329eda407f22101). It successfully fails Git CI.

Another question, is what to do with `cstddef`? It cannot be found neither in my Ubuntu, nor in macOS and I'm a bit confused by all this library business.

wsnyder, 2023-01-25:
You are right.  Mind turning that into a pull request along with fixing the #include?  Note you can use "file_contents" to simplify your code - see t_dist_whitespace.pl

Zvord, 2023-01-25:
Of course, but I'm still wrestling with `cstddef`. Simply changing the include breaks the compilation.

closed 2023-01-26 by wsnyder