verilator/verilator #3906: Module names in verilated cpp

AndrewNolte, 2023-01-25:
Currently I don't think there's any way to get the module name from the verilated cpp. You can get the instance name, and the path, but the actual module name is baked into the cpp class, and is not in the scope data.

I think we could put the module name in with ScopeData for vpi modules, and hence also put that in VerilatedScope. generate scopes would be blank, and top scopes would have a repeat of the name.

wsnyder, 2023-01-25:
I presume you are referring to vpiDefName.  Your analysis and suggested implementation seem correct.  Do you think you can provide a pull request to add this (I'm assuming you want this feature)?

AndrewNolte, 2023-01-25:
Yes, is vpiDefName not implemented currently? I wasn't planning on adding that to the vpi, mainly just as a member to VerilatedScope. But if it's not too much extra I could maybe do that as well. Would that just be overriding defName for VerilatedVpioModule?

wsnyder, 2023-01-25:
That sounds right, and update the t_vpi_module test to check it please.

AndrewNolte, 2023-01-26:
This is sort of touching the same code as https://github.com/verilator/verilator/pull/3840, could we please get that merged?

wsnyder, 2023-02-04:
Thanks for fixing this.

wsnyder, 2023-02-04:
For some reason all the regression tests didn't run against the pull, when it hit master there are many test failures (https://github.com/verilator/verilator/actions/runs/4088934517).  I've reverted.  I can't reopen a pull, so please make a new pull, thanks & sorry.

forensicgarlic, 2023-03-18:
I'm a little confused on the status of this issue, based on being mentioned in the 5.008 release, but getting re-openned on Feb 3. Is this incorporated in 5.008, but work still remains?

wsnyder, 2023-03-18:
It is not in.  I created Changes from the git log, where it was, but I forgot it had been backed out, I will remove from Change history.

ksstrike, 2024-10-31:
Bringing the issue back on the table with #5572 as it's quite a useful functionality to have.
Aiming primarily for cocotb integration, so the assumption is we're running Verilator with `--public-flat-rw`.
I am aware of prior work having been done in #3931 / #3938, but I'm not exactly sure what the roadblock was.
@AndrewNolte seeing as you're still active in the project, could you chime in and elaborate?

closed 2024-11-12 by wsnyder