verilator/verilator #3908: Multithreading and SystemC - Data Races on DefaultContext

TeflonAntihaft, 2023-01-26:
Hi,
We are currently working on framework for parallelizing the evaluation of SystemC partitions with asynchronous communication between them, using a verilated design written in Verilog as the DUT of those partitions. As an example the same verilated design is instantiated in two distinct threads, which communicate with a microcontroller simulation on a third thread, all inside a testbench. 

Through the Google Thread Sanitizer we recently noticed that data races are reported on various parts of the DefaultContext coming from files which part of the Verilator support library. Regarding the behavior of the mulithreaded C++ simulation described in [1] whereas each instanced needs its own context and the fact that the SystemC model does not support setting a custom context, supposedly always using the global default context for each instance, i wanted to clarify that i understood the situation right: 

As SystemC as a language does not support multithreading in its evaluation model, the Verilator SystemC model, in contrast to the C++ version, was also designed with this assumption in mind and thus completely relies on the global default context, ultimately making it not thread safe. Is this what was tried to be said by [2]? Because while reading the docs i personally find it hard to differentiate between which statements  only hold trough for the C++ model and or both. (For example section [3]: Do both models support it? And what happens if i build a SystemC design with "--threads 1"? Is the support library now thread safe or does this only hold true for C++? etc.)

So until something changes inside the model itself it is not possible to have two instances of the same design in different threads without the risk of data races, right? It would be great if someone could clarify that whole situation for me. 

Greetings
David

[1] https://veripool.org/guide/latest/connecting.html#verilated-and-verilatedcontext
[2] https://verilator.org/guide/latest/verilating.html#multithreaded-verilog-and-library-support: "If using [--sc], the SystemC kernel is not thread-safe; therefore, the eval thread and main thread must be the same."
[3] https://verilator.org/guide/latest/verilating.html#multithreading

wsnyder, 2023-01-26:
You are correct that the SystemC kernel is not thread safe.  The VerilatedContext used for all models in SystemC must be the same, I just added that to the docs you mention.

In recent Verilator versions, all code it emits is intended to be threadsafe, using the same or a different context, regardless of --thread settings.  That safety stops though at the edge of Verilator, the SystemC kernel will break if called from multiple threads.

It's a typical use case to have a single thread running the SystemC kernel, calling two Verilator models (from that one thread) sharing the default VerilatedContext.  Each of those Verilated models may use multiple threads internally.

Be sure to use a recent Verilator version, ideally 5.006.  The last race bug I can recall was fixed in 5.002 so certainly nothing older than that.

Closing as I think answered, but feel free to follow up in this issue.

closed 2023-01-26 by wsnyder

TeflonAntihaft, 2023-01-27:
Yeah, the thread safety of the SystemC kernel itself is what our framework is addressing. And the concept itself is working. So if the context in SystemC Verilator itself is thread safe it should be working.

Regarding the version i checked back and it seems like were still at 4.224. Ill try and reach out to the maintainer of the Conan package to see if we can get a update to 5.X.

Thanks a lot for the clarification!