verilator/verilator #3909: SELRANGE lint error only check the bits, not outer packed dimensions

streamradiopi, 2023-01-26:
```
I was surprised to find that SELRANGE only checks for invalid bit-slices in a word, and does not check out-of-range indexing of outer dimensions. The most concise example I could make is below, basically if you select an out-of-range index in the innermost dimension, the Verilator linter catches it for you. When you do that in outer indexes, it does not. I'm not sure if there is a good reason why this is works-as-intended, or if it's just a bug.
```

Thanks for taking the time to report this.

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```
/* verilator lint_off UNUSEDSIGNAL */
/* verilator lint_off UNDRIVEN */
module verilator_bug (
  input logic [3:0][3:0] i1,
  input logic [3:0] i2,
  output logic [3:0] o1,
  output logic o2
);

assign o1 = i1[4];  // beyond range; should be a lint error?
// assign o2 = i2[4];  // bit index beyond range; is a SELRANGE lint error

endmodule
```

What 'verilator --version' are you using?  Did you try it with the git master version?
```
Verilator 5.004
```

What OS and distribution are you using?
```
MacOS 12.5.1
```

May we assist you in trying to fix this in Verilator yourself?
```
No
```

wsnyder, 2023-01-26:
I suspect this is a relic of pre-1995 when there were not multidimensional packed arrays.  Perhaps you'd be willing to provide tests and a fix pull request for this?

paul-demo, 2023-01-28:
I spent a couple hours reading through the code and trying to compile Verilator (both for the first time), but don't have time to go any further and become a code contributor right now. I'm sure you already know this but, and I'm somewhat confused by all the places where widths are checked or changed, but here's what I think:

V3WidthSel.cpp :: visit(AstSelBit* nodep) is turning packed array indexing into AstSel indexing (among other things). Probably this is where a SELRANGE error should be added, before an AstSel is emitted with the raw bit accesses chosen by the outer indexing.

V3width.cpp ::  visit(AstSel* nodep) is doing some checking, but I don't think by the time it has been made an AstSel that it has enough information to evaluate whether the outer packed dimensions were violated or not.

V3width.cpp ::  visit(AstArraySel* nodep) not totally sure what's going on here; this is maybe only for unpacked arrays.

V3const.cpp :: Not sure why there are additional SELRANGE checks in here.

paul-demo, 2023-01-28:
Sorry; I have various github accounts and logged into a different one to reply. I'm the OP.