verilator/verilator #3910: Support #0 delay inactive resumption scheduling

wsnyder, 2023-01-29:
Many dozen of the Icarus Verilog tests run against Verilator in SV-Tests fail with this unsupported message

     %Error-ZERODLY: ...: Unsupported: #0 delays do not schedule process resumption in the Inactive region

e.g. https://chipsalliance.github.io/sv-tests-results/?v=verilator+ivtest+regress-fsv_br971_iv

It would be good to get this resolved so we can see those test results, and get a nice score boost too.

@kbieganski thoughts?

gezalore, 2023-01-29:
FWIW, conceptually this should be easy with an extra loop around the `act` section, i.e.: when the `act` section setlled, look again to see if there are any suspended coroutines pending for the current time step and if so repeat the `act` loop (+ allowing 0 delay suspends). There might be some minor complications to resolve, but should not be hard.

quark17, 2023-01-29:
I was about to file this issue! because we use `#0` in the Verilog generated by [the Bluespec compiler (BSC)](https://github.com/B-Lang-org/bsc).  However, I wanted to request one step further: in a design where the only use of `#` is `#0`, would it be possible to support it without the dynamic runtime overhead that I assume exists when you choose `--timing`, by doing some static analysis to determing a fixed schedule of assignments.

The BSC compiler takes a high-level design and figures out all the scheduling, to produce a very structural design -- all the new values of the state at each clock, based on the previous values, with no loops.  From this representation, BSC can generate C++, which iterates (once) through all the assignments and then steps to the next clock.  BSC can also generate Verilog, that is very amenable to Verilation (with `--no-timing`).  However, the representation is allowed to contain systemtasks (like `$display`) or imported C functions.  And the Verilog language doesn't support calls to these from within a structural design, so BSC has to use always blocks in the output.  (In BSC's C++ output, the calls occur in the appropriate place in the linearized sequence of assignments.)  And even though the design is structural / statically scheduled / linearizable, the Verilog needs to contain `#0` to support the situation when the output of one task feeds into logic that then becomes the input of another task.  Here's an example:
```
module test1 (CLK);
  input  CLK;

  reg [31 : 0] x;
  wire [31 : 0] y;

  assign y = ~x;

  // system tasks
  always@(posedge CLK)
  begin
    x = $stime;
    #0;
    $display("i = %b", x);
    $display("q = %b", y);
    $finish(0);
  end

  // A 2-value simulator and will resort to '0' for the
  // value of 'x' if it is not otherwise initialized
  //
  initial
  begin
    x = 32'hAAAAAAAA;
  end

endmodule // test1
```
What we want to happen is that `x` is assigned the value from `$stime`, then the value of `y` is updated by noticing that its assignment depends on `x` (which has changed), then we return to the always block to execute the display of `y`.  We expect the following output (which iverilog gives):
```
i = 00000000000000000000000000000001
q = 11111111111111111111111111111110
```
Older Verilator or newer Verilator with `--no-timing` or will give this:
```
i = 00000000000000000000000000000001
q = 01010101010101010101010101010101
```
Newer Verilator with `--timing` rejects it, as you've noted.  A first step would be for Verilator with `--timing` to accept it and execute correctly.  However, I wonder if there can be a third option, that gives the correct behavior but without any dynamic looping.  The sequence of assignments should be discoverable from static analysis.  In fact, for the example I've given, BSC now just inlines the assignments into the always block:
```
    x = $stime;
    $display("i = %b", x);
    assign y = ~x;
    $display("q = %b", y);
```
But this is not always possible.  For example, the assignments could be in a separate module:
```
module sub(i, q);
  input  [31 : 0] i;
  output [31 : 0] q;

  wire [31 : 0] q;

  assign q = ~i;
endmodule // sub

module test2 (CLK);
  input  CLK;

  reg [31 : 0] x;
  wire [31 : 0] y;

  sub s(.i(x), .q(y));

  // system tasks
  always@(posedge CLK)
  begin
    x = $stime;
    #0;
    $display("i = %b", x);
    $display("q = %b", y);
    $finish(0);
  end

  // A 2-value simulator and will resort to '0' for the
  // value of 'x' if it is not otherwise initialized
  //
  initial
  begin
    x = 32'hAAAAAAAA;
  end

endmodule // test2
```
I don't know enough about how Verilator works to know, one, if there would be any dynamic overhead to worry about and, two, if avoiding it (particularly in cross-module situations) is possible.  And apologies if this should be opened as a separate issue.  I'm, anyway, happy to see this issue opened for supporting `#0` in some way.  Thanks!

gezalore, 2023-01-30:
> Isn't a lot of Active stuff being moved to nba? 

Yes, I think you are right, probably would have to move all those back to `act`. Still not very difficult to do that, but would hurt performance.

wsnyder, 2023-03-13:
I made ZERODLY a warning instead of error, this makes all but one ivtest pass, so considering this good enough for now.

closed 2023-03-13 by wsnyder