verilator/verilator #3915: lint check rule

KingkingLiu, 2023-01-30:
As far as I know, the quantity of lint rules is about 98. You tell me that I need to provide the specific lint rule to you. Hence I take the rule from spyglass lint, named W88, which flags memories where all their elements are not set in the design. Here is the code for the example :
module test(in1, in2, out1);
input [3:0] in1, in2;
output [3:0] out1;
reg [3:0] out1;
reg [7:0] mem [0:3];
always@(in1) begin
out1 = in1 & in2;
mem[0] = {in1, out1};
mem[2] ={in1, in2};
end
endmodule

wsnyder, 2023-01-30:
This would go into V3Undriven. There are several other improvements people have suggested there too.

KingkingLiu, 2023-01-30:
So,  all the new rules include what i request will be updated in the future?

KingkingLiu, 2023-01-31:
> This would go into V3Undriven. There are several other improvements people have suggested there too.
When will these features be updated? By the way, I'm a C++ beginner and I wonder how long it would take to read the verilator code and add some new lint rule.The reason I need to know this time is because I need to decide whether to delve into the source code myself

wsnyder, 2023-01-31:
To solve this ideally I believe the right way is for each variable to make a used and consumed data structure. That data structure that looks a bit like a [binary] decision tree. A data type that is an array of structs with an array will be tracked as an array containing a map containing an array.  As something is used/driven it sets the appropriate element in that data tree.  The complication is when an array is variably-indexed it needs to link the references of multiple of those trees together. 

Ignoring your knowledge of C++, from a general compsci standpoint if you have no idea what I mean above this might be a hard first project.  (Though I welcome your trying it anyways, or looking at any other bugs to fix!)

I don't expect to get to it myself for months at best, I can't speak as to others doing it.

KingkingLiu, 2023-01-31:
> To solve this ideally I believe the right way is for each variable to make a used and consumed data structure. That data structure that looks a bit like a [binary] decision tree. A data type that is [...]

Thanks for your reply! The paragraph you explained above is difficult for me to understand, because I have no background in computer science. This may sound funny, but I was wondering if there was a way I could add new lint rules without reading the source code.

wsnyder, 2023-01-31:
No, there's no way to add lint without sources.  At least one "dead" commercial tool supported this using a VPI like language, I don't know for sure if others do.

BTW the problem above is to detect that e.g. for "var a[1:0][1:0]" that e.g only "a[1][0]" is unused. It's an easier problem that probably catches all the easy bugs is to check each index separately, so it would only know that e.g. "a[1][*}" or "a[*][0]" is unused.

KingkingLiu, 2023-01-31:
> No, there's no way to add lint without sources. At least one "dead" commercial tool supported this using a VPI like language, I don't know for sure if others do.
> 
> BTW the structure I proposed above is to detect that e.g. for "var a[1:0][1:0]" that e.g only "a[1][0]" is unused. It's an easier problem that probably catches all the easy bugs is to check each index separately, so it would only know that e.g. "`a[1][*}`" or "`a[*][0]`" is unused.

This may be excessive request that I provide you with some new lint rules and you help me add them.

KingkingLiu, 2023-02-02:
I notice in the internals.rst file that Verilator makes a series of passes over the AST, progressively refining and optimizing it. Does it affect the lint check? I mean the refining and optimizing may change the source code of verilog.

wsnyder, 2023-02-02:
Yes, the lint checks must be done at the appropriate place before possible optimizations "break" them.

KingkingLiu, 2023-02-02:
Thank you very much for your prompt reply and generous help！！

KingkingLiu, 2023-02-02:
While I did a quick read of the source code and found that lint checks aren't a single decoupled module, other sources I've read tell me that the general process is to create an AST and then do lint check. I'm a little confused here. Can you give me explain? Thanks a lot

wsnyder, 2023-02-02:
I'm not sure if you are asking a Verilator specific question or a general one.  For general see e.g. https://en.wikipedia.org/wiki/Abstract_syntax_tree for specifics see https://github.com/verilator/verilator/blob/master/docs/internals.rst

KingkingLiu, 2023-02-03:
> I'm not sure if you are asking a Verilator specific question or a general one. For general see e.g. https://en.wikipedia.org/wiki/Abstract_syntax_tree for specifics see https://github.com/verilator/verilator/blob/master/docs/internals.rst
I mean I want to know the relationship between lint and AST in verilator, or how lint works.