verilator/verilator #3916: CMake does not work with system-verilog package files

yodalee, 2023-01-30:
I try to use the `package` provided by system-verilog. Here is the [test project](https://gist.github.com/yodalee/a30e14f86c2601fcfaaa079a6e70cfb8) with 4 files.

In the project, `out.sv` will import `pkg.sv`, so in cmake file I put both `out.sv` and `pkg.sv` into the source lists.
Then I find that:
1. If I put `out.sv` before `pkg.sv`, the verilator will report error that it cannot find `pkg.sv`
2. If I put `pkg.sv` before `out.sv`, the cmake will only include directory of `pkg.dir` and omit the `out.dir`. This cause the compiler cannot find the generated `Vout.h`.

Of course replace import with include is one of the solution, but I think it is not quite right.
I think there should be a correct setting which I missed?

Please help.

wsnyder, 2023-02-01:
@codelec perhaps you can help?

codelec, 2023-02-01:
I am able to compile by adding "PREFIX Vout"

codelec, 2023-02-01:
This line https://github.com/verilator/verilator/blob/master/verilator-config.cmake.in#L145 is the reason the compiler cannot find the generated Vout.h

yodalee, 2023-02-01:
Confirm that designate PREFIX can solve this issue.

yodalee, 2023-02-02:
I think we can emit a warning when there is not PREFIX with multiple SOURCES. Since it is not an error if user specify multiple SOURCES without PREFIX.
I can submit the pull request if the above change is accepted.

wsnyder, 2023-02-02:
A pull would be appreciated.

closed 2023-02-03 by wsnyder