verilator/verilator #3919: Different handles from iteration vs. vpi_handle_by_name

alexforencich, 2023-01-31:
This issue is present in 5.006, but presumably has been around for quite some time (e.g. https://github.com/cocotb/cocotb/issues/2244 is from 2020).  The jist of it is this: iterating over some object in the hierarchy returns different handles vs. using 
`vpi_handle_by_name`.  The problem is that the handles from iteration are read-only, and this trips up cocotb if `dir()` is called at some point as this "primes" cocotb's simulator object cache with a bunch of read-only handles, and then cocotb cannot update any of those signals.

I suppose a workaround that could potentially be implemented in cocotb is simply to never use the results from iteration directly, instead use iteration to get to get the names, and then look those up with `vpi_handle_by_name`, since that returns usable handles.  But other simulators do not seem to have this problem, so it would be better to make sure that iteration returns the same handles as `vpi_handle_by_name` in the first place (or, otherwise returns handles that are equivalent).

To reproduce: run pretty much any test case in cocotb-bus with `make SIM=verilator`.  Specifically I ran `https://github.com/cocotb/cocotb-bus/tree/master/tests/test_cases/test_avalon_stream` and it hangs, waiting for some signal to change, which never does because cocotb can't update any of the signals.  Disabling case-insensitive matching the `Bus` object avoids calling `dir()` and allows the tests to run to completion.

wsnyder, 2023-01-31:
One quick workaround would be to get the full name of the object from the iterate, then call vpi_get_handle_by_name on it.

IEEE: "If a tool creates two handles that refer to the same object, the tool may create two distinct handles or may provide the same handle in both cases. Two distinct handles that refer to the same object are equivalent."

So, as you described the problem, it looks like creating the two separate handles is complant, but the iterator's one being read only isn't compliant.

Looking at Verilator's code though, I agree it makes two different handles between vpi_handle_by_name and vpi_iterate. I see vpi_handle_by_name returns a VerilatedVpioVar, while iterate returns a VerilatedVpioVarIter (a vpiIterator which seems right), then a vpi_scan returns what should be an equivalently capable VerilatedVpioVar on the same object.  So the problem isn't obvious by reading.

Could you modify one of the verilator tests (probably t_vpi_var.cpp) to show the problem please?

alexforencich, 2023-01-31:
How about this?

```
int _mon_check_getput_iter() {
    const char* p;

    TestVpiHandle vh2 = VPI_HANDLE("sub");
    CHECK_RESULT_NZ(vh2);

    TestVpiHandle vh10 = vpi_iterate(vpiReg, vh2);
    CHECK_RESULT_NZ(vh10);
    CHECK_RESULT(vpi_get(vpiType, vh10), vpiIterator);

    TestVpiHandle vh11 = vpi_scan(vh10);
    CHECK_RESULT_NZ(vh11);

    s_vpi_time t;
    t.type = vpiSimTime;
    t.high = 0;
    t.low = 0;
    s_vpi_value v;
    v.format = vpiIntVal;
    v.value.integer = 0;
    vpi_put_value(vh11, &v, &t, vpiNoDelay);

    vpi_get_value(vh11, &v);
    CHECK_RESULT(v.value.integer, 0);

    v.value.integer = 1;
    vpi_put_value(vh11, &v, &t, vpiNoDelay);

    vpi_get_value(vh11, &v);
    CHECK_RESULT(v.value.integer, 1);

    return 0;
}
```

wsnyder, 2023-02-01:
Your test was faulty, you were accessing sub.subsig1 which was marked "public_flat_ro", that is it was correctly read-only.   I modified that in the test along with some other test enhancements and committed the new test.  So back to you to modify the trunk version of the test to show the problem.

alexforencich, 2023-02-01:
Hm.  Well, at this point I may need some help from the cocotb maintainers, because TBH I don't fully understand exactly how cocotb interacts with the simulator.

alexforencich, 2023-02-01:
Working backwards from the setup with cocotb, there is an interesting difference in the logs when `vpi_put_value` is called.  

Working case:

```
-V{t1,111}- vpi: vpi_put_value 0x55bfba0838b8 0x7ffd9549e320
-V{t1,112}- vpi:   vpi_put_value name=TOP.axis_tvalid fmt=6 vali=0
-V{t1,113}- vpi:   varp=0x55bfb9ea178c  putatp=0x55bfb9ea178c
```

Non-working case:

```
-V{t1,225}- vpi: vpi_put_value 0x562641d84eb8 0x7ffc8b5857c0
-V{t1,226}- vpi:   vpi_put_value name=test_axis.axis_tvalid fmt=6 vali=0
-V{t1,227}- vpi:   varp=0x5626417ad796  putatp=0x5626417ad796
```

alexforencich, 2023-02-01:
Strange, there definitely are two scopes:

```
internalsDump:
  Version: Verilator 5.006 2023-01-22
  Argv: sim_build/Vtop +verilator+debug
  scopesDump:
    SCOPE 0x55914dc74bc0: TOP
       VAR 0x55914dc74e68: axis_tdata
       VAR 0x55914dc74ee8: axis_tdest
       VAR 0x55914dc74f68: axis_tid
       VAR 0x55914dc74fe8: axis_tkeep
       VAR 0x55914dc75068: axis_tlast
       VAR 0x55914dc750e8: axis_tready
       VAR 0x55914dc75168: axis_tuser
       VAR 0x55914dc751e8: axis_tvalid
       VAR 0x55914dc75268: clk
       VAR 0x55914dc752e8: rst
    SCOPE 0x55914dc74bf8: test_axis
       VAR 0x55914dc753a8: DATA_WIDTH
       VAR 0x55914dc75428: DEST_WIDTH
       VAR 0x55914dc754a8: ID_WIDTH
       VAR 0x55914dc75528: KEEP_WIDTH
       VAR 0x55914dc755a8: USER_WIDTH
       VAR 0x55914dc75628: axis_tdata
       VAR 0x55914dc756a8: axis_tdest
       VAR 0x55914dc75728: axis_tid
       VAR 0x55914dc757a8: axis_tkeep
       VAR 0x55914dc75828: axis_tlast
       VAR 0x55914dc758a8: axis_tready
       VAR 0x55914dc75928: axis_tuser
       VAR 0x55914dc759a8: axis_tvalid
       VAR 0x55914dc75a28: clk
       VAR 0x55914dc75aa8: rst
```

But, they aren't identical, so it doesn't seem like the same module is getting included multiple times.

The top-level C++ code is quite simple:

https://github.com/cocotb/cocotb/blob/master/cocotb/share/lib/verilator/verilator.cpp

Not sure if anything jumps out at you immediately.  

Maybe the problem is `std::unique_ptr<Vtop> top(new Vtop(""));`?  Is that already created by default and we just need to get a reference to it instead?

wsnyder, 2023-02-01:
Looks like your verilog code has two top modules in it.

alexforencich, 2023-02-01:
This is the verilog code in question: https://github.com/alexforencich/cocotbext-axi/blob/master/tests/axis/test_axis.v .  No `TOP` in there anywhere.

And cocotb is building it with the following command:

verilator -cc --exe -Mdir sim_build -DCOCOTB_SIM=1 --top-module test_axis -Wno-SELRANGE -Wno-WIDTH -GDATA_WIDTH=8 -GID_WIDTH=8 -GUSER_WIDTH=1 -GDEST_WIDTH=8 -GKEEP_WIDTH=1 --debug -CFLAGS "-DVL_DEBUG -DVERILATOR_SIM_DEBUG -g" --vpi --public-flat-rw --prefix Vtop -o Vtop --no-l2name -LDFLAGS "-Wl,-rpath,/home/alex/.local/lib/python3.10/site-packages/cocotb/libs -L/home/alex/.local/lib/python3.10/site-packages/cocotb/libs -lcocotbvpi_verilator" --timescale 1ns/1ns test_axis.v /home/alex/.local/lib/python3.10/site-packages/cocotb/share/lib/verilator/verilator.cpp

(well, this one has some debug options in there, but that should be the only verilog source file)

alexforencich, 2023-02-01:
Oh, I guess I can get rid of those linter flags as well, and the --no-l2name was something I added to see if it did anything (it didn't seem to change anything).

alexforencich, 2023-02-01:
During the initial run of verilator to do the code generation ("verilation" suppose), it looks like this phantom "TOP" module is present, as it appears in Vtop_034_scope.tree:

```
Verilator Tree Dump (format 0x3900) from <e5657> to <e6154>
     NETLIST 0x5555561853a0 <e1> {a0aa}  $root [1ns/1ns]
    1: MODULE 0x5555561aa950 <e718> {c32ai}  __024root  L1 [P] [1ns]
    1:1: CELLINLINE 0x5555561ee840 <e4196> {c32ai}  test_axis -> test_axis [scopep=0x5555561edc40]
    1:2: VAR 0x5555561aacf0 <e722> {c41bj} @dt=0x5555561a9b60@(G/w1)  clk [PI] INPUT [P] [VSTATIC]  WIRE
    1:2: VAR 0x5555561aae70 <e727> {c42bj} @dt=0x5555561a9b60@(G/w1)  rst [PI] INPUT [P] [VSTATIC]  WIRE
    1:2: VAR 0x5555561ab1f0 <e733> {c44bj} @dt=0x5555561a2550@(G/w8)  axis_tdata [PIO] INOUT [P] [VSTATIC]  WIRE
    1:2:3: CONST 0x555556205320 <e5333> {c44bj} @dt=0x5555561a2550@(G/w8)  8'h0
    1:2: VAR 0x5555561ab570 <e739> {c45bj} @dt=0x5555561a38e0@(G/w1)  axis_tkeep [PIO] INOUT [P] [VSTATIC]  WIRE
    1:2:3: CONST 0x5555561ef250 <e5369> {c45bj} @dt=0x5555561a9b60@(G/w1)  1'h0

SNIP

    1:2: TOPSCOPE 0x5555561dc930 <e5661#> {c32ai}
    1:2:2: SCOPE 0x5555561edc40 <e5660#> {c32ai}  TOP [abovep=0] [cellp=0] [modp=0x5555561aa950]
    1:2:2:1: VARSCOPE 0x55555619a530 <e5663#> {c41bj} @dt=0x5555561a9b60@(G/w1)  TOP->clk [T] [scopep=0x5555561edc40] -> VAR 0x5555561aacf0 <e722> {c41bj} @dt=0x5555561a9b60@(G/w1)  clk [PI] INPUT [P] [VSTATIC]  WIRE
    1:2:2:1: VARSCOPE 0x5555561f27b0 <e5666#> {c42bj} @dt=0x5555561a9b60@(G/w1)  TOP->rst [T] [scopep=0x5555561edc40] -> VAR 0x5555561aae70 <e727> {c42bj} @dt=0x5555561a9b60@(G/w1)  rst [PI] INPUT [P] [VSTATIC]  WIRE
    1:2:2:1: VARSCOPE 0x5555561f2880 <e5669#> {c44bj} @dt=0x5555561a2550@(G/w8)  TOP->axis_tdata [T] [scopep=0x5555561edc40] -> VAR 0x5555561ab1f0 <e733> {c44bj} @dt=0x5555561a2550@(G/w8)  axis_tdata [PIO] INOUT [P] [VSTATIC]  WIRE
    1:2:2:1: VARSCOPE 0x5555561f2950 <e5672#> {c45bj} @dt=0x5555561a38e0@(G/w1)  TOP->axis_tkeep [T] [scopep=0x5555561edc40] -> VAR 0x5555561ab570 <e739> {c45bj} @dt=0x5555561a38e0@(G/w1)  axis_tkeep [PIO] INOUT [P] [VSTATIC]  WIRE
    1:2:2:1: VARSCOPE 0x5555561f2a20 <e5675#> {c46bj} @dt=0x5555561a9b60@(G/w1)  TOP->axis_tvalid [T] [scopep=0x5555561edc40] -> VAR 0x5555561afd10 <e745> {c46bj} @dt=0x5555561a9b60@(G/w1)  axis_tvalid [PIO] INOUT [P] [VSTATIC]  WIRE
    1:2:2:1: VARSCOPE 0x5555561f2cc0 <e5678#> {c47bj} @dt=0x5555561a9b60@(G/w1)  TOP->axis_tready [T] [scopep=0x5555561edc40] -> VAR 0x5555561b0090 <e751> {c47bj} @dt=0x5555561a9b60@(G/w1)  axis_tready [PIO] INOUT [P] [VSTATIC]  WIRE
    1:2:2:1: VARSCOPE 0x5555561f2d90 <e5681#> {c48bj} @dt=0x5555561a9b60@(G/w1)  TOP->axis_tlast [T] [scopep=0x5555561edc40] -> VAR 0x5555561b0410 <e757> {c48bj} @dt=0x5555561a9b60@(G/w1)  axis_tlast [PIO] INOUT [P] [VSTATIC]  WIRE
```

In the t_vpi_var test, TOP does not appear in the scopesDump listing, so whatever is going on definitely seems to be specific to this particular run, but I haven't been able to pin down where `TOP` is coming from.

marlonjames, 2023-02-01:
I think I have a better understanding of what is happening now.

The handles being found via iteration are using `vpi_iterate(vpiReg, root_module_handle)`.

But `vpi_handle_by_name("test_axis.axis_tvalid", NULL)` hits a special case where it checks the ports of `TOP` scope first:
https://github.com/verilator/verilator/blob/ef9f44353253d16a01909a8eaaf35db27c7a84a5/include/verilated_vpi.cpp#L1460-L1464

Verilator always overwrites inputs to internal scopes using parent scope values from the top-level ports down.
cocotb is finding internal handles when using `vpi_iterate()`.

marlonjames, 2023-02-01:
I also found something different with the cocotb setup relating to the TOP scope as detailed here:

---
https://github.com/cocotb/cocotb/issues/2346#issuecomment-760369843

For the `t_vpi_module` Verilator test, `Vt_vpi_module__Syms.cpp` sets up the scopes and hierarchy like so:
```cpp
// Setup scopes
    __Vscope_t.configure(this, name(), "t", "t", -12, VerilatedScope::SCOPE_MODULE);
    __Vscope_t__mod_a.configure(this, name(), "t.mod_a", "mod_a", -12, VerilatedScope::SCOPE_MODULE);
    __Vscope_t__mod_a__mod_c__02E.configure(this, name(), "t.mod_a.mod_c__02E", "mod_c.", -12, VerilatedScope::SCOPE_MODULE);
    
    // Set up scope hierarchy
    __Vhier.add(0, &__Vscope_t);
    __Vhier.add(&__Vscope_t, &__Vscope_t__mod_a);
    __Vhier.add(&__Vscope_t__mod_a, &__Vscope_t__mod_a__mod_c__02E);
```

But for cocotb's `sample_module`, in `Vtop__Syms.cpp` we get:
```cpp
    // Setup scopes
    __Vscope_TOP.configure(this, name(), "TOP", "TOP", 0, VerilatedScope::SCOPE_OTHER);
    __Vscope_sample_module.configure(this, name(), "sample_module", "sample_module", -12, VerilatedScope::SCOPE_OTHER);
    
    // Set up scope hierarchy
    
```
The extra `TOP` scope may or may not be relevant, but the `sample_module` scope is `VerilatedScope::SCOPE_OTHER` instead of `VerilatedScope::SCOPE_MODULE`

alexforencich, 2023-02-01:
The thing that I can't figure out is where this `TOP` thing is coming from in the first place.  It doesn't seem to always be present.

Compare the scope dumps from cocotbext-axi and one of the verilator test cases:

```
  scopesDump:
    SCOPE 0x55674ed5dbc0: TOP
       VAR 0x55674ed5de68: axis_tdata
       VAR 0x55674ed5dee8: axis_tdest
       VAR 0x55674ed5df68: axis_tid
       VAR 0x55674ed5dfe8: axis_tkeep
       VAR 0x55674ed5e068: axis_tlast
       VAR 0x55674ed5e0e8: axis_tready
       VAR 0x55674ed5e168: axis_tuser
       VAR 0x55674ed5e1e8: axis_tvalid
       VAR 0x55674ed5e268: clk
       VAR 0x55674ed5e2e8: rst
    SCOPE 0x55674ed5dbf8: test_axis
       VAR 0x55674ed5e3a8: DATA_WIDTH
       VAR 0x55674ed5e428: DEST_WIDTH
       VAR 0x55674ed5e4a8: ID_WIDTH
       VAR 0x55674ed5e528: KEEP_WIDTH
       VAR 0x55674ed5e5a8: USER_WIDTH
       VAR 0x55674ed5e628: axis_tdata
       VAR 0x55674ed5e6a8: axis_tdest
       VAR 0x55674ed5e728: axis_tid
       VAR 0x55674ed5e7a8: axis_tkeep
       VAR 0x55674ed5e828: axis_tlast
       VAR 0x55674ed5e8a8: axis_tready
       VAR 0x55674ed5e928: axis_tuser
       VAR 0x55674ed5e9a8: axis_tvalid
       VAR 0x55674ed5ea28: clk
       VAR 0x55674ed5eaa8: rst
```

```
  scopesDump:
    SCOPE 0x560929f61640: t
       VAR 0x560929f620e8: count
       VAR 0x560929f62168: fourthreetwoone
       VAR 0x560929f621e8: half_count
       VAR 0x560929f62268: onebit
       VAR 0x560929f622e8: quads
       VAR 0x560929f62388: text
       VAR 0x560929f62408: text_byte
       VAR 0x560929f62488: text_half
       VAR 0x560929f62508: text_long
       VAR 0x560929f62588: text_word
       VAR 0x560929f62608: twoone
       VAR 0x560929f62688: x
       VAR 0x560929f62708: y
    SCOPE 0x560929f616b0: t.arr[1].arr
       VAR 0x560929f627c8: check
       VAR 0x560929f62848: rfr
       VAR 0x560929f628c8: sig
       VAR 0x560929f62948: verbose
    SCOPE 0x560929f61720: t.arr[2].arr
       VAR 0x560929f62a08: check
       VAR 0x560929f62a88: rfr
       VAR 0x560929f62b08: sig
       VAR 0x560929f62b88: verbose
    SCOPE 0x560929f61790: t.arr[3].arr
       VAR 0x560929f62c48: check
       VAR 0x560929f62cc8: rfr
       VAR 0x560929f62d48: sig
       VAR 0x560929f62dc8: verbose
    SCOPE 0x560929f61800: t.arr[4].arr
       VAR 0x560929f62e88: check
       VAR 0x560929f62f08: rfr
       VAR 0x560929f62f88: sig
       VAR 0x560929f63008: verbose
    SCOPE 0x560929f61870: t.arr[5].arr
       VAR 0x560929f630c8: check
       VAR 0x560929f63148: rfr
       VAR 0x560929f631c8: sig
       VAR 0x560929f63248: verbose
    SCOPE 0x560929f618e0: t.arr[6].arr
       VAR 0x560929f63308: check
       VAR 0x560929f63388: rfr
       VAR 0x560929f63408: sig
       VAR 0x560929f63488: verbose
    SCOPE 0x560929f61678: t.arr__BRA__1__KET__
    SCOPE 0x560929f616e8: t.arr__BRA__2__KET__
    SCOPE 0x560929f61758: t.arr__BRA__3__KET__
    SCOPE 0x560929f617c8: t.arr__BRA__4__KET__
    SCOPE 0x560929f61838: t.arr__BRA__5__KET__
    SCOPE 0x560929f618a8: t.arr__BRA__6__KET__
    SCOPE 0x560929f61918: t.sub
       VAR 0x560929f63548: subsig1
       VAR 0x560929f635c8: subsig2
```

The verilator test case has no `TOP` at all for some reason.

alexforencich, 2023-02-01:
Alright, so `TOP` does appear in the dumps from t_vpi_var as well:

```
$ cat Vt_vpi_var_034_scope.tree 
Verilator Tree Dump (format 0x3900) from <e5209> to <e5816>
     NETLIST 0x5555561813a0 <e1> {a0aa}  $root [1ps/1ps]
    1: MODULE 0x555556271620 <e4174> {d15ai}  __024root  L1 [P] [1ps]
    1:1: CELLINLINE 0x555556243580 <e4769> {d15ai}  t -> t [scopep=0x555556290b90]
    1:1: CELLINLINE 0x5555562723a0 <e4771> {d99ae}  t__DOT__arr -> __BEGIN__ [scopep=0x555556290b90]
    1:1: CELLINLINE 0x555556271af0 <e4773> {d100ay}  t__DOT__arr__BRA__1__KET__ -> __BEGIN__ [scopep=0x555556290b90]
    1:1: CELLINLINE 0x555556272750 <e4775> {d100ay}  t__DOT__arr__BRA__2__KET__ -> __BEGIN__ [scopep=0x555556290b90]
    1:1: CELLINLINE 0x555556279a50 <e4777> {d100ay}  t__DOT__arr__BRA__3__KET__ -> __BEGIN__ [scopep=0x555556290b90]

SNIP

    1:2: TOPSCOPE 0x555556290c80 <e5213#> {d15ai}
    1:2:2: SCOPE 0x555556290b90 <e5212#> {d15ai}  TOP [abovep=0] [cellp=0] [modp=0x555556271620]
    1:2:2:1: VARSCOPE 0x55555628db80 <e5215#> {d26ak} @dt=0x5555562242d0@(G/w1)  TOP->clk [T] [scopep=0x555556290b90] -> VAR 0x555556271880 <e4178> {d26ak} @dt=0x5555562242d0@(G/w1)  clk [PI] INPUT [P] [clker]  PORT
    1:2:2:1: VARSCOPE 0x55555628cd30 <e5218#> {d26ak} @dt=0x5555562242d0@(G/w1)  TOP->t__DOT__clk [T] [scopep=0x555556290b90] -> VAR 0x555556216fe0 <e4844> {d26ak} @dt=0x5555562242d0@(G/w1)  t__DOT__clk PORT
```

So, it's now a question of why it's appearing in the scopeDump output for only one of these and not both.

alexforencich, 2023-02-01:
Aha, this is related to `--public-flat-rw`.  If that is specified, then `TOP` appears in scopeDump and you get the lines in `__Syms.cpp` setting it up.

alexforencich, 2023-02-01:
> Verilator always overwrites inputs to internal scopes using parent scope values from the top-level ports down.
> cocotb is finding internal handles when using `vpi_iterate()`.

Wait, so you can't set stuff "down in the hierarchy" with verilator?  That's going to end up being a rather major problem, since right now it seems like the only way to handle things like concatenated ports, other than creating additional wrappers that break stuff out.  I wonder what the solution to this might be, maybe back-propagating any changes internally back up to the top-level ports instead of immediately overwriting the values.

marlonjames, 2023-02-01:
I'm not positive about other internal signals, but the ports on the root module always get the values from the TOP wrapper.

Example:
```cpp
// top->eval() will call this:
VL_INLINE_OPT void Vtop___024root___ico_sequent__TOP__0(Vtop___024root* vlSelf) {
    if (false && vlSelf) {}  // Prevent unused
    Vtop__Syms* const __restrict vlSymsp VL_ATTR_UNUSED = vlSelf->vlSymsp;
    VL_DEBUG_IF(VL_DBG_MSGF("+    Vtop___024root___ico_sequent__TOP__0\n"); );
    // Body
    vlSelf->sample_module__DOT__clk = vlSelf->clk;
    ...
}
```

alexforencich, 2023-02-01:
Ah, I guess that's another piece of the puzzle: this strange behavior probably only applies to top-level ports, as if it isn't connected to a top-level port then things might work as expected.  I'll do some more poking around along those lines tomorrow and see if I can put together some useful test cases.

alexforencich, 2023-02-01:
So, I just did some more poking around and I think @marlonjames is right about what's going on here.  The handles are not read-only, they're just pointing at different hierarchy levels and as a result are effectivley read-only as the updated value gets immediately blown away.  Not only are they actually set and overwritten, but the corresponding events are also triggered, as we can see from the cocotbext-axi testbench logs.

Working case:

```
     2.00ns INFO     cocotb.test_axis.axis              Reset asserted
     2.00ns INFO     cocotb.test_axis.axis              Reset asserted
     2.00ns INFO     cocotb.test_axis.axis              Reset asserted
     6.00ns INFO     cocotb.test_axis.axis              Reset de-asserted
     6.00ns INFO     cocotb.test_axis.axis              Reset de-asserted
     6.00ns INFO     cocotb.test_axis.axis              Reset de-asserted
```

Non-working case:

```
     2.00ns INFO     cocotb.test_axis.axis              Reset asserted
     2.00ns INFO     cocotb.test_axis.axis              Reset asserted
     2.00ns INFO     cocotb.test_axis.axis              Reset asserted
     2.00ns INFO     cocotb.test_axis.axis              Reset de-asserted
     2.00ns INFO     cocotb.test_axis.axis              Reset de-asserted
     2.00ns INFO     cocotb.test_axis.axis              Reset de-asserted
```

I had noticed this before and thought it was odd, but the waveform dumps showed the signals in question stuck at 0.  So whatever is going on doesn't even make it to the trace as a "glitch" pulse.

In the verilator test suite, I added a couple of port-related tests, basically copies of getput and getput_iter but pointing at a top-level port.  The getput test got a handle to `TOP.port` instead of `t.port`, and the getput_iter test got a handle to `t.port`.  Both tests passed, BUT I did not trigger an eval operation between setting and reading the port value.

It looks like the real problem is a combination of the special-casing of `TOP`, combined with unconditionally propagating signals instead of only propagating changed signals.

I think the solution to this is going to be some combination of getting rid of/reworking the special-casing of `TOP` and fixing propagation of updates so that only changed signals are propagated.  Presumably changes have to be tracked somewhere anyway to support triggering on rising and falling edges and such, so conditioning updates on this (at least for stuff marked `public_flat_rw`) should be doable.

alexforencich, 2023-02-01:
IMO, personally I also think that it's a bug if you request a handle to `t.port`, you get a handle to a completely different signal (`TOP.port`) instead.

wsnyder, 2023-02-01:
I forget why TOP is exposed in this way, that does seem fishy, maybe TOP doesn't need to be exposed at all.  Anyhow I suspect t.port and TOP.port are correctly (correct by Verilator design, ignoring IEEE) aliases of each other, as they represent the same net after net resolution.

Note there's no way you're going to only get the VPI affected signal changed; Verilator isn't an event mode simulator, it must reevaluate everything appropriately on the next eval() call. For a variable to stay written it must be at a storage point (e.g. represent a flop on a posedge), if you write a variable that is driven by any combo logic, it will typically be rewitten at the next eval(), as Verilator recomputes combo logic whenever it thinks it is best for performance.  Note everything in TOP is most lilely in this will-overwrite category, as TOP simply consists of continuous assignments to lower-level signals.

In summary, in general the VPI is intended to primarily allow VPI read to almost everything, VPI write only to storage for backdoor CSR loading and similar.  Supporting VPI write to "random" signals is intentionally beyond the intent of Verilator's implemented VPI.

alexforencich, 2023-02-01:
They do not appear to be aliases of each other, as at least with the cocotbext-axi testbench I have been running, writes to anything under `test_axis` seem to get immediately overwritten by zeros, while writes to stuff under `TOP` seem to work fine.  I'll do some more poking around in the tests to put together a test case for this.

marlonjames, 2023-02-01:
I would be satisfied if iterating over the root module would return handles to the top-level objects that aren't overwritten by `eval()`. That way at least the top-level can be stimulated from the cocotb testbench.

alexforencich, 2023-02-01:
I need to be able to write to signals that are driven by top-level ports deeper in the hierarchy that do not necessarily correspond 1:1 with top-level ports (for example, there might be a top-level port that is [127:0] that gets split into two internal signals of [63:0], and I would be driving those internal signals with separate calls to `vpi_put_value`), and potentially synchronous to different clocks).  In this case, I will not be driving the corresponding top-level ports at all, so it would be acceptable to back-propagate the updates to the top-level ports.

wsnyder, 2023-04-08:
Is there any action Verilator needs to take to resolve this?

alexforencich, 2023-04-09:
Yes.  I think there are at several potential paths.  

1. Hack Verilator's internal checks to return the faked-out TOP references when enumerating, instead of the internal handles that are immediately overwritten by the values from TOP.  This will solve the issue for connecting to top-level ports, but will do nothing for connecting to deeper ports.
2. Rework signal propagation down the hierarchy so that values are only propagated when changes occur, instead of unconditionally on every call to eval().  This will fix the issue for both top-level and internal ports, but potentially could have some performance implications due to having to detect signal changes.  Possibly this could be done as an extension of public_flat_rw, so it can be enabled only when needed.
3. Remap writes to internal signals back onto the driver (TOP or otherwise).  Basically, for a write to an internal signal, transparently perform the equivalent operation at the source, then propagate this updated value normally.  This will fix the issue for both top-level and internal ports, and should not have any significant performance implications.  

I haven't had time to put together a proper test case for this; the previous test case didn't show the issue as eval() didn't get called between setting and reading, and due to how the test was architected (tests were run from within a VPI callback) it wasn't immediately clear how to make this happen.

wsnyder, 2023-04-09:
I think 1 & 3 are similar approaches, either could work.  2 is a major change just for VPI so seems unlikely.

alexforencich, 2023-04-09:
2 and 3 are going to get you closer to what a normal HDL simulator does, so if 3 is reasonable to implement then that's the one that I would vote for.  

It might also be worth looking at a more explicit method for handling aliased signals across hierarchy levels, which could potentially supersede the check that returns TOP references and potentially work for some internal signals with no performance penalty.  And maybe this could also reduce/remove the need to explicitly copy values when propagating down the hierarchy.

alexforencich, 2023-11-07:
My understanding is that #4618 is an implementation of option 1, which does not fully address the issue (it only fixes top-level ports).  So this should be re-opened.

ArturBieniek4, 2026-09-02:
Fixed by #4618.

PR #4618 (2023-11-07): Fix VPI TOP level variable iteration (#3919)
Closes #3919 by implementing option 1 discussed there.
I followed the pattern set by `vpi_handle_by_name`.
Save the `TOP` scope if found, and check for matching ports when iterating on the module vars.
closed 2026-09-02 by ArturBieniek4