verilator/verilator #3924: Duplicate include in top and sub module causes verilator segmentation fault

yodalee, 2023-02-01:
I encounter this issue when trying to make systemverilog package right. I find that, if I include a package file with:
1. `include in top module
2. `include in sub module
3. import in top module

will cause segmentation fault in verilator.

The [link to the repository](https://github.com/yodalee/test_verilator) that reproduces the issue.

The steps to reproduce the issue:
1. Clone the repository
5. cd test_verilator && mkdir build && cd build
6. cmake ..

You should see the error message of:
```
CMake Error at /usr/share/verilator/verilator-config.cmake:239 (message):
  Verilator command failed (return code=Segmentation fault)
Call Stack (most recent call first):
  CMakeLists.txt:10 (verilate)
```

The OS is Archlinux.

The verilator version:
Verilator 5.006 2023-01-22 rev UNKNOWN.REV
The master branch also has this issue.
AFAIK, the verilator 4.228 is highly possible with same issue.

I can have a look at this issue. I think it is not a serious issue, since no one is gonna use sv package in this way.

wsnyder, 2023-02-02:
Thanks for the clear bug report. I get a MODDUP verilator error not a crash.  I suspect you are using an old version and hitting #3666.

closed 2023-02-02 by wsnyder

yodalee, 2023-02-02:
Can you do a double confirm? Since I can reproduce the issue with master branch.
The build version is `Verilator 5.007 devel rev v5.006-35-g0e955d503`
The command to reproduce is `verilator_bin --cc top.sv pkg.sv`

wsnyder, 2023-02-02:
I get an error before any crash, you get absolutely no output first?  What does --debug --gdbbt say?

yodalee, 2023-02-03:
Yes I will get the warning-MODDUP, then the verilator will crash. Below is the full log w/i `--debug --gdbbt`
```
Starting Verilator 5.007 devel rev v5.006-35-g0e955d503
- Verilator.cpp:587:  Option --verilate: Start Verilation
- V3File.cpp:218:        --check-times failed: no input obj_dir/Vtop__verFiles.dat
- V3ParseImp.cpp:276: parseFile: top
  Preprocessing top.sv
- V3PreShell.cpp:141:     Reading top.sv
- V3PreShell.cpp:141:     Reading sub.sv
- V3PreShell.cpp:141:     Reading pkg.sv
- V3PreShell.cpp:141:     Reading pkg.sv
- V3ParseImp.cpp:325: Lexing top.sv
- V3ParseImp.cpp:276: parseFile: pkg
  Preprocessing pkg.sv
- V3PreShell.cpp:141:     Reading pkg.sv
- V3ParseImp.cpp:325: Lexing pkg.sv
%Warning-MODDUP: pkg.sv:1:9: Duplicate declaration of module: 'pkg'
    1 | package pkg;
      |         ^~~
                 sub.sv:4:1: ... note: In file included from sub.sv
                 top.sv:2:1: ... note: In file included from top.sv
                 pkg.sv:1:9: ... Location of original declaration
    1 | package pkg;
      |         ^~~
-node: PACKAGE 0x55b045105100 {e1aj} u4=0x55b0451029b0  pkg  L0 [LIB] [NONE]
                 ... For warning description see https://verilator.org/warn/MODDUP?v=5.007
                 ... Use "/* verilator lint_off MODDUP */" and lint_on around source to disable this message.
%Warning-MODDUP: pkg.sv:1:9: Duplicate declaration of module: 'pkg'
    1 | package pkg;
      |         ^~~
                 pkg.sv:1:9: ... Location of original declaration
    1 | package pkg;
      |         ^~~
-node: PACKAGE 0x55b0451003b0 {e1aj} u4=0x55b0450ff7f0  pkg  L0 [LIB] [NONE]
- V3LinkLevel.cpp:48: modSortByLevel()
- V3Ast.cpp:1193:     Dumping obj_dir/Vtop_001_cells.tree
zsh: segmentation fault (core dumped)  /home/yodalee/git/verilator/bin/verilator_bin --debug --gdbbt --cc top.sv
```

I think the segmentation fault is happened in the dumping stage. Is it acceptable if the verilator already gives the warning before it crashs.

wsnyder, 2023-02-03:
As long as you get an error out we don't generally fix crashes after that point, as it's an almost never ending battle.  So closing this.

yodalee, 2023-02-03:
Understood.