verilator/verilator #3947: False UNOPTFLAT warning with split_var

uenoku, 2023-02-09:
In the following example, only Foo produces `UNOPTFLAT` warning whereas `Bar` doesn't. I expected `Foo` and `Bar` are equivalent after V3SplitVar pass. 
```verilog
module Foo(
  input  a,
  output b);
  
  wire  [1:0] w /*verilator split_var*/;
  assign w = {w[0], a};
  assign b = w[1];
endmodule

module Bar(
  input  a,
  output b);
  
  wire  w_1, w_0;
  assign {w_1, w_0} = {w_0, a};
  assign b = w_1;
endmodule
```
Verilator 5.004 2022-12-14 rev v5.004

wsnyder, 2023-02-09:
At present a complete assignment of a variable prevents the split from working. This was an intentional decision as otherwise a split would need to break every bit up individually, rather then splitting by which bits in the vector are assigned.  This could perhaps be improved to look at what bits are either used or assigned and take the smaller subset of each.

Perhaps you'd be willing to create a patch to explore that change?

uenoku, 2023-02-09:
> At present a complete assignment of a variable prevents the split from working. This was an intentional decision as otherwise a split would need to break every bit up individually, rather then splitting by which bits in the vector are assigned. 

I see, that makes sense. I'm wondering we can allow spilt if assigned value is array literal `'{...}` or bitconcat `{..}` and corresponding elements are statically determined.

> Perhaps you'd be willing to create a patch to explore that change?

Yeah, I would like to try working on that. Is V3SplitVar.cpp the right place we can modify?

wsnyder, 2023-02-12:
BTW note this should split ok

assign {w[1], w[0]} = {w[0], a};

although an earlier optimize pass might combine the concat before it gets split, I'm not sure.

The split certainly could be taught to work on more constructs.  Yes, V3SplitVar is where to improve. Thanks.