verilator/verilator #3948: Erroneous C++ files generated when using CMake with dash (-) in HDL file names.

Metric-Void, 2023-02-09:
When CMake is used to verilate the project and the project HDL files contain dashes (-) in their file names, Verilator will generate C++ classes with dashes in their names, which is illegal.

The error does not seem to be rise when using `verilator` in command line.

Example of a generated file:
```C++
// Verilated -*- C++ -*-
// DESCRIPTION: Verilator output: Primary design header
//
// This header should be included by all source files instantiating the design.
// The class here is then constructed to instantiate the design.
// See the Verilator manual for examples.

#ifndef _VVERILOG_FILE_WITH_DASH_H_
#define _VVERILOG_FILE_WITH_DASH_H_  // guard

#include "verilated.h"

//==========

class Vverilog-file-with-dash__Syms;      // <-- ILLEGAL CLASS NAME
class Vverilog-file-with-dash___024unit;  // <-- ILLEGAL CLASS NAME

//----------
```

Verilator Version: `Verilator 5.007 devel rev v5.006-72-g4eb280601` (Git master version)

Minimal working example: https://github.com/Metric-Void/verilator-cmake-error

OS/Distro: Ubuntu/22.10 Kinetic

Metric-Void, 2023-02-10:
Source of error here. CMake feeds inappropriate prefixes to the command line.

Filenames are sanitized before getting used as `m_prefix`
https://github.com/verilator/verilator/blob/d34aa10fafef343d03ae940e20448b83d955b95b/src/V3Options.cpp#L956-L964

While command-line args are not.
https://github.com/verilator/verilator/blob/d34aa10fafef343d03ae940e20448b83d955b95b/src/V3Options.cpp#L1371

Changing CMake files seems more appropriate - changing the code to `DECL_OPTION("-prefix", CbVal, [this](const char* unsafe_prefix) { m_prefix = AstNode::encodeName(unsafe_prefix); });` will fix the C++ code generated, but the cmake file will not have the correct path of header files.

wsnyder, 2023-02-11:
Thanks for your first pull!

closed 2023-02-11 by wsnyder