verilator/verilator #3955: forceable and public can't be used in the same project

lucasasselli, 2023-02-11:
On the git master version:

wsnyder, 2023-02-12:
Perhaps you could contribute a pull request with a test and fix - presumably some spot maybe in V3Name needs a !isPublic().

lucasasselli, 2023-02-18:
I will look into it! Is the expectation that the control signals are generated as flat, correct?

wsnyder, 2023-02-21:
> I will look into it! Is the expectation that the control signals are generated as flat, correct?

Yes, currently V3Force works post flattening (after V3Scope and before V3Descope)

wsnyder, 2025-09-21:
Sweeping old bugs... the behavior desired I think can be achieved with:

```
    input a  /* verilator forceable */ /* verilator public_flat */,
```

Will commit a tests to make sure this keeps working.

closed 2025-09-21 by wsnyder