verilator/verilator #3956: verilator.pdf still not reproducible

ldoolitt, 2023-02-12:
My pull request  #3918 failed to achieve its desired result in all cases.
The python "datetime" module is too complicated, and can give different answers depending on the local timezone.
See  https://tests.reproducible-builds.org/debian/rb-pkg/unstable/amd64/verilator.html
and https://lists.reproducible-builds.org/pipermail/rb-general/2023-February/002856.html .

After discussions with the reproducible-build folks (thanks Vagrant!), I attach two patches for your consideration, with different ways to construct the date printed on the title page of verilator.pdf.

1) [dates2.txt](https://github.com/verilator/verilator/files/10715295/dates2.txt) Uses "time" instead of "datetime". Tested (again, sigh) works (for sure, this time!).  Defaults to today's date, but can be over-ridden by the build system using SOURCE_DATE_EPOCH.

2) [dates3.txt](https://github.com/verilator/verilator/files/10715299/dates3.txt) Always uses the date stamp found alongside the version in the Makefile, where it sets the PACKAGE_VERSION variable. Beautiful, stable, simple, and arguably best for actual releases.  Maybe not as good when there's active development.

I'd be happy to turn either one (but probably not both) into a pull request. Any guidance would be appreciated.

wsnyder, 2023-02-12:
dates3 while cleaner for releases doesn't work for devel as there's no date there, it'll break completely.

It seems odd to make the pdf date "today" if it is a very old version.  Maybe we should be taking the github last commit date?  Then it works and doesn't even need SOURCE_DATE_EPOCH (presuming you have .git available - do you?)

wsnyder, 2023-02-12:
BTW src/config_rev does something similar now with the build number.

wsnyder, 2023-02-12:
BTW src/config_rev does something similar now with the build number.  So perhaps it should also write a date to a file that the doc build can include.  The docs makefile would also need to call it to make sure it gets generated.

ldoolitt, 2023-02-12:
First let me summarize what I think is consensus so far:

1. It's not helpful to put the date on which sphinx was run onto the front page of `verilator.pdf`.  That date should have more to do with when the documentation and/or program source was last updated.
2. `conf.py` needs to handle both Release and Devel situations sensibly, which `dates3.txt` above does not do.

> presuming you have .git available - do you?

Debian packaging takes place in _a_ git repo, yes, but not the _upstream_ git repo. OTOH, packagers like Debian should (and are very likely to) use a release with a well defined date, and therefore would use the non-git code path.

Here's another attempt: [dates4.txt](https://github.com/verilator/verilator/files/10717529/dates4.txt)
It borrows the concept from `src/config_rev`, but just calls `git log -n 1 --pretty=%as` on its own.
Work-in-progress: the PDF now has the extraneous word "Release" every time, which I couldn't find the source of. Is that deep inside some sphinx template?
Example for devel:
```
Verilator
Release Devel 5.007
Wilson Snyder
2023-02-12
```
Example for release:
```
Verilator
Release Release 5.006
Wilson Snyder
2023-01-22
```

wsnyder, 2023-02-12:
Looks reasonable, I appreciate this was painful - I learned about time().  I keep seeing an off-by-one date bug when I try to rent cars in Europe and wonder if that's the case.

I think the first Release is built into sphinx, so I'd remove that from your patch, so we get "Release Devel 5.007" or "Release 5.006" that's close enough.

Can you convert to a pull please?

ldoolitt, 2023-02-12:
> Looks reasonable, I appreciate this was painful - I learned about time(). I keep seeing an off-by-one date bug when I try to rent cars in Europe and wonder if that's the case.

Probably, yes.  See a related comment in Debian bug 780259: "Living on the west coast of the US myself, my own blog posts (which use UTC) are frequently dated in the "future" by local time, and I don't think it's that big of a deal."

> I think the first Release is built into sphinx, so I'd remove that from your patch, so we get "Release Devel 5.007" or "Release 5.006" that's close enough.

OK. Deal.

> Can you convert to a pull please?

OK.  I think I should also change `git log -n 1 --pretty=%as` to `git log -n 1 --pretty=%cs`. Otherwise some corner cases could push the listed date unrealistically far back in time.

PR #3957 (2023-02-13): Try harder to put a useful date on the front page of verilator.pdf
Fixes #3956
closed 2023-02-13 by wsnyder