verilator/verilator #3958: Local declaration should hide package wildcard import

algrant-arm, 2023-02-13:
Although the error here is "Unsupported in C", it's a similar case to #599, which is fixed, so it may be that this is intended to work but the fix for #599 just didn't handle the enum case.

```
package pack;
  typedef enum logic[0:0] { P = 1'b0 } e_t;
endpackage

module a
import pack::*;
();
  localparam P = 1;
endmodule
```
output:
```
%Error: a.sv:8:14: Unsupported in C: Variable has same name as ENUMITEM 'P': 'P'
    8 |   localparam P = 1;
      |              ^
%Error: Exiting due to 1 error(s)
        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```
SystemVerilog standard says pack::P is "potentially locally visible" in a, and it ends up not being visible since P is locally declared in a. Seems like once Verilator front end has resolved the scope, it does not need to generate C code for pack::P in a at all, so there should be no clash of names in the C code - but even if it had to somehow reference both pack::P and the localparam, it could achieve that by mangling the names somehow.

algrant-arm, 2023-02-13:
This is similar:
```
package pack;
  typedef enum logic[0:0] { P = 1'b0 } e_t;
endpackage

module b
import pack::*;
#(parameter P = 1)
();
endmodule

module c;
  b #(.P(2)) b1;
endmodule;
```
output:
```
%Error: b.sv:7:13: Unsupported in C: Variable has same name as ENUMITEM 'P': 'P'
    7 | #(parameter P = 1)
      |             ^
%Error: b.sv:12:8: Parameter pin not found: 'P'
   12 |   b #(.P(2)) b1;
      |        ^
%Error: Exiting due to 2 error(s)
        ... See the manual at https://verilator.org/verilator_doc.html for more assistance.
```
b's declaration of its parameter P should hide the wildcard import. As for the previous case, this isn't a clash in generated C code, it's happening well before that.

wsnyder, 2023-02-15:
As long as the enum isn't public this shouldn't cause a problem in the emitted C code.

Would you be willing to make a pull requests with these tests and attempt a fix?

wsnyder, 2023-02-15:
I agree adding the symbol table state seems right.  With that I'm not sure you need to also add an Ast member that it was wildcard imported, but it's fine if you do, nothing else should be upset.  It's ok if the pretty printers don't use it.

wsnyder, 2025-09-09:
Both examples now work on master, fixed earlier.

closed 2025-09-09 by wsnyder