verilator/verilator #3959: Non-sensical warning

jotego, 2023-02-15:
**How to reproduce:**

If you lint the attached file, which I reproduce below, you will get the following error:

```
%Warning-WIDTH: bugreport.v:7:31: Operator SUB expects 32 bits on the LHS, but LHS's VARREF 'UROM_DEPTH' generates 11 bits.
                                : ... In instance uRom
    7 |     reg [UROM_WIDTH-1:0] uRam[UROM_DEPTH];      
      |                               ^~~~~~~~~~
                ... For warning description see https://verilator.org/warn/WIDTH?v=5.006
                ... Use "/* verilator lint_off WIDTH */" and lint_on around source to disable this message.
%Error: Exiting due to 1 warning(s)
```
But this warning is about a non-existing SUB operation and a non-existing LHS. Making `UROM_DEPTH=1024;` fixes the warning, but I don't think the warning should have ever been there.

input file:
```
localparam UADDR_WIDTH = 4'd10;
localparam UROM_WIDTH = 5'd17;
localparam UROM_DEPTH = 11'd1024;

module uRom( input clk, input [UADDR_WIDTH-1:0] microAddr, output logic [UROM_WIDTH-1:0] microOutput);
    reg [UROM_WIDTH-1:0] uRam[UROM_DEPTH];      
    initial begin
        $readmemb("microrom.mem", uRam);
    end
    
    always @( posedge clk) 
        microOutput <= uRam[ microAddr];
endmodule
```

**What 'verilator --version' are you using?  Did you try it with the git master version?**
Tested with 5.06 and _git master_.

**What OS and distribution are you using?**
Ubuntu mate 20.04 LTS

**May we assist you in trying to fix this in Verilator yourself?**
I cannot help with software development

wsnyder, 2023-02-18:
The warning is pointing at the wrong point, it should be at the subtract of UROM_WIDTH-1.

wsnyder, 2023-05-06:
Fixed; suppressing the message seemed better, as was original suggestion.

closed 2023-05-06 by wsnyder