verilator/verilator #3963: Internal Error: ../V3Number.h:196: `num` member accessed when data type is UNINITIALIZED

catkira, 2023-02-18:
I am getting the error

` Internal Error: ../V3Number.h:196: `num` member accessed when data type is UNINITIALIZED`

My verilator version is 5.004 compiled from git.

I am running it with this command line
```
perl /usr/local/bin/verilator -cc --exe -Mdir /mnt/d/git/5G_SSB_sync/sim_build/IN_DW=32_OUT_DW=32_TAP_DW=32_PSS_LEN=128_ALGO=0_WINDOW_LEN=8_CP_ADVANCE=18_CFO=100 -DCOCOTB_SIM=1 --top-module receiver --vpi --public-flat-rw --prefix Vtop -o receiver -LDFLAGS -Wl,-rpath,/mnt/d/git/cocotb/cocotb/libs -L/mnt/d/git/cocotb/cocotb/libs -lcocotbvpi_verilator -y /mnt/d/git/5G_SSB_sync/submodules/verilator-unisims -I/mnt/d/git/5G_SSB_sync/hdl/CIC -I/mnt/d/git/5G_SSB_sync/hdl/fft-core -GIN_DW=32 -GOUT_DW=32 -GTAP_DW=32 -GPSS_LEN=128 -GALGO=0 -GWINDOW_LEN=8 -GCP_ADVANCE=18 -GPSS_LOCAL_0=258502891756376941587586161045879784311204241311776625863883810776614030043198554106213210957132947178442009743845684005108653758967785798299549986597026563108858258634164984098791266121641883707321666674630468572952684076257042217766374610988078749967170533064184594742187764383686231671023670792090951010649380193386437812606879366551788566837297294620064167544523409479281781030884688473504650318271015173983830997499562993215724885937935849911317883869286850752150067222208869537980797036693652536053544794075952303402018957207330387492950827074964312147922270731654813703444836926129790680491744924895543873189081003095293056055014442439329681288066554054688251765482385500629132701939024246189642663583878163197134203725024972971789048566653997170797520977863333246658909049606019371330279989643150241116509441937800369324218955283138256624466901546710821500249000091520246269660249007335376904146252148098613033996998309227329489406108037602174461392868048619736011619057891644297184362788687609168071683165665712185586920062610106957335597026752490326842115252355928203438701927882375609622635629569304642377662596548579531741593162656463539758554553314646353451441336408263015931572522988385319081763675682719161732416207014 -GPSS_LOCAL_1=53524053993988436724331007034498422555267245712309977514631104336541220399100291842678443944322694337295190089079880876371431413155453957430842801934534929907970942402130988101002448235697993152827766270468026386781069444303132754027145809967629989330041837775024052400554207394917810459928224474287421480460201617046041291445869927033706352361012258800368658307386064459489012825037948340907626401497985099440977807126461786488655758162093048125301521233058621908519289031999795710775311078679496836018891848021621432986762870073015267085928169991031212919519839817711608887042434952506489746599483313552039562675972632984113170130936427771360829987953840515723598190236200233656777498213796679831626467810988866891899809830353607006011562865845757936270720150606774737794860261113028504950854442670942041639468341023514356243411557230984348727175474072699794976517120767034358216035530043615139512323327005507953186179735448741001379321177358382599008243855779440116481417651042773094391938542281048958954410260189836683058930728267666507256385058316012741424958433912506557514012667024033942577694126912846096670981027252848515777985832836646043048575687834193633736180889635751692478672389456367831621653140837180046161871960150 -GPSS_LOCAL_2=727692769084738296199514023252920695673110030442313862546494873936543905495465112115440116815497188759705501197535815646587584004276996791752064390383868594373468281020086948060874236171013709961163610340379507171482337110374530681613783202145013092788241815937210870966018139227097764872124474751641815788946053535112477087657136312458412346464311882492009223025761124892768175816113081836351703734815811808091866746647126345392212546930379911037828830015901316964722094413997851972518098152588844953943989631956252982179713758548699641252079762660878419761048466354457474404621561351873599525730175985582816209389093268802777458896604049172997098801344634666255667809975101223655006143928299719459732650385256333085359378423915656658073936416189406066453060753211327862638631082429491463981174442406347800928707185018063163844717919172077760369562510766221642403859597949340448739407881830030591052787606094267650225809326781620844915882640690415553409790547109072354439213916437092450968553340222620544271455309910007978982204458776367926065672863081889771206329426079874861760146265338141627687895347081066879697688631730952361555541164239652950942434867989007569712233445450099054556358665324201755063801958547716130561744500874 /mnt/d/git/cocotb/cocotb/share/lib/verilator/verilator.cpp /mnt/d/git/5G_SSB_sync/hdl/receiver.sv /mnt/d/git/5G_SSB_sync/hdl/frame_sync.sv /mnt/d/git/5G_SSB_sync/hdl/channel_estimator.sv /mnt/d/git/5G_SSB_sync/hdl/PSS_detector.sv /mnt/d/git/5G_SSB_sync/hdl/Peak_detector.sv /mnt/d/git/5G_SSB_sync/hdl/PSS_correlator.sv /mnt/d/git/5G_SSB_sync/hdl/SSS_detector.sv /mnt/d/git/5G_SSB_sync/hdl/LFSR/LFSR.sv /mnt/d/git/5G_SSB_sync/hdl/FFT_demod.sv /mnt/d/git/5G_SSB_sync/hdl/complex_multiplier/complex_multiplier.v /mnt/d/git/5G_SSB_sync/hdl/CIC/cic_d.sv /mnt/d/git/5G_SSB_sync/hdl/CIC/comb.sv /mnt/d/git/5G_SSB_sync/hdl/CIC/downsampler.sv /mnt/d/git/5G_SSB_sync/hdl/CIC/integrator.sv /mnt/d/git/5G_SSB_sync/hdl/FFT/fft/fft.v /mnt/d/git/5G_SSB_sync/hdl/FFT/fft/int_dif2_fly.v /mnt/d/git/5G_SSB_sync/hdl/FFT/fft/int_fftNk.v /mnt/d/git/5G_SSB_sync/hdl/FFT/math/int_addsub_dsp48.v /mnt/d/git/5G_SSB_sync/hdl/FFT/math/cmult/int_cmult_dsp48.v /mnt/d/git/5G_SSB_sync/hdl/FFT/math/cmult/int_cmult18x25_dsp48.v /mnt/d/git/5G_SSB_sync/hdl/FFT/twiddle/rom_twiddle_int.v /mnt/d/git/5G_SSB_sync/hdl/FFT/delay/int_align_fft.v /mnt/d/git/5G_SSB_sync/hdl/FFT/delay/int_delay_line.v /mnt/d/git/5G_SSB_sync/hdl/FFT/buffers/inbuf_half_path.v /mnt/d/git/5G_SSB_sync/hdl/FFT/buffers/outbuf_half_path.v /mnt/d/git/5G_SSB_sync/hdl/FFT/buffers/int_bitrev_order.v
```
The repo with the code is here: https://github.com/catkira/open5G_rx

wsnyder, 2023-02-18:
Can you extract into a minimal example please?

catkira, 2023-02-18:
minimal example looks like this
```verilog
`timescale 1ns / 1ns

module PSS_correlator
(
    input                                 clk_i,
    output  reg                           test
);

reg regA;

always @(posedge clk_i) begin
    test <= regA;
end

endmodule
```

catkira, 2023-02-18:
A minimal example looks like this

```verilog
`timescale 1ns / 1ns

module receiver
(
    input                                 clk_i,
    output  reg            [23:0]         m_axis_out_tdata
);

reg [32 - 1: 0] filter_result;

always @(posedge clk_i) begin
    m_axis_out_tdata <= {{(24 - 32){1'b0}}, filter_result};
end

endmodule
```

catkira, 2023-02-19:
When I run it with v4.106 I get this interesting error message
```
%Warning-WIDTHCONCAT: receiver.sv:22:57: More than a 8k bit replication is probably wrong: 4294967291
                                       : ... In instance receiver
   22 |         m_axis_out_tdata <= {{(OUT_DW - REQUIRED_OUT_DW){1'b0}}, filter_result};
      |                                                         ^
                      ... Use "/* verilator lint_off WIDTHCONCAT */" and lint_on around source to disable this message.
%Error: Verilator internal fault, sorry. Suggest trying --debug --gdbbt
%Error: Command Failed /usr/local/bin/verilator_bin -cc --exe --top-module receiver -o receiver receiver.sv
```

catkira, 2023-02-19:
Debug output
```
benja@DESKTOP:/mnt/d/git/5G_SSB_sync/hdl$ perl /usr/local/bin/verilator -cc --exe --top-module receiver  -o receiver receiver.sv --debug
Starting Verilator 4.106 2020-12-02 rev v4.106 (mod)
- Verilator.cpp:539:  Option --verilate: Start Verilation
- V3File.cpp:234:        --check-times failed: different command line
- V3ParseImp.cpp:271: parseFile: receiver
  Preprocessing receiver.sv
- V3PreShell.cpp:147:     Reading receiver.sv
- V3ParseImp.cpp:320: Lexing receiver.sv
- V3LinkCells.cpp:201:Link Module: MODULE 0x55555636ade0 <e3#> {c3ai} u4=0x55555636af70  receiver  L0 [1ns]
- V3LinkCells.cpp:222:Link --top-module: MODULE 0x55555636ade0 <e3#> {c3ai} u1=0x555556363b20 u4=0x55555636af70  receiver  L0 [1ns]
dot -Tpdf -o ~/a.pdf obj_dir/Vreceiver_001_linkcells.dot
- V3LinkLevel.cpp:45: modSortByLevel()
- V3Ast.cpp:1138:     Dumping obj_dir/Vreceiver_002_cells.tree
- V3LinkDot.h:37:     linkDotPrimary:
- V3LinkJump.cpp:292: linkJump:
- V3Ast.cpp:1138:     Dumping obj_dir/Vreceiver_007_link.tree
- V3LinkInc.cpp:248:  linkIncrements:
- V3Ast.cpp:1138:     Dumping obj_dir/Vreceiver_008_linkInc.tree
- V3Param.cpp:1106:   param:
- V3LinkDot.h:42:     linkDotParamed:
- V3Ast.cpp:1138:     Dumping obj_dir/Vreceiver_010_paramlink.tree
- V3Dead.cpp:471:     deadifyModules:
- V3Width.cpp:5980:   width:
- V3Ast.cpp:1138:     Dumping obj_dir/Vreceiver_012_width.tree
- V3Width.cpp:6023:   widthCommit:
- V3Const.cpp:2659:   constifyAllLive:
%Warning-WIDTHCONCAT: receiver.sv:22:57: More than a 8k bit replication is probably wrong: 4294967291
                                       : ... In instance receiver
-node: REPLICATE 0x555556372170 <e550> {c22cf} @dt=0x55555636a660@(G/w-5)
   22 |         m_axis_out_tdata <= {{(OUT_DW - REQUIRED_OUT_DW){1'b0}}, filter_result};
      |                                                         ^
                      ... Use "/* verilator lint_off WIDTHCONCAT */" and lint_on around source to disable this message.
```

catkira, 2023-02-19:
This is a workaround for the bug
```verilog
`timescale 1ns / 1ns

module receiver
#(
    parameter IN_DW = 32,
    parameter OUT_DW = 24
)
(
    input                                       clk_i,
    output  reg            [OUT_DW-1:0]         m_axis_out_tdata
);

localparam REQUIRED_OUT_DW = OUT_DW + 5;
reg [REQUIRED_OUT_DW - 1: 0] filter_result;
localparam PAD_BITS = REQUIRED_OUT_DW >= OUT_DW ? 0 : OUT_DW - REQUIRED_OUT_DW;

always @(posedge clk_i) begin
    if (REQUIRED_OUT_DW >= OUT_DW) begin
        // take some MSBs
        m_axis_out_tdata <= filter_result[REQUIRED_OUT_DW - 1 -: OUT_DW];
    end else begin
        // do zero padding
        m_axis_out_tdata <= {{(PAD_BITS){1'b0}}, filter_result};
    end
end
endmodule
```

wsnyder, 2023-02-23:
Your code before workaround has effectively "{24-29}{1'b0}", and since concatenate sizes are unsigned this is a huge vector, which causes the crash.  The way you worked around it, or similar, I believe is required by IEEE.
That the warning wasn't given was a bug, this I just fixed and made it an error as it's clearly a mistake being so large.

closed 2023-02-23 by wsnyder