verilator/verilator #3964: Verilator is truncating large untyped parameter

catkira, 2023-02-19:
I have a large untyped parameter with this value:
```verilog
727692769084738296199514023252920695673110030442313862546494873936543905495465112115440116815497188759705501197535815646587584004276996791752064390383868594373468281020086948060874236171013709961163610340379507171482337110374530681613783202145013092788241815937210870966018139227097764872124474751641815788946053535112477087657136312458412346464311882492009223025761124892768175816113081836351703734815811808091866746647126345392212546930379911037828830015901316964722094413997851972518098152588844953943989631956252982179713758548699641252079762660878419761048466354457474404621561351873599525730175985582816209389093268802777458896604049172997098801344634666255667809975101223655006143928299719459732650385256333085359378423915656658073936416189406066453060753211327862638631082429491463981174442406347800928707185018063163844717919172077760369562510766221642403859597949340448739407881830030591052787606094267650225809326781620844915882640690415553409790547109072354439213916437092450968553340222620544271455309910007978982204458776367926065672863081889771206329426079874861760146265338141627687895347081066879697688631730952361555541164239652950942434867989007569712233445450099054556358665324201755063801958547716130561744500874
```
it gets truncated to
```verilog
32317006071311007300714876688669951960444102669715484032130345427524655138867890893197201411522913463688717960921898019494119559150490921095088152386448283120630877367300996091750197750389652106796057638384067568276792218642619756161838094338476170470581645852036305042887575891541065808607552399123930385521914333389668342420684974786564569494856176035326322058077805659331026192708460314150258592864177116725943603718461857357598351152301645904403697613233287231227125684710820209725157101726931323469678542580656697935045997268352998638215525166389437335543602135433229604645318478604952148193555853611059596230655
```
I use verilator v5.006

wsnyder, 2023-02-19:
Did Verilatoe give you a WIDTH warning?  IEEE says these may be limited.  Try it with a size.  See if the C++ is correct.

catkira, 2023-02-20:
Verilator did not give a WIDTH warning.
I will dive deeper into this after we have handled my other issue https://github.com/verilator/verilator/issues/3963

None of my issues are reallly urgent since I have work arounds, my brain is just too small to dive into multiple issues at the same time ^^

wsnyder, 2023-02-23:
I think recompiling with VL_VALUE_STRING_MAX_WORDS 256 might fix this, please give it a try.

catkira, 2023-02-27:
should I do `VALUE_MAX_STRING_MAX_WORDS=256 ./configure` ?

wsnyder, 2023-02-27:
It isn't a configure switch but in the C code.  Either edit the define in verilatedos.h, or compile your code with that defined, or verilate you model with "-CFLAGS -DVL_VALUE_STRING_MAX_WORDS=256"

catkira, 2023-02-27:
Verilating my files like this does not help
```
INFO cocotb: ccache g++  -std=c++11 -I.  -MMD -I/usr/local/share/verilator/include -I/usr/local/share/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=1 -DVM_TRACE_FST=1 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_VALUE_STRING_MAX_WORDS=256   -Os -c -o verilator.o /mnt/d/git/cocotb/cocotb/share/lib/verilator/verilator.cpp
```

wsnyder, 2023-02-28:
I don't see how that message could be printed with a proper value of VL_VALUE_STRING_MAX_WORDS.  Maybe it isn't getting applied.  See if the error message is printing the right value.  See if changing the verilatedos.h code directly to the new value works.

Or ideally, perhaps contribute a patch that used a dynamic string or something so this define could be eliminated.

wsnyder, 2023-04-11:
Assuming this got sorted out, if not more info or a patch to make dynamic described above would be appreciated.  Feel free to follow up here if needed.

closed 2023-04-11 by wsnyder