verilator/verilator #3965: IGNOREDRETURN - is not disabled with -Wno-lint option

svenka3, 2023-02-20:
Hello,
  I see -Wno-lint does not disable IGNOREDRETURN warning.

```sh
verilator --sc   ../soc/riscv_soc --Mdir verilated -I./../soc -I../core/core/rv32imsu -I../soc --pins-sc-uint -Wno-lint --trace
```
Log:
``` log
%Warning-IGNOREDRETURN: ../core/core/rv32imsu/riscv_decode.v:447:15: Ignoring return value of non-void function (IEEE 1800-2017 13.4.1)
                                                                   : ... In instance riscv_soc.u_core.u_core.u_decode
  447 |     u_regfile.set_register(r,value);
``` 
Is this a known issue/limitation? 

Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)

```verilog
module m;

  function bit test_bit ();
  endfunction : test_bit
  /* THISWORKS-verilator lint_off IGNOREDRETURN */
  initial test_bit();

endmodule : m

```
What 'verilator --version' are you using?  Did you try it with the git master version?

> Verilator 5.007 devel rev v5.006-7-g4a8cfe367

What OS and distribution are you using?

> WSL

May we assist you in trying to fix this in Verilator yourself?

> Yes, please

wsnyder, 2023-02-20:
This is a IEEE mandated warning, not just a lint warning. Use -Wno--IGNOREDRETURN.

wsnyder, 2023-02-20:
And, instead of disabling you really should instead have the author fix the bad code, use void'test_bit()

closed 2023-02-20 by wsnyder