verilator/verilator #3968: error disable isn't underneath a begin with name

gkamendje, 2023-02-21:
Using Verilator 5.006 on Ubuntu 20.04, and trying to verilate the following code
```
always begin    
fork
    begin: u_check_1
        #delay_1;
        #1 disable u_check_2;
        fork
            begin: l_check_1
                #delay_2;
                #1 disable l_check_2;
            end
            begin: l_check_2
                @(posedge clk);
                #1 disable l_check_1;
            end
        join 
    end
    
    begin: u_check_2
        @(posedge clk);
        #1 disable u_check_1;
    end
join
end 
```
I get the following error messages. 
```
%Error: ../../ver/th/test.v:119:12: disable isn't underneath a begin with name: 'u_check_2'
  119 |         #1 disable u_check_2;
      |            ^~~~~~~
%Error: ../../ver/th/test.v:123:20: disable isn't underneath a begin with name: 'l_check_2'
  123 |                 #1 disable l_check_2;
      |                    ^~~~~~~
%Error: ../../ver/th/test.v:127:20: disable isn't underneath a begin with name: 'l_check_1'
  127 |                 #1 disable l_check_1;
      |                    ^~~~~~~
%Error: ../../ver/th/test.v:134:12: disable isn't underneath a begin with name: 'u_check_1'
  134 |         #1 disable u_check_1;

```
Which seems to point out that Verilator only accept `disable` statements when the target named block is the current named block. I think that this is  incorrect. I looked at the LRM but I could not find anything suggesting that it should be restricted this way.

wsnyder, 2023-02-21:
Correct, verilator does not support disables in this way.  This is likely complicated to support so likely will not be improved soon, unless you want to contribute it.  Meanwhile, the error should be fixed to say UNSUPPORTED.

gkamendje, 2023-02-21:
I would definitely like to contribute. However, I will not be able to work on this until mid June or later. Let's close this for now and re-open later if required.

closed 2023-02-21 by gkamendje