verilator/verilator #3972: timescale option not working.

minired2154, 2023-02-22:
I currently work on Ubuntu 20.04/WSL2. And I follow the package installation guide which verilator version is 4.038.

I use the following command to make the testbench `verilator adder.v --exe --build testbench_adder.cpp -cc --trace --timescale 1ps/1ps`. After execution, the time unit of my .vcd file is at 10s. I also try the  `--timescale-override 1ps/1ps`, but it's still not working. Also, I don declare any `timescale` directives in my .v file. Do I do anything wrong or miss understanding something?

wsnyder, 2023-02-22:
That is a very old version, please try the current stable version.  I believe we have tests that check this case (t_trace_timescale), so if it still fails please attach an example.

minired2154, 2023-02-23:
It's worked. Sorry for asking such a dumb question. Thank you so much.

closed 2023-02-23 by minired2154