verilator/verilator #3973: max delay # seems to be limited to 32bit

xdch47, 2023-02-22:
Using Verilator 5.007 devel rev v5.006-101-g47a7e758

```
`timescale 1ns/1ps

module time();

	initial begin
		#5ms;
		$display("finish at %t", $realtime);
		$finish();
	end

endmodule
```
results in `finish at            705032704`
but 5000000000 would have been expected. Looks like uint32 - overflow `5e9 % 2**32 == 705032704`

wsnyder, 2023-02-22:
Huh, the math should have been 64-bit, but...   Would you be willing to make this into a test_regress test and attempt a fix?

xdch47, 2023-02-23:
hi,
test_regress should be quite straight forward (the example above is fine, right?) .
however, for the delay schedule a pointer to the sources would be helpful...

wsnyder, 2023-02-23:
Similar to the example, yes, need standard header and All Finished message.

Run with --debug and look at .tree files to see where the number goes wrong, there are several possibilities, might parse bad in verilog.y, or maybe V3Width.

xdch47, 2023-02-28:
Closed by a3ff375ce7853a9e376ad73ac4795c54d79222e9

closed 2023-02-28 by xdch47