verilator/verilator #3975: Incrementation and decrementation double's expressions side-effects

kboronski-ant, 2023-02-22:
AST for accessing value seems to get cloned for read and write access for pre/post incremenatation and decrementation operations. This results in any side-effects happening twice, first when reading the value and then when writing it.

Here's a minimal example for reproducing this behavior:
```systemverilog
int i = 0;

function int postincrement_i;
    return i++;
endfunction

module t;
    initial begin
        int arr [1:0] = {0, 0};
        arr[postincrement_i()]++;
        $display("Value: %d", i);
    end
endmodule
```

The expected output is
```
Value:           1
```
but instead we get
```
Value:           2
```

Note, I wrote a separate function `postincrement_i` because incrementing `i` directly in this context is yet another issue (which I'm already solving at the moment).

I'm afraid that solving the underlying issue might require reworking the way references are being handled in AST (unless I missed something and it's actually possible to hold a reference an unnamed lvalue) or at least implementing `PreAdd`/`PreSub`/`PostAdd`/`PostSub` operators to be translated directly to C++.

I'm using the latest version of Verilator under Linux (Gentoo):
```
Verilator 5.007 devel rev v5.006-101-g47a7e7584
```

kboronski-ant, 2023-02-22:
Here's a patch I made FTTB that should cause Verilator to throw an error when statement with side-effects is used in context of incrementation or decrementation: https://github.com/verilator/verilator/pull/3976

wsnyder, 2023-05-03:
I looked at a wider warning for this, I'm thinking most (but not all) places there is a cloneTree we should check for side effects.  I haven't implemented that yet though.

wsnyder, 2025-09-09:
This example currently works on master; fixed earlier.  Note there are still many side effect issues, however those others that remain should give a SIDEEFFECT warning so it is obvious.