verilator/verilator #3981: Unable to compile generated C++ file with systemverilog with struct input

yodalee, 2023-02-23:
I think the problem is related to latest change to support systemverilog struct in verilator. I found it when implementing my RSA verilog project.

# The example to show this issue:

```systemverilog
typedef struct {
    logic [255:0] msg;
} StructType;

module TestMod (
    input logic clk,
    StructType in
);

logic [255:0] msg;

always_ff @( posedge clk ) begin
    msg <= in.msg;
end

endmodule
```

Execute following commands:
```shell
verilator --cc test.sv
make -C obj_dir -f Vtest.mk
```

# The error message:
I share the error message using the [gist link](https://gist.github.com/yodalee/7b2d96216f98f50870bc7577d6aca28c)

# What 'verilator --version' are you using?
I compile this case using the git master commit
Verilator 5.007 devel rev v5.006-101-g47a7e7584

# What OS and distribution are you using?
Archlinux kernel 6.1.8-arch1-1

May we assist you in trying to fix this in Verilator yourself?
Yes I think I can take a look into this. It seems that there are lack some of the implementation to  `VL_TO_STRING_W`

wsnyder, 2023-02-23:
```
Vtest___024unit__DepSet_h9b08f7a8__0.cpp: In function 'std::string VL_TO_STRING(const Vtest_StructType__struct1&)':
Vtest___024unit__DepSet_h9b08f7a8__0.cpp:13:42: error: invalid user-defined conversion from 'const VlWide<8>' to 'int' [-fpermissive]
   13 |     out += "'{msg:" + VL_TO_STRING_W(obj.msg);
```

This should be just VL_TO_STRING, though because it is a wide value I can see why it gets confused.  Can you try adding another templated function in verilated_types.h for VL_TO_STRING_W that calls VL_TO_STRING.  And add a test to the pull too.

yodalee, 2023-02-24:
Understood, I can implement this on weekend.

yodalee, 2023-02-24:
Adding template function VL_TO_STRING_W with argument `VlWide<T_Words>` can solve the error.
However, when I want to add the test, I found that there is already similar test case `t_struct_port.v`.
Do you think this issue should be fixed? or should I just use packed struct?

wsnyder, 2023-02-24:
Does your testcase pass on master?  If not a new test case would be appreciated.

yodalee, 2023-02-25:
My testcase can pass on master **if** I use `struct packed`.
Do you think it is needed to support non-packed struct?
----
Also, after solving the `VL_TO_STRING_W` issue, there is another one:
```
Vtest___024root__DepSet_hc07518e4__0__Slow.cpp: In function 'void Vtest___024root___ctor_var_reset(Vtest___024root*)':
Vtest___024root__DepSet_hc07518e4__0__Slow.cpp:63:37: error: request for member 'msg' in 'vlSelf->Vtest___024root::in', which is of non-class type 'CData' {aka 'unsigned char'}
   63 |     VL_RAND_RESET_W(256, vlSelf->in.msg);
      |                                     ^~~
```
It seems that verilator consider `in.msg` to be a `VlWide<>` so it uses `VL_RAND_RESET_W`. However the actually type of `in.msg` is `CData`, which causes this error.

wsnyder, 2023-02-25:
For primary IO only struct packed is supported. Internal to a design either.

yodalee, 2023-02-25:
Then I think this issue can be closed since it is not a `struct packed`.

closed 2023-02-25 by wsnyder