verilator/verilator #3983: "make test" fails on cmake_protect_lib starting with 93d50c449

ldoolitt, 2023-02-24:
Under stock Debian (pre-)Bookworm,
`git clean -fdx && autoconf && ./configure --prefix=$HOME/test && make -j 6 && make test`
output ends with
```
make[1]: Entering directory '/redacted/verilator/examples/cmake_protect_lib'

-- Verilator CMake protect_lib example
[...]
CMake Error at /redacted/verilator/verilator-config.cmake:244 (message):
  Verilator command failed (return code=Segmentation fault)
Call Stack (most recent call first):
  CMakeLists.txt:58 (verilate)
```
starting at commit 93d50c449.
That's a big commit, and I have no idea how it triggers the failure.
The error message steers me to a CMakeError.log file, and the fault looks like
```
Building CXX object CMakeFiles/cmTC_85bd6.dir/src.cxx.o
/usr/bin/c++ -D_mt   -o CMakeFiles/cmTC_85bd6.dir/src.cxx.o -c /redacted/verilator/examples/cmake_protect_lib/build/CMakeFiles/CMakeScratch/TryCompile-buqlpY/src.cxx
Linking CXX executable cmTC_85bd6
/usr/bin/cmake -E cmake_link_script CMakeFiles/cmTC_85bd6.dir/link.txt --verbose=1
/usr/bin/c++ CMakeFiles/cmTC_85bd6.dir/src.cxx.o -o cmTC_85bd6  -mt 
c++: error: unrecognized command-line option ‘-mt’; did you mean ‘-MT’?
gmake[3]: *** [CMakeFiles/cmTC_85bd6.dir/build.make:99: cmTC_85bd6] Error 1
gmake[3]: Leaving directory '/redacted/verilator/examples/cmake_protect_lib/build/CMakeFiles/CMakeScratch/TryCompile-buqlpY'
gmake[2]: *** [Makefile:127: cmTC_85bd6/fast] Error 2
gmake[2]: Leaving directory '/redacted/verilator/examples/cmake_protect_lib/build/CMakeFiles/CMakeScratch/TryCompile-buqlpY'
```
I hope this helps.  LMK if more info is needed.
Oh, c++ in this case is `c++ (Debian 12.2.0-14) 12.2.0`

ldoolitt, 2023-02-24:
I don't see a lot of "-mt" strings floating around this code base.  The config.log file includes
```
configure:8426: checking whether g++ linker accepts -mt
[...]
configure:8452: result: no
```
which is good.  But then I do see
```
S["CFG_LDFLAGS_THREADS_CMAKE"]=" -mt -pthread -lpthread -latomic"
```
show up in config.status.  Mysterious.

ldoolitt, 2023-02-24:
Running this under gdb by hand gives
```
(No debugging symbols found in /redacted/verilator/bin/verilator_bin)
[...]
Program received signal SIGSEGV, Segmentation fault.
0x000055555585aa1d in AstNode::unlinkFrBack(VNRelinker*) ()
(gdb) bt
#0  0x000055555585aa1d in AstNode::unlinkFrBack(VNRelinker*) ()
#1  0x0000555555d167fb in V3LinkLevel::modSortByLevel() ()
#2  0x000055555581319d in process() ()
#3  0x000055555580dda3 in main ()
(gdb) 
```
I tried to reproduce with verilator_bin_dbg, but that run does _not_ segfault.

wsnyder, 2023-02-24:
Try --debug --no-debug-leak to see if that crashes.

ldoolitt, 2023-02-25:
```Starting Verilator 5.007 devel rev v5.006-73-g93d50c449
- Verilator.cpp:587:  Option --verilate: Start Verilation
- V3File.cpp:234:        --check-times failed: different command line
- V3ParseImp.cpp:276: parseFile: verilated_std
  Preprocessing /redacted/verilator/include/verilated_std.sv
- V3PreShell.cpp:141:     Reading /redacted/verilator/include/verilated_std.sv
- V3ParseImp.cpp:325: Lexing /redacted/verilator/include/verilated_std.sv
- V3ParseImp.cpp:276: parseFile: top
  Preprocessing ../make_protect_lib/top.v
- V3PreShell.cpp:141:     Reading ../make_protect_lib/top.v
- V3ParseImp.cpp:325: Lexing ../make_protect_lib/top.v
- V3ParseImp.cpp:276: parseFile: verilated_secret [LIB]
  Preprocessing verilated_secret
- V3PreShell.cpp:141:     Reading /redacted/verilator/examples/cmake_protect_lib/build/verilated_secret/verilated_secret.sv
- V3ParseImp.cpp:325: Lexing verilated_secret
- V3LinkLevel.cpp:48: modSortByLevel()
Segmentation fault
```
Same thing with or without `--no-debug-leak`

ldoolitt, 2023-02-25:
Sorry, my cut-and-paste left off the
```
Starting Verilator 5.007 devel rev v5.006-73-g93d50c449
```

ldoolitt, 2023-02-25:
I'm less than an expert at tracing the internals of verilator, although I do have about 50 years practice debugging C programs in general.  I convinced myself the segfault happens in
```
for (AstNodeModule* nodep : mods) nodep->unlinkFrBack();
```
specifically its second time through the loop, processing `verilated_secret`.

@gezalore, you're the last person to touch this line of code.  Any ideas?

I suppose memory could have been corrupted earlier in the process, and it just happens to get caught here.
Do people ever run verilator under valgrind?

wsnyder, 2023-02-25:
Almost certainly something corrupted earlier, e.h. something is getting deleted twice.
All the 22.04 github action regressions should be running with the address sanitizer, which catches most valgrind like issues.

ldoolitt, 2023-02-26:
I think I was able to turn on the address sanitizer for `verilator_bin` itself by adding `LDFLAGS="-fsanitize=address,undefined"` to the configure step. Then getting to this bug involved a bit of hacking
(like `mkdir -p  /redacted/verilator/examples/cmake_protect_lib/build/CMakeFiles/example.dir/Vtop.dir`).
New output is almost the same, maybe a smidgen more info:
```
$ VERILATOR_ROOT=$HOME/git/verilator /redacted/verilator/bin/verilator_bin --debug --debugi-V3LinkLevel 9 --compiler gcc --prefix Vtop --Mdir /redacted/verilator/examples/cmake_protect_lib/build/CMakeFiles/example.dir/Vtop.dir --make cmake --cc -I/redacted/verilator/examples/cmake_protect_lib/build/verilated_secret ../make_protect_lib/top.v
Starting Verilator 5.007 devel rev v5.006-73-g93d50c449 (mod)
- Verilator.cpp:587:  Option --verilate: Start Verilation
- V3File.cpp:218:        --check-times failed: no input /redacted/verilator/examples/cmake_protect_lib/build/CMakeFiles/example.dir/Vtop.dir/Vtop__verFiles.dat
- V3ParseImp.cpp:276: parseFile: verilated_std
  Preprocessing /redacted/verilator/include/verilated_std.sv
- V3PreShell.cpp:141:     Reading /redacted/verilator/include/verilated_std.sv
- V3ParseImp.cpp:325: Lexing /redacted/verilator/include/verilated_std.sv
- V3ParseImp.cpp:276: parseFile: top
  Preprocessing ../make_protect_lib/top.v
- V3PreShell.cpp:141:     Reading ../make_protect_lib/top.v
- V3ParseImp.cpp:325: Lexing ../make_protect_lib/top.v
- V3ParseImp.cpp:276: parseFile: verilated_secret [LIB]
  Preprocessing verilated_secret
- V3PreShell.cpp:141:     Reading /redacted/verilator/examples/cmake_protect_lib/build/verilated_secret/verilated_secret.sv
- V3ParseImp.cpp:325: Lexing verilated_secret
- V3LinkLevel.cpp:48: modSortByLevel()
- V3LinkLevel.cpp:81: modSortByLevel() loop2 done
- V3LinkLevel.cpp:87: modSortByLevel() sorted
- V3LinkLevel.cpp:89: modSortByLevel() nodep iteration
MODULE 0x612000000dc0 {d9ai}  top  L2 [1ps]- V3LinkLevel.cpp:89: modSortByLevel() nodep iteration
MODULE 0x6120000010c0 {e6ai}  verilated_secret  L3 [LIB] [1ps]- V3LinkLevel.cpp:89: modSortByLevel() nodep iteration
AddressSanitizer:DEADLYSIGNAL
=================================================================
==4729==ERROR: AddressSanitizer: SEGV on unknown address 0x000800000203 (pc 0x55e98e4ad400 bp 0x6020000001c0 sp 0x7ffca3e7fbe0 T0)
==4729==The signal is caused by a READ memory access.
    #0 0x55e98e4ad400 in V3LinkLevel::modSortByLevel() (/redacted/verilator/bin/verilator_bin+0x826400)
    #1 0x55e98dfa9d8c in process() (/redacted/verilator/bin/verilator_bin+0x322d8c)
    #2 0x55e98dfa4992 in main (/redacted/verilator/bin/verilator_bin+0x31d992)
    #3 0x7f655a246189 in __libc_start_call_main ../sysdeps/nptl/libc_start_call_main.h:58
    #4 0x7f655a246244 in __libc_start_main_impl ../csu/libc-start.c:381
    #5 0x55e98dfa9010 in _start (/redacted/verilator/bin/verilator_bin+0x322010)

AddressSanitizer can not provide additional info.
SUMMARY: AddressSanitizer: SEGV (/redacted/verilator/bin/verilator_bin+0x826400) in V3LinkLevel::modSortByLevel()
==4729==ABORTING
```
You'll see a couple more messages from UINFO() above, because I added debug lines to `V3LinkLevel.cpp`:
```
diff --git a/src/V3LinkLevel.cpp b/src/V3LinkLevel.cpp
index 9718b9b67..72f19318f 100644
--- a/src/V3LinkLevel.cpp
+++ b/src/V3LinkLevel.cpp
@@ -78,13 +78,19 @@ void V3LinkLevel::modSortByLevel() {
                                        << warnTopModules(secp->warnMore(), tops));
         }
     }
+    UINFO(9, "modSortByLevel() loop2 done\n");  // Comment required?
 
     timescaling(mods);
 
     // Reorder the netlist's modules to have modules in level sorted order
     stable_sort(mods.begin(), mods.end(), CmpLevel());  // Sort the vector
     UINFO(9, "modSortByLevel() sorted\n");  // Comment required for gcc4.6.3 / bug666
-    for (AstNodeModule* nodep : mods) nodep->unlinkFrBack();
+    for (AstNodeModule* nodep : mods) {
+        UINFO(9, "modSortByLevel() nodep iteration\n");  // Comment required?
+        cout << nodep;
+        nodep->unlinkFrBack();
+    }
+    UINFO(9, "modSortByLevel() nodep loop done\n");  // Comment required for gcc4.6.3 / bug666
     UASSERT_OBJ(!v3Global.rootp()->modulesp(), v3Global.rootp(), "Unlink didn't work");
     for (AstNodeModule* nodep : mods) v3Global.rootp()->addModulesp(nodep);
     UINFO(9, "modSortByLevel() done\n");  // Comment required for gcc4.6.3 / bug666
```
But, no evidence to support the memory corruption hypothesis.
Again, switching to `verilator_bin_dbg` succeeds, as in gives no segfault.  That approach also finds and reports
```
SUMMARY: AddressSanitizer: 200 byte(s) leaked in 5 allocation(s).
```
which I think is expected.

I'm out of ideas.
I'll also be away from my normal Bookworm machine for a week.  I can still probably run experiments, but it will be on a machine with 1/4 the single-core MIPS, 1/4 the cores, 1/4 the RAM, and 1/4 the pixels.  :-(

ldoolitt, 2023-02-27:
Using UINFO like printf, I could detect that a
```
for (AstNodeModule* nodep : mods) {}
```
loop works fine (enumerates `top` and `verilated_secret`) before starting the `nodep->unlinkFrBack()` function calls.
But when you insert that call into the loop, you end up with nodep == nullptr on the second loop iteration, and unlinkFrBack crashes on that.
I can supply patches and traces if anyone cares.

ldoolitt, 2023-03-20:
Fixed by commit d60a3e2b (Fix random internal crashes).

closed 2023-03-20 by ldoolitt