verilator/verilator #3984: Internal Erorr: Interface without tracing reference

rapiz1, 2023-02-24:
I'm using Verilator 5.006 from Arch Linux repo

To reproduce:
```
// b.sv
interface AXI_BUS;
  logic         aw_id;
endinterface

module test(input id_i, output id_o);	
  AXI_BUS m_axi;
  assign m_axi.aw_id = id_i;
  assign id_o = m_axi.aw_id;
endmodule
```
```
verilator --trace --binary b.sv -O0
```

Note that `-O0` is necessary to produce this internal error.

rapiz1, 2023-02-24:
Just realized that I may misunderstand what `-O0` means. What I intended to do is turn off the GCC optimization. `-O0` seems to turn off verilator optimization for models. However, I suppose an internal error should not be present to users

wsnyder, 2023-02-26:
"-mno-inline" seems sufficient to cause this. Needs more debug.

wsnyder, 2023-02-26:
"-fno-inline" seems sufficient to cause this. Needs more debug.

closed 2023-10-03 by wsnyder