verilator/verilator #3985: cmake linker error on macos ventura (M2)

hyperswine, 2023-02-26:
**Can you attach an example that shows the issue?  (Must be openly licensed, ideally in test_regress format.)**
Its the [cmake example starter](https://veripool.org/guide/latest/verilating.html#cmake) for verilog and verilator.

The error:
```
[build] FAILED: Vour 
[build] : && /opt/homebrew/bin/aarch64-apple-darwin22-g++-12 -g -arch arm64 -isysroot /Library/Developer/CommandLineTools/SDKs/MacOSX13.1.sdk -Wl,-search_paths_first -Wl,-headerpad_max_install_names  CMakeFiles/Vour.dir/sim_main.cpp.o CMakeFiles/Vour.dir/CMakeFiles/Vour.dir/Vour.dir/Vour.cpp.o CMakeFiles/Vour.dir/CMakeFiles/Vour.dir/Vour.dir/Vour___024root__DepSet_h63d63036__0.cpp.o CMakeFiles/Vour.dir/CMakeFiles/Vour.dir/Vour.dir/Vour___024root__DepSet_hd6c43f43__0.cpp.o CMakeFiles/Vour.dir/CMakeFiles/Vour.dir/Vour.dir/Vour___024root__Slow.cpp.o CMakeFiles/Vour.dir/CMakeFiles/Vour.dir/Vour.dir/Vour___024root__DepSet_hd6c43f43__0__Slow.cpp.o CMakeFiles/Vour.dir/CMakeFiles/Vour.dir/Vour.dir/Vour__Dpi.cpp.o CMakeFiles/Vour.dir/CMakeFiles/Vour.dir/Vour.dir/Vour__Syms.cpp.o CMakeFiles/Vour.dir/opt/homebrew/share/verilator/include/verilated.cpp.o CMakeFiles/Vour.dir/opt/homebrew/share/verilator/include/verilated_dpi.cpp.o CMakeFiles/Vour.dir/opt/homebrew/share/verilator/include/verilated_threads.cpp.o -o Vour  -pthread  -lpthread  -latomic && :
[build] Undefined symbols for architecture arm64:
[build]   "__ZN12VerilatedVpi12callValueCbsEv", referenced from:
[build]       _main in sim_main.cpp.o
[build]   "__ZTHN9Verilated3t_sE", referenced from:
[build]       __ZTWN9Verilated3t_sE in sim_main.cpp.o
[build]       __ZTWN9Verilated3t_sE in Vour.cpp.o
[build]       __ZTWN9Verilated3t_sE in verilated_dpi.cpp.o
[build]       __ZTWN9Verilated3t_sE in verilated_threads.cpp.o
[build]   "_vpi_get_str", referenced from:
[build]       __Z14read_and_checkv in sim_main.cpp.o
[build]   "_vpi_get_value", referenced from:
[build]       __Z14read_and_checkv in sim_main.cpp.o
[build]   "_vpi_handle_by_name", referenced from:
[build]       __Z14read_and_checkv in sim_main.cpp.o
[build] ld: symbol(s) not found for architecture arm64
[build] collect2: error: ld returned 1 exit status
[build] ninja: build stopped: subcommand failed.
[proc] The command: /opt/homebrew/bin/cmake --build /Users/jasonqin/Documents/GitHub/Playground/systemverilog/build --config Debug --target Vour -- exited with code: 1
[build] Build finished with exit code 1
```

I have tried clang 14 as well, and it also had a similar linker error.

**What 'verilator --version' are you using?  Did you try it with the git master version?**
`Verilator 5.006 2023-01-22 rev UNKNOWN.REV`
I installed it with `brew install verilator`

**What OS and distribution are you using?**
MacOS Ventura 13.2.1

wsnyder, 2023-02-26:
There were some recent fixes in this, please try the github version e.g. #3979 , please see if that works.

wsnyder, 2023-04-08:
Assuming this is now working for you, if not we'll need more info.

closed 2023-04-08 by wsnyder