verilator/verilator #3986: What is the minimum gcc version required to compile Verilator from source?

TallAllie, 2023-02-27:
Compiling recent Verilator versions (5.006, 4.228) with gcc-6.4.0 yields a handful of errors like this:
```
In file included from ../V3Assert.h:23:0,
                 from ../Verilator.cpp:19:
../V3Ast.h:2375:33: error: specialization of ‘template<class _Tp> struct std::hash’ in different namespace [-fpermissive]
 struct std::hash<VNRef<T_Node>> final {
                                 ^~~~~
In file included from /tools/gnu/gcc-6.4.0/include/c++/6.4.0/bits/basic_string.h:5643:0,
                 from /tools/gnu/gcc-6.4.0/include/c++/6.4.0/string:52,
                 from /tools/gnu/gcc-6.4.0/include/c++/6.4.0/bits/locale_classes.h:40,
                 from /tools/gnu/gcc-6.4.0/include/c++/6.4.0/bits/ios_base.h:41,
                 from /tools/gnu/gcc-6.4.0/include/c++/6.4.0/ios:42,
                 from /tools/gnu/gcc-6.4.0/include/c++/6.4.0/ostream:38,
                 from /tools/gnu/gcc-6.4.0/include/c++/6.4.0/iostream:39,
                 from ../config_build.h:65,
                 from ../V3Active.h:20,
                 from ../Verilator.cpp:17:
/tools/gnu/gcc-6.4.0/include/c++/6.4.0/bits/functional_hash.h:58:12: error:   from definition of ‘template<class _Tp> struct std::hash’ [-fpermissive]
     struct hash;
            ^~~~
...
```

TallAllie, 2023-02-27:
Ah, this seems to be a gcc-6 bug ...
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=56480

wsnyder, 2023-02-27:
Note we regress with the GCC in Ubuntu 20.04 and 22.04, any patches to fix other issues to keep it C++14 compliant or work around bugs that are not overly intrusive to fix are welcome.  Please submit a pull request as needed.

Closing as answered, feel free to followup here if needed.

closed 2023-02-27 by wsnyder