verilator/verilator #3988: Conflicting pull directions on different signals

algrant-arm, 2023-02-28:
```
module M;
  wire [1:0] PIN;
  pullup(PIN[0]);
  pulldown(PIN[1]);
endmodule
```
reports
```
%Error-UNSUPPORTED: a.v:4:11: Unsupported: Conflicting pull directions.
                            : ... In instance M
    4 |   pulldown(PIN[1]);
      |           ^
                    a.v:3:9: ... Location of conflicting pull.
    3 |   pullup(PIN[0]);
      |         ^
                    ... For error description see https://verilator.org/warn/UNSUPPORTED?v=5.007

```
Appreciate it says UNSUPPORTED, but the message says that conflicting pull directions are unsupported, when there are no conflicting pull directions here as they are on different pins. Is it unsupported to pull a specific pin or range of pins?

algrant-arm, 2023-02-28:
I guess this might be another manifestation of #364 (BLKNANDNBLK reported for different bits in a vector).

wsnyder, 2023-02-28:
All bits in PIN, to Verilator, are the "same signal".    It's possible this could be fixed in V3Tristate to track individual bits, if you would like to submit a patch to do so.  Alternatively, try "wire [1:0] PIN  /*verilator split_var*/;"

Closing as answered but feel free to follow up.

algrant-arm, 2023-03-06:
If this is a non-compliance and a fix would be welcomed, should the issue remain open?

closed 2026-10-18 by wsnyder