verilator/verilator #3992: Installed directory structure doesn't work

kcameron4456, 2023-03-01:
I want to install verilator in a local directory so I'm not always dependent on the build tree being available and built. Unfortunately, the installed directory tree doesn't work.

I checked out master (06661ab6760cc619ebf029783785774d1cc6d216) and built with:

make distclean
autoconf
./configure --prefix /home/kcameron/.local/verilator/master --enable-longtests --disable-defenv
make -j 9
rm -r ~/.local/verilator
make install

When I try to compile a simple test case, I get this:
$ setenv VERILATOR_ROOT /home/kcameron/.local/verilator/master
$ $VERILATOR_ROOT/bin/verilator --timing -Mdir vlrobj_sz4/ --binary Test.v
%Error: Cannot find verilated_std.sv containing built-in std:: definitions:/home/kcameron/.local/verilator/master/include/verilated_std.sv
%Error: This may be because there's no search path specified with -I<dir>.
        ... Looked in:
             /home/kcameron/.local/verilator/master/include/verilated_std.sv
             /home/kcameron/.local/verilator/master/include/verilated_std.sv.v
             /home/kcameron/.local/verilator/master/include/verilated_std.sv.sv
             vlrobj_sz4///home/kcameron/.local/verilator/master/include/verilated_std.sv
             vlrobj_sz4///home/kcameron/.local/verilator/master/include/verilated_std.sv.v
             vlrobj_sz4///home/kcameron/.local/verilator/master/include/verilated_std.sv.sv
%Error: Exiting due to 2 error(s)
$ 

If I set VERILATOR_ROOT to the build dir, it works:
$ 
$ setenv VERILATOR_ROOT /home/kcameron/src/eda/verilator
$ $VERILATOR_ROOT/bin/verilator --timing -Mdir vlrobj_sz4/ --binary Test.v
make: Entering directory '/home/kcameron/projects/othello/verilog/bug3vlr/vlrobj_sz4'
ccache g++  -I.  -MMD -I/home/kcameron/src/eda/verilator/include -I/home/kcameron/src/eda/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_TIME_CONTEXT   -Os -c -o verilated.o /home/kcameron/src/eda/verilator/include/verilated.cpp
ccache g++  -I.  -MMD -I/home/kcameron/src/eda/verilator/include -I/home/kcameron/src/eda/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_TIME_CONTEXT   -Os -c -o verilated_threads.o /home/kcameron/src/eda/verilator/include/verilated_threads.cpp
/usr/bin/python3 /home/kcameron/src/eda/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VTest.cpp VTest___024root__DepSet_h755816ae__0.cpp VTest___024root__DepSet_h2d9f3804__0.cpp VTest__main.cpp VTest___024root__Slow.cpp VTest___024root__DepSet_h2d9f3804__0__Slow.cpp VTest__Syms.cpp > VTest__ALL.cpp
ccache g++  -I.  -MMD -I/home/kcameron/src/eda/verilator/include -I/home/kcameron/src/eda/verilator/include/vltstd -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fcf-protection=none -Wno-bool-operation -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow     -DVL_TIME_CONTEXT   -Os -c -o VTest__ALL.o VTest__ALL.cpp
echo "" > VTest__ALL.verilator_deplist.tmp
Archive ar -rcs VTest__ALL.a VTest__ALL.o
g++    verilated.o verilated_threads.o VTest__ALL.a    -pthread -lpthread -latomic   -o VTest
rm VTest__ALL.verilator_deplist.tmp
make: Leaving directory '/home/kcameron/projects/othello/verilog/bug3vlr/vlrobj_sz4'
$ 

Here's the installed directory tree:
$ 
$ find /home/kcameron/.local/verilator
/home/kcameron/.local/verilator
/home/kcameron/.local/verilator/master
/home/kcameron/.local/verilator/master/bin
/home/kcameron/.local/verilator/master/bin/verilator
/home/kcameron/.local/verilator/master/bin/verilator_coverage
/home/kcameron/.local/verilator/master/bin/verilator_gantt
/home/kcameron/.local/verilator/master/bin/verilator_profcfunc
/home/kcameron/.local/verilator/master/bin/verilator_bin
/home/kcameron/.local/verilator/master/bin/verilator_bin_dbg
/home/kcameron/.local/verilator/master/bin/verilator_coverage_bin_dbg
/home/kcameron/.local/verilator/master/share
/home/kcameron/.local/verilator/master/share/verilator
/home/kcameron/.local/verilator/master/share/verilator/bin
/home/kcameron/.local/verilator/master/share/verilator/bin/verilator_includer
/home/kcameron/.local/verilator/master/share/verilator/bin/verilator_ccache_report
/home/kcameron/.local/verilator/master/share/verilator/bin/verilator_difftree
/home/kcameron/.local/verilator/master/share/verilator/include
/home/kcameron/.local/verilator/master/share/verilator/include/gtkwave
/home/kcameron/.local/verilator/master/share/verilator/include/gtkwave/fastlz.c
/home/kcameron/.local/verilator/master/share/verilator/include/gtkwave/fastlz.h
/home/kcameron/.local/verilator/master/share/verilator/include/gtkwave/fstapi.c
/home/kcameron/.local/verilator/master/share/verilator/include/gtkwave/fstapi.h
/home/kcameron/.local/verilator/master/share/verilator/include/gtkwave/fst_config.h
/home/kcameron/.local/verilator/master/share/verilator/include/gtkwave/fst_win_unistd.h
/home/kcameron/.local/verilator/master/share/verilator/include/gtkwave/lz4.c
[... 89 lines cut ...]
/home/kcameron/.local/verilator/master/share/verilator/examples/cmake_tracing_sc/CMakeLists.txt
/home/kcameron/.local/verilator/master/share/verilator/examples/cmake_tracing_sc/Makefile
/home/kcameron/.local/verilator/master/share/verilator/examples/cmake_protect_lib
/home/kcameron/.local/verilator/master/share/verilator/examples/cmake_protect_lib/CMakeLists.txt
/home/kcameron/.local/verilator/master/share/verilator/examples/cmake_protect_lib/Makefile
/home/kcameron/.local/verilator/master/share/verilator/examples/xml_py
/home/kcameron/.local/verilator/master/share/verilator/examples/xml_py/sub.v
/home/kcameron/.local/verilator/master/share/verilator/examples/xml_py/top.v
/home/kcameron/.local/verilator/master/share/verilator/examples/xml_py/Makefile
/home/kcameron/.local/verilator/master/share/verilator/examples/xml_py/vl_file_copy
/home/kcameron/.local/verilator/master/share/verilator/examples/xml_py/vl_hier_graph
/home/kcameron/.local/verilator/master/share/verilator/verilator-config.cmake
/home/kcameron/.local/verilator/master/share/verilator/verilator-config-version.cmake
/home/kcameron/.local/verilator/master/share/man
/home/kcameron/.local/verilator/master/share/man/man1
/home/kcameron/.local/verilator/master/share/man/man1/verilator.1
/home/kcameron/.local/verilator/master/share/man/man1/verilator_coverage.1
/home/kcameron/.local/verilator/master/share/man/man1/verilator_gantt.1
/home/kcameron/.local/verilator/master/share/man/man1/verilator_profcfunc.1
/home/kcameron/.local/verilator/master/share/pkgconfig
/home/kcameron/.local/verilator/master/share/pkgconfig/verilator.pc
$ 

verilated_std.sv is present but it's under an addition layer of dirs (share/verilator) compared to the build tree.  The other include files seem to have the same issue.
v5.006 has a similar issue but with verilator_bin

xdch47, 2023-03-01:
Looks like verilator root has to be `pkgdatadir`.

So something like the following could do the job as kind of workaround. 
```
export PATH=/home/kcameron/.local/verilator/master/bin
export VERILATOR_ROOT=/home/kcameron/.local/verilator/master/share/verilator
```

```
diff --git a/bin/verilator b/bin/verilator
index 587a9b85f..519947259 100755
--- a/bin/verilator
+++ b/bin/verilator
@@ -139,23 +139,28 @@ sub debug {
 sub verilator_bin {
     my $bin = "";
     # Use VERILATOR_ROOT if defined, else assume verilator_bin is in the search path
+    #
     my $basename = ($ENV{VERILATOR_BIN}
                     || ($Debug ? "verilator_bin_dbg" : "verilator_bin"));
+
+    if (-x "$RealBin/$basename"
+        || -x "$RealBin/$basename.exe") {
+        $bin = "$RealBin/$basename";  # From path/to/verilator with verilator_bin installed
+    } else {
+        $bin = $basename;  # Find in PATH
+    }
+
     if (defined($ENV{VERILATOR_ROOT})) {
         my $dir = $ENV{VERILATOR_ROOT};
         if (-x "$dir/bin/$basename"
             || -x "$dir/bin/$basename.exe") {  # From a "make install" into VERILATOR_ROOT
             $bin = "$dir/bin/$basename";
         } else {
-            $bin = "$dir/$basename";  # From pointing to kit directory
+            if (-x"$dir/$basename") {
+                $bin = "$dir/$basename";  # From pointing to kit directory
+            }
         }
     } else {
-        if (-x "$RealBin/$basename"
-            || -x "$RealBin/$basename.exe") {
-            $bin = "$RealBin/$basename";  # From path/to/verilator with verilator_bin installed
-        } else {
-            $bin = $basename;  # Find in PATH
-        }
         # Note we don't look under bin/$basename which would be right if running
         # in the kit dir. Running that would likely break, since
         # VERILATOR_ROOT wouldn't be set and Verilator won't find internal files.
diff --git a/test_regress/driver.pl b/test_regress/driver.pl
index 2a1685e4e..63f4a4b53 100755
--- a/test_regress/driver.pl
+++ b/test_regress/driver.pl
@@ -875,7 +875,7 @@ sub compile_vlt_cmd {
     return 1 if $self->errors || $self->skips;
 
     my @vlt_cmd = (
-        "perl", "$ENV{VERILATOR_ROOT}/bin/verilator",
+        "verilator",
         $self->compile_vlt_flags(%param),
         $param{top_filename},
         @{$param{v_other_filenames}},
```

wsnyder, 2023-03-01:
You can build as you did above, but when install do not set VERILATOR_ROOT.  Alternatively do not "make install" at all, and instead point VERILATOR_ROOT at your build directory with sources.  These options are described in https://verilator.org/guide/latest/install.html

closed 2023-03-01 by wsnyder