verilator/verilator #3994: help2man runs error during compilation

LuJiuxi, 2023-03-01:
I compile the verilator on WSL2 ubuntu22.04LTS, but there is someting wrong:

``` shell
make[2]: Entering directory '/home/suyang/verilator/src/obj_opt'
      Compile flags:  g++ -O3 -MMD -I. -I.. -I.. -I../../include -I../../include -MP -faligned-new -Wno-unused-parameter -Wno-shadow -fno-builtin-malloc -fno-builtin-calloc -fno-builtin-realloc -fno-builtin-free -DDEFENV_SYSTEMC="" -DDEFENV_SYSTEMC_ARCH="" -DDEFENV_SYSTEMC_INCLUDE="" -DDEFENV_SYSTEMC_LIBDIR="" -DDEFENV_VERILATOR_ROOT="/usr/local/share/verilator"
make[2]: Leaving directory '/home/suyang/verilator/src/obj_opt'
make[1]: Leaving directory '/home/suyang/verilator/src'
help2man --no-info --no-discard-stderr --version-string=- bin/verilator_gantt -o verilator_gantt.1
help2man: can't get `--help' info from bin/verilator_gantt
make: *** [Makefile:191: verilator_gantt.1] Error 13
```
Does anyone know how to solve this problem？

wsnyder, 2023-03-01:
This works for me on WSL.  See what "bin/verilator_gantt --help" prints, maybe your python is broken.

LuJiuxi, 2023-03-03:
It is caused by running permissions and Windows encoding. Modify the operation permission and file code to solve the problem

closed 2023-03-03 by LuJiuxi