verilator/verilator #3998: Errors during `--coverage` run when using class

timkpaine, 2023-03-02:
## Repro
Example problem, adapted from the example repo here:
https://github.com/timkpaine/verilator-coverage-repro

If you uncomment `--coverage` in the makefile flags, you'll reproduce the error. Otherwise, the program will run exactly as expected.

Error:
```
clang++  -I.  -MMD -I/opt/homebrew/Cellar/verilator/5.006/share/verilator/include -I/opt/homebrew/Cellar/verilator/5.006/share/verilator/include/vltstd -DVM_COVERAGE=1 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++17 -fcoroutines-ts -Os -c -o verilated_threads.o /opt/homebrew/Cellar/verilator/5.006/share/verilator/include/verilated_threads.cpp
/usr/bin/python3 /opt/homebrew/Cellar/verilator/5.006/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include Vdut.cpp Vdut___024root__DepSet_hbd7b84b1__0.cpp Vdut___024root__DepSet_hf386fab0__0.cpp Vdut___024root__DepSet_hbf6d145a__0.cpp Vdut_ifc__DepSet_h26685362__0.cpp Vdut_ifc__DepSet_h4a5f660b__0.cpp Vdut___024unit__03a__03atransaction__Vclpkg__DepSet_hbfca68f4__0.cpp Vdut___024unit__03a__03atransaction__Vclpkg__DepSet_hc0bd7095__0.cpp Vdut___024root__Slow.cpp Vdut___024root__DepSet_hbd7b84b1__0__Slow.cpp Vdut___024root__DepSet_hbf6d145a__0__Slow.cpp Vdut___024unit__Slow.cpp Vdut___024unit__DepSet_h3a999569__0__Slow.cpp Vdut_ifc__Slow.cpp Vdut_ifc__DepSet_h26685362__0__Slow.cpp Vdut_ifc__DepSet_h4a5f660b__0__Slow.cpp Vdut___024unit__03a__03atransaction__Vclpkg__Slow.cpp Vdut___024unit__03a__03atransaction__Vclpkg__DepSet_hbfca68f4__0__Slow.cpp Vdut___024unit__03a__03atransaction__Vclpkg__DepSet_hc0bd7095__0__Slow.cpp Vdut__Syms.cpp > Vdut__ALL.cpp
clang++  -I.  -MMD -I/opt/homebrew/Cellar/verilator/5.006/share/verilator/include -I/opt/homebrew/Cellar/verilator/5.006/share/verilator/include/vltstd -DVM_COVERAGE=1 -DVM_SC=0 -DVM_TRACE=0 -DVM_TRACE_FST=0 -DVM_TRACE_VCD=0 -faligned-new -fbracket-depth=4096 -fcf-protection=none -Qunused-arguments -Wno-bool-operation -Wno-tautological-bitwise-compare -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-but-set-variable -Wno-unused-parameter -Wno-unused-variable -Wno-shadow      -std=gnu++17 -fcoroutines-ts -Os -c -o Vdut__ALL.o Vdut__ALL.cpp
In file included from Vdut__ALL.cpp:3:
In file included from ./Vdut.cpp:5:
In file included from ./Vdut__Syms.h:17:
In file included from ./Vdut___024root.h:10:
/opt/homebrew/Cellar/verilator/5.006/share/verilator/include/verilated_timing.h:35:36: warning: unknown warning group '-Wdeprecated-experimental-coroutine', ignored [-Wunknown-warning-option]
#  pragma clang diagnostic ignored "-Wdeprecated-experimental-coroutine"
                                   ^
In file included from Vdut__ALL.cpp:19:
./Vdut___024unit__03a__03atransaction__Vclpkg__Slow.cpp:50:21: error: use of undeclared identifier 'vlSymsp'
    VL_COVER_INSERT(vlSymsp->_vm_contextp__->coveragep(), count32p,  "filename",filenamep,  "lineno",lineno,  "column",column,
                    ^
./Vdut___024unit__03a__03atransaction__Vclpkg__Slow.cpp:50:21: error: use of undeclared identifier 'vlSymsp'
./Vdut___024unit__03a__03atransaction__Vclpkg__Slow.cpp:50:21: error: use of undeclared identifier 'vlSymsp'
./Vdut___024unit__03a__03atransaction__Vclpkg__Slow.cpp:50:5: error: use of undeclared identifier 'name'
    VL_COVER_INSERT(vlSymsp->_vm_contextp__->coveragep(), count32p,  "filename",filenamep,  "lineno",lineno,  "column",column,
    ^
/opt/homebrew/Cellar/verilator/5.006/share/verilator/include/verilated_cov.h:73:39: note: expanded from macro 'VL_COVER_INSERT'
        covcontextp->_insertp("hier", name(), __VA_ARGS__); \
                                      ^
In file included from Vdut__ALL.cpp:19:
./Vdut___024unit__03a__03atransaction__Vclpkg__Slow.cpp:51:28: error: use of undeclared identifier 'name'
        "hier",std::string{name()} + hierp,  "page",pagep,  "comment",commentp,  (linescovp[0] ? "linescov" : ""), linescovp);
                           ^
In file included from Vdut__ALL.cpp:20:
./Vdut___024unit__03a__03atransaction__Vclpkg__DepSet_hbfca68f4__0__Slow.cpp:15:31: error: use of undeclared identifier 'vlSymsp'
    vlSelf->__vlCoverInsert(&(vlSymsp->__Vcoverage[42]), first, "transaction.svh", 4, 19, "", "v_line/transaction", "block", "4-5");
                              ^
./Vdut___024unit__03a__03atransaction__Vclpkg__DepSet_hbfca68f4__0__Slow.cpp:16:31: error: use of undeclared identifier 'vlSymsp'
    vlSelf->__vlCoverInsert(&(vlSymsp->__Vcoverage[43]), first, "transaction.svh", 8, 19, "", "v_line/transaction", "block", "8-9");
                              ^
./Vdut___024unit__03a__03atransaction__Vclpkg__DepSet_hbfca68f4__0__Slow.cpp:17:31: error: use of undeclared identifier 'vlSymsp'
    vlSelf->__vlCoverInsert(&(vlSymsp->__Vcoverage[44]), first, "transaction.svh", 12, 18, "", "v_line/transaction", "block", "12-13");
                              ^
./Vdut___024unit__03a__03atransaction__Vclpkg__DepSet_hbfca68f4__0__Slow.cpp:18:31: error: use of undeclared identifier 'vlSymsp'
    vlSelf->__vlCoverInsert(&(vlSymsp->__Vcoverage[45]), first, "transaction.svh", 1, 1, "", "v_line/transaction", "block", "1");
                              ^
1 warning and 9 errors generated.
```

## Version
`Verilator 5.006 2023-01-22 rev UNKNOWN.REV`

## OS / Dist
MacOS

May we assist you in trying to fix this in Verilator yourself?
Absolutely

wsnyder, 2023-03-04:
I looked at this, and it is putting the coverage into the wrong generated class, so easy fix committed.  Also I disabled toggle coverage on member variables as that is hard and seems somewhat non-sensical to cover.

closed 2023-03-04 by wsnyder

timkpaine, 2023-03-04:
Oh wow amazing, thanks so much!