verilator/verilator #4002: Verilator version isn't properly updated when using worktrees

AndrewNolte, 2023-03-03:
In cmakelists.txt we have 
```
set(gitHead ${srcdir}/../.git/logs/HEAD)
if (NOT EXISTS ${githead})
    set(gitHead "")
endif()
add_custom_command(
    OUTPUT config_rev.h
    MAIN_DEPENDENCY ${gitHead}
    DEPENDS ${CONFIG_REV}
    COMMAND ${PYTHON3} ARGS
        ${CONFIG_REV} "${srcdir}" > "$<SHELL_PATH:${CMAKE_CURRENT_BINARY_DIR}/config_rev.h>"
)
```
But if you're operating in a worktree, the HEAD file is actually at ```<vltrepo>/.git/worktrees/<myworktree>/logs/HEAD``` And so this head commit id wont update in the config_rev.h. Could we maybe change this to recompute each time if it doesn't exist?

wsnyder, 2023-03-04:
Please feel free to propose a patch to fix this.  Note the cmake matches src/Makefile.in that works similarly, so please update both.

wsnyder, 2025-08-21:
Closing as not planned as this is a somewhat sophisticated developer configuration, open to getting a pull to resolve it.

closed 2025-08-21 by wsnyder