verilator/verilator #4003: Queue find method - support for "with item"

svenka3, 2023-03-05:
While bringing up SVUnit on Verilator we hit the below issue, we have a workaround for now. 

```sh
verilator -V
Verilator 5.007 devel rev v5.006-7-g4a8cfe367
```

What would you like added/supported?

Array query/find methods need support for "with item", e.g.

```verilog
    svunit_testsuite match[$] = list_of_suites.find() with (item.get_results() == PASS);
```

I get error as below:
```log
Vaf_vlt_issue_1___024root__DepSet_hf4869d12__0__Slow.cpp:20:19: error: ‘item’ was not declared in this scope
   20 |     VL_NULL_CHECK(item, "/home/srini/af_vlt_issue_1.sv", 14)->__VnoInFunc_get_results(vlSymsp, __Vtask_get_results__0__Vfuncout);
      |                   ^~~~
```

Can you attach an example that runs on other simulators?  (Must be openly licensed, ideally in test_regress format.)

```verilog
typedef enum bit {FAIL, PASS} results_t;
class svunit_testsuite;
  protected results_t success = PASS;

  function results_t get_results();
    return success;
  endfunction : get_results
endclass : svunit_testsuite

module top;
  svunit_testsuite list_of_suites[$];

  initial begin
    svunit_testsuite match[$] = list_of_suites.find() with (item.get_results() == PASS);
    $display ("%p", match);
  end
endmodule : top

```

May we assist you in trying to fix this in Verilator yourself?

Yes, will attempt.

wsnyder, 2023-03-05:
A pull would be appreciated, it should be straight forward.  t_with already tests some "queue with", but is missing the case of a function call.

There's some code inserted around function calls, I suspect V3Task is putting it outside the with() and needs to insert it under the AstWith.

wsnyder, 2023-07-08:
This example now passes, probably fixed earlier with 3979f8a8b

closed 2023-07-08 by wsnyder