verilator/verilator #4004: no Registers in dumped VCD waveform

sujay-pandit, 2023-03-06:
I am using Verilator 4.038 2020-07-11 rev v4.036-114-g0cd4a57ad on Ubuntu 22.04.2 LTS. 

I compiled Verilator with trace enabled for an open-source RISC-V core (https://gitlab.com/shaktiproject/cores/c-class) , my generated vcd there doesn't have any registers. In comparison, I get registers in my VCD when I generate it using VCS.  My generated VCD has no registers, so when I feed it to tools like Spyglass, my power numbers are way off.

I have a zip file containing Verilator and Vcs VCDs for the RISC-V core. (Shakti C-class)

[cclass_vcd.zip](https://github.com/verilator/verilator/files/10902862/cclass_vcd.zip)

wsnyder, 2023-03-06:
There should have been comments in the output file as to why something wasn't traced, but I see a recent optimization broke that.  I added a debug print to put it back, so please use the master version, run with "--debugi-V3TraceDecl 4" and see why it doesn't trace it.  Most likely it's a very wide vector or array not traced for speed reasons, see the various trace control options in the documentation please.

wsnyder, 2023-03-06:
Feel free to post followups if you have questions.

sujay-pandit, 2023-03-07:
Thanks for your help! 
Sorry, this is my bad; I should have been clearer. The problem isn't that registers aren't getting dumped. They are; they are just set written as wires. 

Below is the VCD snippet from VCS:

[image]

this is the VCD from Verilator:
[image]

Could this be an issue because the Verilog is generated from Bluespec SV (High level HDL)?
I faced the same issue while simulating Gemmini  (https://github.com/ucb-bar/gemmini) written in Chisel.

I have attached the generated Verilog, which contains the register in the image. 

[mkTbSoc.zip](https://github.com/verilator/verilator/files/10904008/mkTbSoc.zip)

wsnyder, 2023-03-07:
Verilator should put in "logic" if the signal is declared as reg/logic, so check your input file.  It won't say "reg" as that's pre-SystemVerilog.

wsnyder, 2023-03-07:
Verilator should put in "logic" if the signal is declared as reg/logic, so check your input SystemVerilog file (no idea what Bluesec does). It won't say "reg" as that's pre-SystemVerilog.

sujay-pandit, 2023-03-07:
Thanks! Bluespec compiler generates standard Verilog. Below is a snippet of the input generated Verilog code, on Line 60, "rg_read_rx" is declared as a reg. (example from the image above). So I should expect Verilator to generate reg/logic, right?

[image]

wsnyder, 2023-03-08:
Yes, it looks like the examples/make_tracing_c count_c likewise doesn't create a logic.  Can you dig in, and create a patch?

sujay-pandit, 2023-03-08:
Sure, I will try my best to do that. 
Can you point me to a file I should start with?

anmolsahoo25, 2023-03-08:
Can you test #4011 with c-class?

wsnyder, 2023-08-30:
Fixed earlier.

closed 2023-08-30 by wsnyder