verilator/verilator #4006: Help understanding poor multithreaded scaling

shareefj, 2023-03-07:
Hi,

we're using Verilator on a fairly small MCU design and I've been looking, or rather trying to look, at how to optimise our run time.  So far we're not seeing any real benefit from using multithreading and I'm trying to work out why.

The data below was compiled by running a decent sized C test on our MCU and times are in seconds.

| Threads | Execution Time |
| :--- | :---|
| 1 | 176 |
| 2 | 189 |
| 4 | 144 |
| 6 | 165 |
| 8 | 166 |
| 10 | 180 |
| 12 | 189 |
```

I've run the 4 thread case with profiling enabled and I get this output from gprof:

```
Flat profile:

Each sample counts as 0.01 seconds.
  %   cumulative   self              self     total
 time   seconds   seconds    calls   s/call   s/call  name
 67.47    198.62   198.62 16368132     0.00     0.00  VlMTaskVertex::waitUntilUpstreamDone(bool) const
  9.94    227.87    29.25  2007781     0.00     0.00  VlWorkerThread::dequeWork(VlWorkerThread::ExecRec*)
  2.58    235.46     7.59  5411940     0.00     0.00  Vchip_sim_tb___024root____Vthread__eval__1(void*, bool)
  1.65    240.32     4.87  9396749     0.00     0.00  Vchip_sim_tb___024root____Vthread__eval__2(void*, bool)
  1.59    245.00     4.68   503926     0.00     0.00  Vchip_sim_tb___024root____Vthread__eval__0(void*, bool)
  1.40    249.12     4.12   233180     0.00     0.00  Vchip_sim_tb___024root____Vthread__eval__3(void*, bool)
  0.43    250.38     1.25 15260626     0.00     0.00  VlWorkerThread::addTask(void (*)(void*, bool), void*, bool)
  0.30    251.26     0.88 14798051     0.00     0.00  VerilatedScope::exportFind(VerilatedScope const*, int)
<snip>
```

And here's the gantt chart of the threads.

![gantt](https://user-images.githubusercontent.com/99925711/223432465-cc798f20-7199-4e89-b64e-080263adb163.jpg)

So where do I go from here?

shareefj, 2023-03-07:
Using profile guided optimisation makes the 4 thread case worse.  142 seconds vs 137.

wsnyder, 2023-03-07:
Can you post the text output of the Gantt report?

shareefj, 2023-03-07:
```
Verilator Gantt report

Argument settings:
  +verilator+prof+exec+start+1
  +verilator+prof+exec+window+2

Analysis:
  Total threads             = 4
  Total mtasks              = 44
  Total cpus used           = 4
  Total yields              = 9
  Total evals               = 2
  Total eval loops          = 2
  Total eval time           = 1631538 rdtsc ticks
  Longest mtask time        = 446690 rdtsc ticks
  All-thread mtask time     = 4245246 rdtsc ticks
  Longest-thread efficiency = 27.4%
  All-thread efficiency     = 65.0%
  All-thread speedup        = 2.6

Prediction (what Verilator used for scheduling):
  All-thread efficiency     = 94.3%
  All-thread speedup        = 3.8

MTask statistics:
  min log(p2e) = -4.686  from mtask 1 (predict 27, elapsed 2926)
  max log(p2e) = 3.394  from mtask 27 (predict 113218, elapsed 3800)
  mean = -0.598
  stddev = 1.558
  e ^ stddev = 4.751

CPUs:
  cpu 0: cpu_time=1191908
  cpu 2: cpu_time=1056970 socket=0 core=1  AMD Ryzen 9 3900X 12-Core Processor
  cpu 4: cpu_time=1508600 socket=0 core=2  AMD Ryzen 9 3900X 12-Core Processor
  cpu 6: cpu_time=487768 socket=0 core=3  AMD Ryzen 9 3900X 12-Core Processor

Writing profile_exec.vcd
```

wsnyder, 2023-03-08:
>Total eval time           = 1631538 rdtsc ticks
Assuming an around 3 GHz core, this corresponds to around 1800 cycles/second. What are you seeing?  This seems an order of magnitude slower than what I would expect from a "small MCU" design.  Anyhow if around 1KHz this is a big enough model Verilator should be doing better.

>  All-thread speedup        = 2.6
Verilator's data says you should be getting 2.6x rather than the 1.2x wall time.  Are you sure you ran the gantt report at a representative cycle (e.g. not startup)?

>67.47    198.62   198.62 16368132     0.00     0.00  VlMTaskVertex::waitUntilUpstreamDone(bool) const
This means a thread is waiting for an earlier mtask. This is the idle times in the gantt report.  The gantt report estimated 65% efficiency or 33% wait time, which is quite different from the 67% wait in this function, so that also suggests something inconsistent in the data.

shareefj, 2023-03-08:
I'm running a 3900X so base clock of 3.8GHz.  My C++ wrapper (I'm using OpenTitan's SimCtrl) prints the following statistics:
```
Simulation statistics
=====================
Executed cycles:  5270158
Wallclock time:   158.856 s
Simulation speed: 33175.7 cycles/s (33.1757 kHz)
```

Regarding the Gantt report, I notice that there are two plusargs that control a start value and window.  How should these be set?  Am I right in thinking that the default just logs 2 eval loops from eval loop count 1?  So I want to pick some number further into my model running when it's doing "useful" stuff?

wsnyder, 2023-03-08:
Yes, pick some cycles well into "normal" operation.  You can pick a lot more than 2 cycles if you want, but the data will get large, which is why that is not the default.

shareefj, 2023-03-08:
OK, I ran a few cycles this morning and actually, each eval loop looks pretty much the same which I guess is what you'd expect. 

Here is some data for you when sweeping across 2-12 cores:

```
{
    "6f4f524a0b499b489b450f9acf095791d9f6b2a6": {
        "2": {
            "cycles": "571909",
            "freq": "2415.47",
            "wallclock": "236.769"
        },
        "4": {
            "cycles": "571909",
            "freq": "2677.08",
            "wallclock": "213.632"
        },
        "6": {
            "cycles": "571909",
            "freq": "2713.62",
            "wallclock": "210.755"
        },
        "8": {
            "cycles": "571909",
            "freq": "2674.23",
            "wallclock": "213.859"
        },
        "10": {
            "cycles": "571909",
            "freq": "2609.47",
            "wallclock": "219.167"
        },
        "12": {
            "cycles": "571909",
            "freq": "2593.66",
            "wallclock": "220.503"
        }
    }
}
```

Looking at the 4 thread data in particular, and after updating the start/window for Gantt generations, I get:

```
Verilator Gantt report

Argument settings:
  +verilator+prof+exec+start+10000
  +verilator+prof+exec+window+4

Analysis:
  Total threads             = 4
  Total mtasks              = 44
  Total cpus used           = 4
  Total yields              = 0
  Total evals               = 4
  Total eval loops          = 4
  Total eval time           = 2754240 rdtsc ticks
  Longest mtask time        = 778278 rdtsc ticks
  All-thread mtask time     = 7564508 rdtsc ticks
  Longest-thread efficiency = 28.3%
  All-thread efficiency     = 68.7%
  All-thread speedup        = 2.7

Prediction (what Verilator used for scheduling):
  All-thread efficiency     = 66.3%
  All-thread speedup        = 2.7

MTask statistics:
  min log(p2e) = 1.696  from mtask 23 (predict 495477, elapsed 90858)
  max log(p2e) = 3.646  from mtask 24 (predict 349445, elapsed 9120)
  mean = 2.479
  stddev = 0.346
  e ^ stddev = 1.413

CPUs:
  cpu 0: cpu_time=2491470
  cpu 2: cpu_time=1398362 socket=0 core=1  AMD Ryzen 9 3900X 12-Core Processor
  cpu 4: cpu_time=1575252 socket=0 core=2  AMD Ryzen 9 3900X 12-Core Processor
  cpu 6: cpu_time=2099424 socket=0 core=3  AMD Ryzen 9 3900X 12-Core Processor

Writing profile.4.vcd
```

and the gprof output is:

```
Flat profile:

Each sample counts as 0.01 seconds.
  %   cumulative   self              self     total
 time   seconds   seconds    calls   s/call   s/call  name
 59.90    139.97   139.97 14177620     0.00     0.00  VlMTaskVertex::waitUntilUpstreamDone(bool) const
 14.97    174.96    34.99  1508542     0.00     0.00  VlWorkerThread::dequeWork(VlWorkerThread::ExecRec*)
  2.79    181.49     6.53 12944892     0.00     0.00  Vchip_sim_tb___024root____Vthread__eval__1(void*, bool)
  2.41    187.11     5.62   553452     0.00     0.00  Vchip_sim_tb___024root____Vthread__eval__0(void*, bool)
  1.66    190.98     3.87   228729     0.00     0.00  Vchip_sim_tb___024root____Vthread__eval__3(void*, bool)
```

shareefj, 2023-03-08:
And the gantt chart.  Which seems to have a predicted parallelism that now more closely matches the measured.

[image]

wsnyder, 2023-03-09:
With the new data your model is a lot smaller/faster than I was thinking so it's possible there isn't much more to be gained after doing the tuning, the relatively flat parallelism results suggest that too.

Did you try profile feedback on the new sampling window?

Did you try profiling the single threaded model?  Any improvements there (in Verilog or Verilator) will likely also make multithreaded faster.  (It's just easier to see what is needed in single threaded.)

BTW you can set CPUs at the same frequency w/ "cpufreq-set", that will help you see the results between experiments more consistently.

shareefj, 2023-03-09:
OK, thanks.  I'll try looking at the single threaded mode and see what that looks like.

What did you mean by "Did you try profile feedback on the new sampling window?"?  Did you mean profile guided optimization?  Is that controlled by the same window length?

shareefj, 2023-03-17:
So I managed to get back to this and have run a single threaded model.  I get the following at the top of my report:

```
Flat profile:

Each sample counts as 0.01 seconds.
  %   cumulative   self              self     total           
 time   seconds   seconds    calls   s/call   s/call  name    
  8.20      5.20     5.20  1286807     0.00     0.00  Vchip_sim_tb___024root___eval__0(Vchip_sim_tb___024root*)
  5.42      8.63     3.44  1286807     0.00     0.00  Vchip_sim_tb___024root___eval__1(Vchip_sim_tb___024root*)
  3.82     11.05     2.42  1286807     0.00     0.00  Vchip_sim_tb___024root___eval__2(Vchip_sim_tb___024root*)
  0.57     11.41     0.36   571915     0.00     0.00  Vchip_sim_tb___024root___multiclk__TOP__18637__PROF__cm3_nvic_main__l1087(Vchip_sim_tb___024root*)
  0.52     11.74     0.33   571915     0.00     0.00  Vchip_sim_tb___024root___multiclk__TOP__19094__PROF__cm3_nvic_int_state__l120(Vchip_sim_tb___024root*)
```

I did also note that we still have some UNOPTFLAT warnings that we thought we'd dealt with so I'll open a second ticket on that. as I can't seem to get rid of them.

But do you expect the majority of the time to be spent in the eval function?

wsnyder, 2023-03-18:
Large evals are not necessarily a problem but you should look at what the code is doing in them. This might indicate that there are a lot of clocks, e.g. from fine grained clock gaters.  Verilator will likely run much faster if your base flops have "if (enable) ...." rather than little fine grained clock AND gates.

shareefj, 2023-03-20:
I guess this is the hard part - I'm not sure what sort of code structure I should be looking out for that might be "bad".

We do have some architectural clock gates in the design and I've just tried bypassing them such that the clocks were free running and this resulted in a 40% performance drop.  So small changes do have quite a large impact.

I've also just gone through the profile data in more detail and noticed that we were spending time in blocks that I didn't think were active.  This highlighted an issue I hadn't thought about - we're currently early in our development cycle so don't have all architectural clock gates in place.  So the model is evaluating stuff that just isn't active every clock cycle.  It looks like quite a lot of time was spent calculating a CRC.  Tying off that clock and re-running resulted in an improvement from 40KHz to 54KHz.  So I guess we'll naturally improve as we add the missing functionality.  Multi-threaded performance still drops though.

wsnyder, 2023-04-08:
Discussion seemed to finish so closing - feel free to follow up.

closed 2023-04-08 by wsnyder